espressif/esp-idf #3904: esp32 configure ble gatt to spp and after perform spp issue of ble bonding old pair device

sagarvegg, 2019-08-09:
## Environment

- Development Kit:     [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v4]
- Module or chip used:   [ESP32-WROVER-B]
- IDF version (run git describe --tags to find it): v4.0-dev-478-gdf61612f8-dirty
- Build System:         [Make]
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [external 5V]

## Problem Description

I perform several steps to perform ble gatt and spp by simple procedure,
first of all advertising ble device that securely connect to nRF connect app with bonding and data of bonding device address and other details auto save into esp chip..
now I send data to particular custom characteristics for enable "spp accepter" and that works fine... 
NOTE: I use another esp custom board use as "spp initiator" 

No any problem for spp transmission but after successfully all operation done and reset board than issue occurs..

Main issue: after [spp initiate and reset board]/ [spp complete and auto reset] or [halt spp and reset board] try to reconnect ble device thats auto bonded delete by "bta_dm_remove_sec_dev_entry (bda);" in bta_dm_act.c file and getting error: authentication fail = 0x51

and main issue I don't understand that " after reset board by any other above 3 case; wait for 5 to 6 minutes than some time don't get authentication fail issue, and successfully connected after passing that interval of time".
 
before reset esp I already deinitialized using "esp_spp_deinit();"

If I don't enable spp transmission and reset board every time esp ble device able to connect with pairing [bond] / already bonded only connect successfully... 

I use esp board for multi connect and only issue occurs with requested [characteristics write for enable spp initiator] central device. other device perform connection successfully without any issue. 

### Expected Behavior

after ble to spp transmission and reset board connect ble device with already bonded central device nRF connect app

### Actual Behavior

after ble to spp transmission and reset board don't able connect ble device with already bonded central device nRF connect app and delete bonded information from bonding list

### Steps to repropduce

1. connect with paring one device to esp board notation as "Device 1"
2. connect with paring second device to esp board notation as "Device 2"
3. now "Device 1" characteristics request for spp accepter
4. another esp board [spp initiator] enable and connection establish successfully
5. perform spp transmission at that time both "Device 1 and Device 2" connected
6. after spp transmission done auto reset esp main board and advertising start for multi device as previous 
7. now try to reconnect "Device 1" --> issue: don't connect and bonding information of "Device 1" deleted from above bond list with all that particular "Deceive 1" data
8. now try to reconnect "Device 2" --> successfully connected as already saved bonded device

If interchange process of "Device 1" and "Device 2" issues occurs vice versa...

sagarvegg, 2019-08-16:
hey all of you if any one know solution than please respond...

gengyuchao, 2019-08-16:
Hey, have you deleted the binding information with ESP32 on your phone?
If not, please delete the binding information on your phone before connecting.

sagarvegg, 2019-08-20:
@gengyuchao I don't need to delete binding information with ESP32 on my phone cause of we require that already connected device don't ask again for authentication [bonding] process.

our need to direct reconnect already binding device if already that device in binding list...

and second thing: after any above declared reset method only that device have binding issue which requested to SPP transmission, and don't get this issue if after reset we wait for 4 to 5 minutes and try to reconnect [that time don't get issue of reconnect]