espressif/esp-idf #3905: unit-test-app "GPIO falling edge interrupt test" no deinit after fail

ChrisIdema, 2019-08-09:
## Environment

- Development Kit:      DOIT
- Kit version: [v1
- Module or chip used:  ESP32-WROOM-32
- IDF version : v3.1.5 (also on v3.1.3)
- Build System:        Make
- Compiler version:1.22.0-80-g6c4433a
- Operating System:     Ubuntu 18.04.2
- Power Supply:         USB

## Problem Description

If I build and run unit-test-app the interrupts on pin 19 are not disabled after test fail.
This causes two problems:
1. toggling pin 19 prints messages: GPIO[19] intr, val: 1
2. running the test again produces this error: E (109179) gpio: gpio_install_isr_service(394): GPIO isr service already installed

### Expected Behavior

1. After test failure no more messages afterwards
2. running test a second time should yield the same result: no error message

### Actual Behavior

1. After test failure pin 19 generates messages
2. running test a second time should yields message: E (109179) gpio: gpio_install_isr_service(394): GPIO isr service already installed

### Steps to repropduce

1. run unit-test-app:
```
cd $IDF_PATH/tools/unit-test-app
make clean
make menuconfig
make TESTS_ALL=1
make flash monitor
enter
```
2. connect 18 + 19
3. 67 + enter (65 in v3.1.3) + enter (pass)
4. toggle 19 (no messages)
5. disconnect 18 + 19
6. 67 + enter (failure, no additional error)
7. toggle 19 (messages)
8. 67 + enter (65 in v3.1.3) (failure + error message)

## Debug Logs

```
Running GPIO falling edge interrupt test...
I (85921) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (85922) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
GPIO[19] intr, val: 0
/home/e2/esp/esp-idf/components/driver/test/test_gpio.c:179:GPIO falling edge interrupt test:PASS
Test ran in 235ms

-----------------------
1 Tests 0 Failures 0 Ignored 
OK
Enter next test, or 'enter' to see menu
Running GPIO falling edge interrupt test...
I (94168) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (94168) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
/home/e2/esp/esp-idf/components/driver/test/test_gpio.c:196:GPIO falling edge interrupt test:FAIL: Expected 0 Was 1
MALLOC_CAP_8BIT potential leak: Before 228900 bytes free, After 228544 bytes free (delta 356)
MALLOC_CAP_32BIT potential leak: Before 263704 bytes free, After 263348 bytes free (delta 356)

Test ran in 142ms

-----------------------
1 Tests 1 Failures 0 Ignored 
FAIL
Enter next test, or 'enter' to see menu
GPIO[19] intr, val: 1
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 0
GPIO[19] intr, val: 1

-----------------------
0 Tests 0 Failures 0 Ignored 
OK
Enter next test, or 'enter' to see menu
Running GPIO falling edge interrupt test...
I (109169) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (109170) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
E (109179) gpio: gpio_install_isr_service(394): GPIO isr service already installed
/home/e2/esp/esp-idf/components/driver/test/test_gpio.c:196:GPIO falling edge interrupt test:FAIL: Expected 0 Was 1

Test ran in 133ms

-----------------------
1 Tests 1 Failures 0 Ignored 
FAIL
Enter next test, or 'enter' to see menu
GPIO[19] intr, val: 0

```

wilhelmw-97, 2020-06-18:
The problem is the test installs the driver's isr and after the test fails, it does not perform clean-ups and does not uninstall this isr. In the next call, the test tries to install the isr again when it is no uninstalled, and fails and produces the message you've seen.

As of now, the test app, after a failed assertion (indicating that the test failed), was not written to clean up and be responsible for the success of further tests (but will always clean up correctly if the test passes). So this undefined behavior is ....kind of..... the expected behavior.

Patsjemoe, 2020-07-14:
Hi,
So what is then the solution ?
Do we need to change the driver, the program ....?
regards

wilhelmw201, 2020-07-15:
Hi,

May I ask what are you using this program for? These tests are usually run one at a time, is primarily meant for firmware testing for those who tweak with drivers, and requires the GPIOs be correctly connected.

ChrisIdema, 2020-07-15:
I posted this almost a year ago (I haven't checked if the error still persists).
The unit test should simply de-initialize every resource used after the test or reset the microcontroller.
If the testpins are connected, the test should succeed.
If the testpins are not connected, the test should fail.
The unit-test menu allows you to repeat the test without resetting the microcontroller so it should produce the same results every time.

wilhelmw201, 2020-08-10:
Yes.. We have only a partial support of the test platform (it's called unity test platform). You have to reset the controller every time after performing a test.

closed 2023-09-21 by Alvin1Zhang