espressif/esp-idf #3906: I2C timeout

tatulea, 2019-08-09:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it):  v3.2-2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

Hi,

I have a timeout problem while trying to read BQ27220 device number.
I posted it here (https://www.esp32.com/viewtopic.php?f=13&t=11847) with a lot more details

negativekelvin, 2019-08-09:
`i2c_set_timeout(I2C_NUM_0, 400000);`

tatulea, 2019-08-12:
Hi @negativekelvin, it works. Could you tell me why the "400000" is the "magic number"?

closed 2019-08-15 by tatulea

alprshn, 2023-10-11:
bro try it. I solved problem
SSD1306Wire display(0x3c, I2C_SDA, I2C_SCL, GEOMETRY_128_64, I2C_ONE, -1);
You can look this link https://github.com/ThingPulse/esp8266-oled-ssd1306