espressif/esp-idf #3907: Crash in bootloader with encrypted flash: bug introduced by commit d77c74770a879fa25058bea78719d5e66ab34bdd

cfillot, 2019-08-09:
In "components/esp32/cpu_start.c" the following code has been added in commit d77c74770a879fa25:

```
    bootloader_flash_update_id();
#if !CONFIG_SPIRAM_BOOT_INIT  // If psram is uninitialized, we need to improve some flash configuration.
    esp_image_header_t fhdr;
    const esp_partition_t *partition = esp_ota_get_running_partition();
    spi_flash_read(partition->address, &fhdr, sizeof(esp_image_header_t));
    bootloader_flash_clock_config(&fhdr);
    bootloader_flash_gpio_config(&fhdr);
    bootloader_flash_dummy_config(&fhdr);
    bootloader_flash_cs_timing_config();
#endif
```

However, when flash encryption is used, "spi_flash_read" does return raw data from flash, not the decrypted data.

spi_flash_read_encrypted() or esp_partition_read() should be used to correctly read data (-> `esp_partition_read(partition, 0, &fhdr, sizeof(esp_image_header_t));` )

Alvin1Zhang, 2019-08-20:
@cfillot  Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

negativekelvin, 2019-08-20:
@Alvin1Zhang the information here fully documents the issue and provides the fix, it is not necessary

projectgus, 2019-08-20:
Hi @cfillot ,

Thanks for reporting this. We have a fix for this and some related flash encryption regressions in our internal review and merge queue, this issue will be updated when they're merged (should be soon).

Angus

projectgus, 2019-10-14:
Hi @cfillot ,

Sorry for the long delay between replies. This should be fixed in release/v4.0 with commit fae124a as linked above, and similar commits are present in master branch and v3.x release branches as needed.

closed 2019-10-14 by projectgus