espressif/esp-idf #3908: UDP

jponko, 2019-08-10:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

## Environment

- ESP32-LyraT v4.3
- IDF version (run ``git describe --tags`` to find it): 
   v3.2.2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux Mint 19.2 Tina
- Power Supply:         USB

## Problem Description
example program udp_client does not work nor does any udp client
//Detailed problem description goes here.
I'm trying to use ESP32 to send UDP packets with various messages to my Android app. The example program is supposed to demenstrate sending UDP messages to a ip address over wireless connection as a station. The program is configured as IPV4 only , destination address 192.168.1.121 (my Linux laptop running "Packet Sender".. The esp32 connects to the router and is configured with ip address  192.168.1.9. The progarm runs (bug note: it shows an ipv6 address when app configured for only IPV4) and here is the monitor output:
------------------------ monitor output ------------------------------
I (0) cpu_start: App cpu up.
I (331) heap_init: Initializing. RAM available for dynamic allocation:
I (338) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (344) heap_init: At 3FFB8DD0 len 00027230 (156 KiB): DRAM
I (350) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (357) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (363) heap_init: At 4008FFAC len 00010054 (64 KiB): IRAM
I (369) cpu_start: Pro cpu start user code
I (52) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (142) wifi: wifi driver task: 3ffc09f8, prio:23, stack:3584, core=0
I (152) wifi: wifi firmware version: 693c7b6
I (152) wifi: config NVS flash: enabled
I (152) wifi: config nano formating: disabled
I (152) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (162) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (192) wifi: Init dynamic tx buffer num: 32
I (192) wifi: Init data frame dynamic rx buffer num: 32
I (192) wifi: Init management frame dynamic rx buffer num: 32
I (202) wifi: Init management short buffer num: 32
I (202) wifi: Init static rx buffer size: 1600
I (212) wifi: Init static rx buffer num: 10
I (212) wifi: Init dynamic rx buffer num: 32
I (212) example: Setting WiFi configuration SSID linksys...
I (312) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0
I (312) wifi: mode : sta (c4:4f:33:14:ef:4d)
I (322) example: Waiting for AP connection...
I (322) example: SYSTEM_EVENT_STA_START
I (442) wifi: n:7 2, o:1 0, ap:255 255, sta:7 2, prof:1
I (442) wifi: state: init -> auth (b0)
I (442) wifi: state: auth -> assoc (0)
I (562) wifi: state: assoc -> run (10)
I (732) wifi: connected with linksys, channel 7
I (732) wifi: pm start, type: 1

I (1642) event: sta ip: 192.168.1.9, mask: 255.255.255.0, gw: 192.168.1.1
I (1642) example: SYSTEM_EVENT_STA_GOT_IP
I (2142) example: SYSTEM_EVENT_STA_GOT_IP6
I (2142) example: IPv6: FE80::C64F:33FF:FE14:EF4D
I (2142) example: Connected to AP
I (2142) example: Socket created
I (2152) example: Message sent
------------------------------------- END monitor---------------------------------------
The "Packet Sender" program doesn't receive any packets from the esp32 and a Wireshark capture confirmes that no packets are sent because it cannot communicate to the laptop at 192.168.1.121 (host unreachable). I suspect there is no arp address in its cache for the destination since it can't ping the destination. Wireshark show icmp pings from the esp32 to 192.168.1.121 as host unreachable. However, the laptop can ping the esp32 (with some 3% loss) as shown below:
...
64 bytes from 192.168.1.9: icmp_seq=584 ttl=255 time=471 ms
64 bytes from 192.168.1.9: icmp_seq=585 ttl=255 time=93.8 ms
64 bytes from 192.168.1.9: icmp_seq=586 ttl=255 time=311 ms
64 bytes from 192.168.1.9: icmp_seq=587 ttl=255 time=539 ms
64 bytes from 192.168.1.9: icmp_seq=588 ttl=255 time=460 ms
64 bytes from 192.168.1.9: icmp_seq=589 ttl=255 time=171 ms
64 bytes from 192.168.1.9: icmp_seq=590 ttl=255 time=609 ms
^C
--- 192.168.1.9 ping statistics ---
590 packets transmitted, 568 received, 3% packet loss, time 590399ms
rtt min/avg/max/mdev = 3.363/537.590/3842.343/532.952 ms, pipe 4
------------------------------- END Laptop ping of ESP32-lyraT -------------------

Additional information:
1) The laptop and Esp32 are located within 10 feet (3meters) of the ASUS RT-AC88U wireless router.
2) I've tried this code with the ESP32 configured as an Access Point but with the same results.
3) I've tried Linux udp client code on the ESP32 with the same results.
4) Oddly enough, the udp echo server works but I suspect it's because the "Packet Sender" sends a packet first to the server so that it extracts the  

### Expected Behavior
To send a UD packet to the destination ip address and to receive UDP packets from the destination ip when sent and then to display them in monitor console.
### Actual Behavior
Niether sends or receives UDP packets

____________________________
Wireshark capture attached:
[esp32-lyraT-udp_client.pcapng.zip](https://github.com/espressif/esp-idf/files/3488423/esp32-lyraT-udp_client.pcapng.zip)

### Steps to reproduce

1. make menuconfig and set Example Confguration settings for SSID, password, IP Version IPV4, set ipv4 address for destination, Port (use defautt) , make flash monitor
2. Setup destination to capture UDP packets (PAcket Sender, Netcat, etc)
3. make flash monitor.  ..

// It helps if you 4 a picture of your setup/wiring here.

### Code to reproduce this issue
Get code from Espressif Systems · GitHub 
example programs udp_client in esp-idf/examples/protocols/sockets

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3488443/sdkconfig.zip)
[udp_client.elf.zip](https://github.com/espressif/esp-idf/files/3488445/udp_client.elf.zip)
[udp_client.zip](https://github.com/espressif/esp-idf/files/3488446/udp_client.zip)

sagb2015, 2019-08-14:
Hi @jponko, 
We are unable to reproduce the issue on our side. 
The ICMP "host unreacable" packets in the capture indicate that ESP32 port 33331 was not reachable. This maybe because of some initial delay at ESP32 between DHCP address reception  till the time recvfrom is called. It will help if you can try the following- 
1) Use nc -u -l 192.168.1.121 -p 33333 command instead of PacketSender ( parameters are per sdkconfig you have shared)
2) Provide wireless sniffer capture if you still observe the issue.

jponko, 2019-08-14:
Hello sagb2015,

The first time I uploaded the program to the lyra-T board I was able to reproduce the problem. No matter how many times I reset or reprogrammed the lyra-T board I couldn't ping the ip address assigned by the router. So I decided that the esp32 WiFi must be in some sort of weird state so I recycled power to the board by pulling both of the USB cables from the laptop and plugging them back in after 30 seconds. The ping requests then started to work and the message was received by netcat. From then on I couldn't reproduce the problem no matter what I tried with this board.

So I connected a new lyra-T board, programmed it and tried to ping it without success. So this time I pulled the USB cables from the board and plugged them back in and ping started working and nc received the UDP message.

I then unplugged the board waited 10 minutes and plugged it in and the ping was working as it was previously. I then reprogrammed it again and after the reset I was unable to ping it and no matter if I reset, powered down and backup, the ping was still unsuccessful. However, after about 2 minutes, the pings were successful.

This leads me to believe that this is a firmware/OS/hardware problem as I didn't change anything in the program and was able to reproduce this several times even though the board said it was connected to the router but it doesn't seem to be sending or receiving packets.

What procedures did you try to reproduce the problem?

Jerry

sagb2015, 2019-08-16:
Hi Jerry, 
Since ping is not working, the problem is now reduced from UDP to just WiFi connectivity.  Can try the same experiment with getting_started station example? A wireless sniffer capture will help to debug the issue.

jponko, 2019-08-18:
Hi Sagar,

I tried the station app and was unable to get either of my ESP32-lyraT
boards to connect to my router. I didn't see any UDP dhcp client requests
at all, just a few DNS and NTP queries from my laptop so there was nothing
to capture from the esp32-lyraT.
I tried both esp-idf v3.2.2 and v3.3-beta3 to compile the station app.

Here's the console output:

jponko@jponko-ideapad:~/esp/station$ make flash monitor
Toolchain path: /home/jponko/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from
/home/jponko/esp/esp-idf-v3.3-beta3/requirements.txt are satisfied.

Project is not inside a git repository, will not use 'git describe' to
determine PROJECT_VER.
App "wifi_station" version: 1
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____.....___
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding
Scheme None
MAC: c4:4f:33:14:ef:bd
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 24480 bytes to 14613...
Wrote 24480 bytes (14613 compressed) at 0x00001000 in 0.1 seconds
(effective 2108.8 kbit/s)...
Hash of data verified.
Compressed 593888 bytes to 360738...
Wrote 593888 bytes (360738 compressed) at 0x00010000 in 4.8 seconds
(effective 997.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective
8269.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6364
load:0x40078000,len:11308
ho 0 tail 12 room 4
load:0x40080400,len:6700
entry 0x40080764
I (30) boot: ESP-IDF v3.3-beta3-dirty 2nd stage bootloader
I (30) boot: compile time 19:14:06
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
[... 87 lines cut ...]
I (12633) wifi station: connect to the AP fail

I (14693) wifi station: retry to connect to the AP
I (14693) wifi station: connect to the AP fail

I (14813) wifi: new:<5,2>, old:<5,0>, ap:<255,255>, sta:<5,2>, prof:1
I (14813) wifi: state: init -> auth (b0)
I (14813) wifi: state: auth -> assoc (0)
I (14823) wifi: state: assoc -> run (10)
I (22823) wifi: state: run -> init (fc0)
I (22823) wifi: new:<5,0>, old:<5,2>, ap:<255,255>, sta:<5,2>, prof:1
I (22823) wifi station: retry to connect to the AP
I (22823) wifi station: connect to the AP fail

I (22943) wifi: new:<5,0>, old:<5,0>, ap:<255,255>, sta:<5,0>, prof:1
I (22943) wifi: state: init -> auth (b0)
I (22953) wifi: state: auth -> assoc (0)
I (22953) wifi: state: assoc -> run (10)
I (26963) wifi: state: run -> init (2c0)
I (26963) wifi: new:<5,0>, old:<5,0>, ap:<255,255>, sta:<5,0>, prof:1
I (26963) wifi station: retry to connect to the AP
I (26963) wifi station: connect to the AP fail

I (29013) wifi station: connect to the AP fail
:

sagb2015, 2019-08-19:
Hi Jerry, 
It seems like you are facing authentication failure. Can you check if password is okay?

jponko, 2019-08-19:
Hi Sagar,

I did screw up with the password and the station example does work and I
can ping the ESP32. The ping response times are rather inconsistent ranging
from about 32ms to 600ms. Wat did you wanted me to capture with Wireshark?

Jerry

sagb2015, 2019-08-20:
Hi Jerry, 
The variable delay is because of wifi power save which is expected ( You can use esp_wifi_set_ps to disable power save if you wish). To address the original issue, can you try UDP client example again on the same board with the same access point and provide the sniffer capture if it does not work.

jponko, 2019-08-30:
Hi Sagar,

I tried the UDP client program again but with a different destination on
the same access point and it works.

What doesn't seem to make sense is that if I use the computer that the
program is uploaded from as the destination in the UDP client
configuration, I cannot capture any UDP packets coming from the ESP32
including the DHCP request. However, if I setup the ESP32 as an access
point I can connect to the AP and the UDP client works as expected. I
wonder if this may have something to do with the dual band nature of my
access point (2.4GHz or 5GHz).

Thanks for the tip on disabling WiFi power saving mode. I wouldn't have
expected power saving mode to be on as the default.

Jerry
AC9NM

Jerry

sagb2015, 2019-08-31:
Hi Jerry, 
If the ping works fine between ESP32 and your computer, there is no reason because of which UDP traffic cannot go through. The dual band nature of the AP should not be an issue as AP typically takes care of forwarding traffic between clients on different channel. There may be some forwarding rules setup on your machine. If it is a linux machine, you can use "sudo iptables --flush"  to clear the rules and check if it works.

jponko, 2019-09-01:
I tried the iptables --flush when I was able to ping the ESP32 but the UDP
packet sent was never received. I suspect it's because of the rather weak
signal from the AP to the laptop and ESP32. The laptop is showing 75%
signal strength I'm only about 10 meters from the AP but the ESP32 is
having great difficulty obtaining a IPV4 address and regularly gets a IPV6
address even though I've configured the udp_client program for IPV4. The
example program should just timeout saying it can't obtain a IPV4 address
rather than telling me it has obtained a IPV6 address.

The signal between the ESP32 and the access point is extremely weak. It's
either due to a poor built-in antenna or the WiFi chip has problems (WiFi
power saving was disabled). I notice that with the ESP8266 (built-in
antenna) can easily communicate at least 40 meters or more in AP mode with
an Android tablet. I'll have to test how far away I can get from the
ESP32-lyraT board with WiFi setup for AP mode. I haven't looked at the API
but are there some WiFi diagnostic routines like those for the ESP8266 to
show RSSI or RCPI signal strength between the ESP32 and an AP?

Jerry
AC9NM

sagb2015, 2019-09-01:
Hi Jerry, 
Unfortunately, we do not have an API to get RSSI when ESP32 is connected. However, for your testing you can use esp_wifi_scan_get_ap_records API to the RSSI after scanning. You need to keep scanning as you change the distance though.

sagb2015, 2019-10-11:
@jponko, Do you still have any concerns here?

jponko, 2019-10-11:
I haven't had time to try the setting:

Component config-->Bluetooth-->Bluetooth controller-->Bluetooth controller
mode:
Set to : *BR/EDR Only*

Is this set as the default for the examples now?

sagb2015, 2019-10-12:
Please open a separate issue for BT related queries. BT is not relevant for UDP client use-case.

jponko, 2019-10-14:
Sorry, I thought I was replying to a BT issue. No, the example still
doesn't work for me in WiFi station mode but works fine in AP mode. It
obviously has something to do with my Wifi hub or laptop but there is no
way for me to diagnose the issue. So you can close this case.

JP

Alvin1Zhang, 2019-10-15:
@jponko Thanks for the updates and confirmation. @sagb2015 Thanks for the efforts. @jponko Feel free to reopen if you still have the issue, thanks.

closed 2019-10-15 by Alvin1Zhang