espressif/esp-idf #3910: make: execvp: python: Permission denied when using eclipse

Alexgoshih, 2019-08-10:
make: execvp: python: Permission denied
/usr/bin/make -C /home/Alexg/esp/esp-idf/examples/bluetooth/ble_ibeacon/build/app_trace -f /home/Alexg/esp/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/home/Alexg/esp/esp-idf/components/app_trace/component.mk COMPONENT_NAME=app_trace build
make: *** [C:/msys32/home/Alexg/esp/esp-idf/make/project.mk:245: check_python_dependencies] Error 127
make: *** Waiting for unfinished jobs....

Alvin1Zhang, 2019-08-12:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @Alexgoshih

dobairoland, 2019-08-12:
Hi @Alexgoshih. In addition to the "environment" variables suggested in the issue template please also post the value of your PATH variable. You can find it in the Project Properties -> C/C++ Build -> Environment.

Please make sure that you've followed all of the instructions from here:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/eclipse-setup.html

Are you able to build the project from terminal, i.e. by issuing `make all`?

What terminal are you using?

What is the output of `python --version` in your terminal?

Alexgoshih, 2019-08-13:
Thank you for replying.

Eclipce version: 2019-03, 2019-06.
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
PROJECT_PATH= COMPONENT_DIRS= /usr/bin/make -C /home/Alexg/esp/esp-idf/components/bootloader/subproject V=1 BUILD_DIR_BASE=/home/Alexg/esp/esp-idf/examples/bluetooth/ble_ibeacon/build/bootloader TEST_COMPONENTS= TESTS_ALL= EXCLUDE_COMPONENTS= /home/Alexg/esp/esp-idf/examples/bluetooth/ble_ibeacon/build/bootloader/bootloader.bin
python /home/Alexg/esp/esp-idf/tools/check_python_dependencies.py

Eclipce PATH variable is: C:/Program Files/Java/jdk-12.0.1/bin/server;C:/Program Files/Java/jdk-12.0.1/bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Users\Alexg\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\Java\jdk-12.0.1\bin;C:\msys32\mingw32\bin;C:\msys32\opt\xtensa-esp32-elf\bin;C:\msys32\usr\bin 

I can build the project from msys32 by using **make all** and **flash**.

$ python --version
Python 2.7.15

dobairoland, 2019-08-13:
Hi @Alexgoshih. Thanks for the additional information.

Please make sure that you have setup the IDF_PATH variable inside Eclipse as per instruction from here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/eclipse-setup.html#project-properties

Note that the path must be a full path to IDF and a valid Windows path, i.e. `c:\...` and not `/home/...` because Eclipse doesn't operate inside the MSYS environment.

Alexgoshih, 2019-08-17:
My another computer which can use eclipse to compile and flash the project before. Has run into the same issue just after installed the newest updates of windows 10.

dobairoland, 2019-08-19:
Unfortunately, I don't know what those windows updates broke so you need to check the environment variables set-up in Eclipse according to the documentation for the IDF version you use.

If the build works in MSYS then your setup of Eclipse is somehow miss-configured.

It fails on this line:  
```
$(PYTHON) $(IDF_PATH)/tools/check_python_dependencies.py
```
This means that either it cannot invoke python from MSYS and/or the IDF_PATH variable is not set-up in Eclipse correctly. E.g. the windows update could have changed your PATH and now Eclipse cannot find the python interpreter.

dobairoland, 2019-08-22:
Hi @Alexgoshih. I gave this another try. I updated my Windows 10 and then followed the steps from the documentation to set up a clean project. It works as it should be.

Please try to move the three `c:\msys..` paths from the end of PATH to the beginning. That could be a another possible issue.

Alexgoshih, 2019-08-23:
Hello @dobairoland.  Thanks for your suggestion. The problem has been solved after rearrange the PATH variable.

closed 2019-08-23 by Alexgoshih

dobairoland, 2019-08-23:
I'm happy and thank you for reporting back.

dasistdevnull, 2019-10-23:
Moving the path to the beginning worked for me too.