espressif/esp-idf #3912: PICO-D4 Core  0 panic'ed (LoadProhibited)

vangalvin, 2019-08-10:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [X ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [|ESP32-PICO-D4]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB|Battery] 

## Problem Description
I have tested the code on an ESP32-WROOM-32 and works fine however the intended target is a PICO-D4. I am getting the following error imediatly upon startup.

`ets Jun  8 2016 00:22:57

rst:0x9 (RTCWDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6248
load:0x40078000,len:10272
ho 0 tail 12 room 4
load:0x40080400,len:6608
entry 0x40080764
W (63) boot: PRO CPU has been reset by WDT.
W (63) boot: WDT reset info: PRO CPU PC=0x400621ce
W (63) boot: WDT reset info: APP CPU PC=0xbf43c4ca
I (69) boot: ESP-IDF v3.2.2-42-g4a9f33944 2nd stage bootloader
I (76) boot: compile time 10:38:12
I (80) boot: Enabling RNG early entropy source...
I (85) boot: SPI Speed      : 40MHz
I (89) boot: SPI Mode       : DIO
I (93) boot: SPI Flash Size : 4MB
I (97) boot: Partition Table:
I (101) boot: ## Label            Usage          Type ST Offset   Length
I (108) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (116) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (123) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (131) boot:  3 factory          factory app      00 00 00010000 00140000
I (138) boot:  4 ota_0            OTA app          00 10 00150000 00140000
I (146) boot:  5 ota_1            OTA app          00 11 00290000 00140000
I (154) boot:  6 coredump         Unknown data     01 03 003d0000 00010000
[... 687 lines cut ...]
//---main.h
#ifndef __MESH_MAIN__
#define __MESH_MAIN__

#define FIRMWARE_VERSION "0.0.12"

void node_mesh_write(char *data, int len);

#endif

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2019-08-12:
@vangalvin Thanks for reporting the issue, would you help provide the ELF file of the bin file which is programmed? Thanks.

vangalvin, 2019-08-12:
> 

[esp32hotspot-elf.zip](https://github.com/espressif/esp-idf/files/3493082/esp32hotspot-elf.zip)
Hopefully these are the correct ones :)

> @vangalvin Thanks for reporting the issue, would you help provide the ELF file of the bin file which is programmed? Thanks.

krzychb, 2019-08-13:
> I have tested the code on an ESP32-WROOM-32 and works fine however the intended target is a PICO-D4. 
> Any sugestions would be appreciated as to what I have done wrong or if i need to change a setting for compatibility

Hi @vangalvin,

One of suggestions is to check usage of GPIO16 and GPIO17. These pins are available on ESP32-WROOM-32 but used for connecting the embedded flash in ESP32-PICO-D4. I see some GPIOs initialized early in this code but not exact numbers to tell is this may be the issue.

vangalvin, 2019-08-14:
Yes Checked the pinout and GPIO16 and GPIO17 are being used for FLASH_CS and FLASH_SD0