espressif/esp-idf #3913: What did I do and How do I fix it

jhinklejr, 2019-08-12:
ESP32 CAM module

Have been fine for two days - build, flash, and view the startup output when it starts.

Now -- I get this ...

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5640
load:0x40078000,len:12204
load:0x40080400,len:7084
entry 0x4008076c

From here on it's garbage as if the baudrate has changed from 115200.

I replaced my application with the simple WIFI Station code - same output.

I thought that maybe the boot or partition code had change since rebuild does not touch that so I removed the build folder to forced a total new build - same garbage.

Is root-cause a baudrate other than 115200 - if so why did it change and how to fix it.

Is there another possible issue?

Any help or comments please - i'm dead in the water.

Thanks.

Joe

negativekelvin, 2019-08-12:
https://github.com/espressif/esp-idf/issues/959

jhinklejr, 2019-08-12:
It was not a 26 mhz crystal issue as suggested.

New day - now it works as expected - I hate issues I don't understand.

closed 2019-08-12 by jhinklejr