espressif/esp-idf #3917: esp-idf windows  CMD下搭建编译环境  步骤idf.py menuconfig 错误

UserInitpass, 2019-08-13:
在按照官方文档windows10 CMD 进行步骤到正确设置环境变量后，在windows CMD下使用idf.py menuconfig 会出现“undefined variable "$IDF_PATH/components/bt/host/nimble/Kconfig.in"
FAILED: CMakeFiles/menuconfig”
解决方案：替换 esp-idf\components\bt\Kconfig  中391行source "$IDF_PATH/components/bt/host/nimble/Kconfig.in"为 source "esp-idf的绝对路径/components/bt/host/nimble/Kconfig.in"
替换 esp-idf\components\bt\Kconfig  中404行 "$IDF_PATH/components/bt/esp_ble_mesh/Kconfig.in"为 source "esp-idf的绝对路径/components/bt/esp_ble_mesh/Kconfig.in"

然后保存文件，再执行 idf.py menuconfig 就不会出错。

dobairoland, 2019-08-13:
Hi @UserInitpass. The permanent solution for this issue is to upgrade your `mconf-idf`. If you are using the Windows installer then please download and install the latest one (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#esp-idf-tools-installer) which already contains the fixed `mconf-idf`. 

If you prefer to do it manually then the related tools can be found here: https://github.com/espressif/kconfig-frontends/releases/
Note in the changelog of v4.6.0.0-idf-20190628 the following line which is related to this issue:
`Expand mixed environment variables and static text in source lines, similar to wordexp functionality but much simpler (no quoting, no escaping)`.

Alvin1Zhang, 2019-08-23:
@UserInitpass 您好，请问您的问题解决掉了吗？按照 @dobairoland 建议的方法，可否提供一些反馈？谢谢。

dobairoland, 2019-09-09:
I'm closing this for lack of response. Please feel free to reopen it if the problem still exists.

closed 2019-09-09 by dobairoland