espressif/esp-idf #3918: abort() was called at PC 0x400839d7 on core 0

kuba1342, 2019-08-13:
Here is my error:

abort() was called at PC 0x400839d7 on core 0
0x400839d7: vPortAssertIfInISR at /home/jakub/CLionProjects/NS_Internship/prodlinesys_freertos/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

Backtrace: 0x4008684c:0x3ffc04d0 0x40086a1b:0x3ffc04f0 0x400839d7:0x3ffc0510 0x400851c2:0x3ffc0530 0x400856a3:0x3ffc0550 0x40080dbd:0x3ffc0570 0x400d1318:0x3ffc0590 0x400d0880:0x3ffc05c0 0x400818d2:0x3ffc05e0 0x4008152e:0x3ffc0600 0x400d23ad:0x00000000
0x4008684c: invoke_abort at /home/jakub/CLionProjects/NS_Internship/prodlinesys_freertos/amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:676

0x40086a1b: abort at /home/jakub/CLionProjects/NS_Internship/prodlinesys_freertos/amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:676

0x400839d7: vPortAssertIfInISR at /home/jakub/CLionProjects/NS_Internship/prodlinesys_freertos/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

Im using ESP32-WROOM-32D and CLion

The error occured as I was trying to use pointers as function params
`// button test
void isr_button_pressed(void *args) {
    struct LEDTest *test = args; // <----------------------- HERE
    int btn_state = gpio_get_level(test->button_gpio);
    if (btn_state == 0) {
        gpio_set_level(test->LED_gpio, 0);
    } else {
        gpio_set_level(test->LED_gpio, 1);
    }
}

void interrupt_test(void *args) {
    struct LEDTest *test = args; // <----------------------- HERE
    configure_button(test->button_gpio);

    gpio_install_isr_service(0);
    gpio_isr_handler_add(test->button_gpio, isr_button_pressed, test);

    // Wait
    while(1) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }

}`

This is from my main:
`LEDIndicator yellowLED = {.GPIOPort = 26, .mode = GPIO_MODE_OUTPUT, .time = 1000, ._TaskHandler = NULL};
    LEDIndicatorInit(&yellowLED);
    printf("LED configured");

    struct LEDTest test = {.button_gpio = 25, .LED_gpio = 26};

    xTaskCreate(&interrupt_test, "testInterrupt", 2048, &test, 5, NULL);`

Im trying to make LED blinking when I press the button. When I was just using params declared inside functions (not pointers) everything was working properly. Now everytime I press the button it keeps rebooting.

Thanks in advance for any help on this.

Alvin1Zhang, 2019-08-13:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @kuba1342

boarchuz, 2019-08-13:
I suspect "struct LEDTest test" has gone out of scope when the above functions use their pointer to it.

kuba1342, 2019-08-14:
@boarchuz Thanks for reply. Do you have any suggestions how can I fix this? I tried using malloc but result was the same

boarchuz, 2019-08-14:
It's probably not that then. You're going to have to provide more info.
To test, I would've just moved "struct LEDTest test" to global scope. If it still aborts, it's something else.

Dazza0, 2019-08-19:
@kuba1342 
```
Backtrace: 0x4008684c:0x3ffc04d0 0x40086a1b:0x3ffc04f0 0x400839d7:0x3ffc0510 0x400851c2:0x3ffc0530 0x400856a3:0x3ffc0550 0x40080dbd:0x3ffc0570 0x400d1318:0x3ffc0590 0x400d0880:0x3ffc05c0 0x400818d2:0x3ffc05e0 0x4008152e:0x3ffc0600 0x400d23ad:0x00000000
```
It would be helpful if you could decode the entire backtrace using `addr2line`. It will help narrow down what lead to the call of `vPortAssertIfInISR()`

closed 2021-01-22 by projectgus