espressif/esp-idf #3920: build fails if project is empty git repo

morganrallen, 2019-08-14:
## Environment

- Development Kit: ESP32-PICO-D4
- IDF version: v4.0-dev-1443-g39f090a4f
- Build System: CMake
- Compiler version:crosstool-ng-1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description
Part of the cmake build systems tries to determine the revision of the current project (not esp-idf version). If the project has run `git init` but there are no commits (therefor no revision) the build will fail will the output below. This looks to start right around https://github.com/espressif/esp-idf/blob/master/tools/cmake/third_party/GetGitRevisionDescription.cmake#L48

# Build Output

```
CMake Error: File /tmp/esp32-mcurses/.git/packed-refs does not exist.
CMake Error at build/CMakeFiles/git-data/grabRef.cmake:27 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/project.cmake:37 (git_describe)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/project.cmake:337 (__project_get_revision)
  CMakeLists.txt:3 (project)

CMake Error at build/CMakeFiles/git-data/grabRef.cmake:28 (file):
  file failed to open for reading (No such file or directory):

    /tmp/esp32-mcurses/build/CMakeFiles/git-data/packed-refs
Call Stack (most recent call first):
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/project.cmake:37 (git_describe)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/project.cmake:337 (__project_get_revision)
  CMakeLists.txt:3 (project)

CMake Warning (dev) at build/CMakeFiles/git-data/grabRef.cmake:29 (if):
  uninitialized variable 'PACKED_REFS'
Call Stack (most recent call first):
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/project.cmake:37 (git_describe)
  /home/morgan/devel/ESP32/esp-idf/tools/cmake/project.cmake:337 (__project_get_revision)
  CMakeLists.txt:3 (project)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

renzbagaporo, 2019-08-21:
You can get around this by specifying PROJECT_VER in top-level CMakeLists.txt or specifying the version in a `version.txt` file in the project directory.

morganrallen, 2019-08-21:
Sure, you can also work around it by making a commit, it's still broken.

renzbagaporo, 2019-08-27:
Hi @morganrallen. Fix for this is on the way.

closed 2019-08-30 by igrr

elasticdotventures, 2020-07-28:
jeezus just spent more time than I should have figuring out wtf.

leoribg, 2023-09-14:
OMG, this is still happening on ESP-IDF V5.1 :(

sarahkjls, 2023-12-09:
> OMG, this is still happening on ESP-IDF V5.1 :(

Yap same here did you find something out in order to fix it? or did you just downgrade?

leoribg, 2023-12-12:
> > OMG, this is still happening on ESP-IDF V5.1 :(
> 
> Yap same here did you find something out in order to fix it? or did you just downgrade?

You can set PROJECT_VER variable in project CMakeLists.txt file, or initialize a git repository at your project and make a commit.

Check out this section -> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#build-project-variables

kryvel, 2024-01-03:
> 
> You can set PROJECT_VER variable in project CMakeLists.txt file, or initialize a git repository at your project and make a commit.
> 
> Check out this section -> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#build-project-variables

There are no requirements to set PROJECT_VER in documentation. It says

>  Otherwise, PROJECT_VER will be "1".

So this strange behaviour should be fixed. Should I open new issue or this one could be reopened ?

igrr, 2024-01-03:
@kryvel opening the new issue is probably the best option! Please mention the IDF version, operating system you are using, the version of git, and attach the full log of `idf.py build`. Thank you.