espressif/esp-idf #3921: Could not find the file: "lwip/apps/sntp.h"

innc11, 2019-08-14:
## Environment
- Development Kit:     none
- Module or chip used:  ESP32
- IDF version: esp-idf-v3.3-rc
- Build System:   Make
- Compiler version :1.22.0-61-gab8375a
- Operating System:   Windows
- Power Supply:   USB

## Problem Description

SNTP example code failed to compile, Could not find the file: "lwip / apps / sntp.h"
terminal output: 
```
Administrator@shengyang7 ~/esp-idf-v3.3-rc/sntp
$ make all
fatal: Not a git repository (or any parent up to mount point /cygdrive/d)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: Not a git repository (or any parent up to mount point /cygdrive/d)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
WARNING: Toolchain version is not supported: 1.22.0-61-gab8375a
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
fatal: Not a git repository (or any parent up to mount point /cygdrive/d)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: Not a git repository (or any parent up to mount point /cygdrive/d)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
WARNING: Toolchain version is not supported: 1.22.0-61-gab8375a
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/main/sntp_example_main.o
/home/aithinker/esp-idf-v3.3-rc/sntp/main/sntp_example_main.c:24:28: fatal error: lwip/apps/sntp.h: No such file or directory
compilation terminated.
make[1]: *** [/cygdrive/d/AiThinkerIDE_V0.5/cygwin/home/aithinker/esp-idf/make/component_wrapper.mk:286：sntp_example_main.o] 错误 1
make: *** [/cygdrive/d/AiThinkerIDE_V0.5/cygwin/home/aithinker/esp-idf/make/project.mk:468：component-main-build] 错误 2
```

### Code to reproduce this issue

```cpp
file: esp-idf-v3.3-rc\examples\protocols\sntp\main\sntp_example_main.c
/* LwIP SNTP example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <time.h>
#include <sys/time.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_attr.h"
#include "esp_sleep.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/apps/sntp.h"

/* The examples use simple WiFi configuration that you can set via
   'make menuconfig'.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_WIFI_SSID CONFIG_WIFI_SSID
#define EXAMPLE_WIFI_PASS CONFIG_WIFI_PASSWORD
......
```

KonstantinKondrashov, 2019-08-14:
Hi @innc11! I have checked it now for v3.3 and it works for me. I do not see any errors.
Could you try `git submodule update` and `make clean` maybe it will help.

innc11, 2019-08-14:
> Hi @innc11! I have checked it now for v3.3 and it works for me. I do not see any errors.
> Could you try `git submodule update` and `make clean` maybe it will help.

Thank you, the problem disappeared after I downloaded ESP_IDF again.

closed 2019-08-14 by innc11