espressif/esp-idf #3922: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

kuba1342, 2019-08-14:
## Environment

Hardwere: ESP32
IDE used: Clion
Operating System: Linux
Version of Amazon FreeRTOS: 201906.00_Major-225-g523c08a3
Build System: CMake
Power Supply: USB cable

## Problem Description

Im trying to establish mqtt connection on my ESP based on this [documentation](https://www.freertos.org/Documentation/api-ref/mqtt/iot__mqtt_8h.html#ad3c7dfde82e2bd71c2da9f2752d0249c). I found that some functions from mqtt_connect example are not working. When I run monitor it shows that AP connected successfully, expected address IS NOT: 0.0.0.0 but after Connecting... ```I get Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.``` and then it reboots. After that ESP tries to connect once again but this this access point connection is failed, expected address is 0.0.0.0 and after few seconds I get ```Failed to establish new MQTT connection, error NETWORK ERROR.``` and ```Connect failure. Error 5```

### Code to reproduce this issue

```c
MqttInit(void){
    IotMqttError_t status = IOT_MQTT_SUCCESS;

    status = IotMqtt_Init();
    if (status != IOT_MQTT_SUCCESS){
        ESP_LOGW(MQTT_TAG, "Init failure");
        return;
    }

    status = _MqttConnect();
    if (status != IOT_MQTT_SUCCESS){
        ESP_LOGW(MQTT_TAG, "Connect failure");
        return;
    }
}

IotMqttError_t _MqttConnect(void){
// Parameters to MQTT connect.
    IotMqttConnection_t mqttConnection = IOT_MQTT_CONNECTION_INITIALIZER;
    IotMqttNetworkInfo_t networkInfo = IOT_MQTT_NETWORK_INFO_INITIALIZER;
    IotMqttConnectInfo_t connectInfo = IOT_MQTT_CONNECT_INFO_INITIALIZER;
    IotMqttPublishInfo_t willInfo = IOT_MQTT_PUBLISH_INFO_INITIALIZER;
// Example network abstraction types.
    IotNetworkServerInfo_t serverInfo = AWS_IOT_NETWORK_SERVER_INFO_AFR_INITIALIZER;
    /*
    {
            .pHostName = clientcredentialMQTT_BROKER_ENDPOINT,
            .port = clientcredentialMQTT_BROKER_PORT
    };
    */
    ESP_LOGI(MQTT_TAG, "Host: %s, port: %d", serverInfo.pHostName, serverInfo.port);
    IotNetworkCredentials_t credentialInfo = AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER;
    /*
    {
            .maxFragmentLength = 0,
            .disableSni = false,
            .pRootCa = NULL,
            .rootCaSize = 0,
            .pClientCert = keyCLIENT_CERTIFICATE_PEM,
            .clientCertSize = sizeof(keyCLIENT_CERTIFICATE_PEM),
            .pPrivateKey = keyCLIENT_PRIVATE_KEY_PEM,
            .privateKeySize = sizeof(keyCLIENT_PRIVATE_KEY_PEM)
    };
     */
    IotNetworkInterface_t networkInterface = {
            .create             = IotNetworkAfr_Create,
            .send               = IotNetworkAfr_Send,
            .receive            = IotNetworkAfr_Receive,
            .setReceiveCallback = IotNetworkAfr_SetReceiveCallback,
            .close              = IotNetworkAfr_Close,
            .destroy            = IotNetworkAfr_Destroy
    };
// Example using a generic network implementation.
    networkInfo.createNetworkConnection = true;
    networkInfo.u.setup.pNetworkServerInfo = &serverInfo;
    networkInfo.u.setup.pNetworkCredentialInfo = &credentialInfo;
#ifdef BLE_ENABLED
    networkInfo.pNetworkInterface = IOT_NETWORK_INTERFACE_BLE;
#else
    networkInfo.pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;
#endif
// Set the members of the connection info (password and username not used).
    connectInfo.cleanSession = true;
    connectInfo.keepAliveSeconds = 300;
    connectInfo.pClientIdentifier = clientcredentialIOT_THING_NAME;
    connectInfo.clientIdentifierLength = sizeof(clientcredentialIOT_THING_NAME) - 1;
// Set the members of the will info (retain and retry not used).
    willInfo.qos = IOT_MQTT_QOS_1;
    willInfo.pTopicName = "will/topic/name";
    willInfo.topicNameLength = 15;
    willInfo.pPayload = "MQTT client unexpectedly disconnected.";
    willInfo.payloadLength = 38;
// Set the pointer to the will info.
    connectInfo.pWillInfo = &willInfo;
// Call CONNECT with a 5 second block time. Should return
// IOT_MQTT_SUCCESS when successful.

    uint32_t expectedIp = SOCKETS_GetHostByName( clientcredentialMQTT_BROKER_ENDPOINT );
    char expectedIpBuffer[16] = "";
    /* Convert to string. */
    SOCKETS_inet_ntoa( expectedIp, expectedIpBuffer );
    ESP_LOGI( MQTT_TAG, "expected ip: %s", expectedIpBuffer );

    ESP_LOGI(MQTT_TAG, "Connecting...");
    IotMqttError_t result = IotMqtt_Connect( &networkInfo,
                                             &connectInfo,
                                             5000,
                                             &mqttConnection );
    if(result != IOT_MQTT_SUCCESS){
        ESP_LOGE(MQTT_TAG, "Connect failure. Error %d", result);
    }
    return result;
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (3377) [ESP32_Kuba3][WiFi]: Connected to AP success
4 306 [main] [INFO ][MQTT][3060] MQTT library successfully initialized.
I (3387) [ESP32_Kuba3][MQTT]: Host: a24k1vf8fdqaq9-ats.iot.eu-central-1.amazonaws.com, port: 8883
I (3647) [ESP32_Kuba3][MQTT]: expected ip: 35.158.242.21
I (3647) [ESP32_Kuba3][MQTT]: Connecting...
5 906 [main] [INFO ][MQTT][9060] Establishing new MQTT connection.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4009346a  PS      : 0x00060a30  A0      : 0x80091599  A1      : 0x3ffd4950  
0x4009346a: vListInsert at /home/jakub/CLionProjects/NS_Internship/prodlinesys_freertos/amazon-freertos/freertos_kernel/list.c:150 (discriminator 1)

A2      : 0x3ffc3b2c  A3      : 0x3ffd4d0c  A4      : 0x00001388  A5      : 0x3ffd4b50  
A6      : 0x3fff2d9c  A7      : 0x3ffd4b54  A8      : 0x00000000  A9      : 0x3ffc3b34  
A10     : 0x00000000  A11     : 0x00000018  A12     : 0x00000000  A13     : 0x401198b0  
0x401198b0: _IotMqtt_ProcessSend at /home/jakub/CLionProjects/NS_Internship/prodlinesys_freertos/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_operation.c:832

A14     : 0x3ffe373c  A15     : 0x3fff2db8  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4009346a:0x3ffd4950 0x40091596:0x3ffd4970 0x40090e57:0x3ffd4990 0x40174e4e:0x3ffd49d0 0x40174e6f:0x3ffd4a10 0x40173a50:0x3ffd4a40 0x40173acf:0x3ffd4a80 0x40119321:0x3ffd4ae0 0x40118152:0x3ffd4b10 0x400d2c79:0x3ffd4b40 0x400d2cb8:0x3ffd4c20 0x400d2e0b:0x3ffd4c40 0x400d8d5a:0x3ffd4c60

```
and after reboot
```
W (2698) [ESP32_Kuba3][WiFi]: Connected to AP Fail
2 241 [main] [INFO ][MQTT][2410] MQTT library successfully initialized.
I (2708) [ESP32_Kuba3][MQTT]: Host: a24k1vf8fdqaq9-ats.iot.eu-central-1.amazonaws.com, port: 8883
I (13308) [ESP32_Kuba3][MQTT]: expected ip: 0.0.0.0
I (13308) [ESP32_Kuba3][MQTT]: Connecting...
3 2501 [main] [ERROR][NET][25010] Failed to resolve a24k1vf8fdqaq9-ats.iot.eu-central-1.amazonaws.com.
4 2501 [main] [ERROR][MQTT][25010] Failed to establish new MQTT connection, error NETWORK ERROR.
E (25308) [ESP32_Kuba3][MQTT]: Connect failure. Error 5
W (25318) [ESP32_Kuba3][MQTT]: Connect failure
```

Thanks in advance!

Dazza0, 2019-08-19:
@kuba1342  Could you decode the backtrace addresses using `addr2line` so that we can find where the program crashed?

Alvin1Zhang, 2020-05-28:
@kuba1342 Would you please help share if any updates for the issue? Thanks.

mahavirj, 2020-09-17:
Closing due to lack of update, please feel free to re-open if there is any update or more information available.