espressif/esp-idf #3923: crash after succesful OTA over BLE

chegewara, 2019-08-14:
Hi,
i am trying to write OTA over BLE example and it is working, but after restart i have multiple times app crash. 
Full story:
- simple BLE client app which is connecting to other esp32 working as ble server,
- on button write to ble server characteristic, now ble server starts sending indications with binary data
- ble client receive data and perform ota flash,
- basically the same code has been working about 9-10 months ago with esp-idf v3.x,
- currently i am trying to work with esp-idf master branch, before OTA update ble client works fine, issue is after OTA update with exactly the same binary,
- this is interesting 
```
ELF file SHA256: ad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0ba
```
Environment:
- ubuntu 16.04
- both CMake and make
- esp-idf master
- different esp32 boards
- elf file: [hello-world.elf.txt](https://github.com/espressif/esp-idf/files/3500459/hello-world.elf.txt)

Here are logs:
```
I (445) esp_image: segment 5: paddr=0x002675d8 vaddr=0x40080400 size=0x12648 ( 75336) load
I (497) boot: Loaded app from partition at offset 0x190000
I (497) boot: Disabling RNG early entropy source...
I (498) cpu_start: Pro cpu up.
I (501) cpu_start: Application information:
I (506) cpu_start: Project name:     hello-world
I (511) cpu_start: App version:      1
I (516) cpu_start: Compile time:     Aug 14 2019 09:15:39
I (522) cpu_start: ELF file SHA256:  4d3c7dd765532430...
I (528) cpu_start: ESP-IDF:          v4.0-dev-1443-g39f090a
I (534) cpu_start: Starting app cpu, entry point is 0x40081228
0x40081228: call_start_cpu1 at /home/chegewara/esp/esp-idf/components/esp32/cpu_start.c:281

I (524) cpu_start: App cpu up.
I (545) heap_init: Initializing. RAM available for dynamic allocation:
I (552) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (558) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (564) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (570) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (576) heap_init: At 3FFC3870 len 0001C790 (113 KiB): DRAM
I (582) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (588) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (595) heap_init: At 40092A48 len 0000D5B8 (53 KiB): IRAM
I (601) cpu_start: Pro cpu start user code
I (620) spi_flash: detected chip: generic
I (620) spi_flash: flash io: dio
I (621) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (768) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 2
// esp_app_desc_t dump from image which is used to flash
E (1958) compare_app_versions: 32 54 cd ab 00 00 00 00 00 00 00 00 00 00 00 00 
E (1958) compare_app_versions: 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (1958) compare_app_versions: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (1968) compare_app_versions: 68 65 6c 6c 6f 2d 77 6f 72 6c 64 00 00 00 00 00 
E (1978) compare_app_versions: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (1988) compare_app_versions: 30 39 3a 31 35 3a 33 39 00 00 00 00 00 00 00 00 
E (1988) compare_app_versions: 41 75 67 20 31 34 20 32 30 31 39 00 00 00 00 00 
E (1998) compare_app_versions: 76 34 2e 30 2d 64 65 76 2d 31 34 34 33 2d 67 33 
E (2008) compare_app_versions: 39 66 30 39 30 61 00 00 00 00 00 00 00 00 00 00 
E (2018) compare_app_versions: 4d 3c 7d d7 65 53 24 30 08 77 8c 77 f0 9b e9 9a 
E (2028) compare_app_versions: 78 4c e8 e4 e5 39 44 89 bd 1d 5e 48 d1 61 09 1d 
E (2028) compare_app_versions: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (2038) compare_app_versions: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (2048) compare_app_versions: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (2058) compare_app_versions: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (2068) compare_app_versions: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
// esp_app_desc_t from currently running partition after flash (newly flashed image)
E (2068) compare_app_versions: 32 54 cd ab 00 00 00 00 00 00 00 00 00 00 00 00 
E (2078) compare_app_versions: 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... 170 lines cut ...]
E (2164) TAG: Starting OTA example...
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x40127ae8  PS      : 0x00060e34  A0      : 0x800466c6  A1      : 0x3ffbe210  
0x40127ae8: r_platform_reset at ??:?

A2      : 0x00000000  A3      : 0x40127ae8  A4      : 0x00000000  A5      : 0x00000010  
0x40127ae8: r_platform_reset at ??:?

A6      : 0x3ffb93cc  A7      : 0x3ffb8360  A8      : 0x80019fb8  A9      : 0x00002060  
A10     : 0xf2f2f2f2  A11     : 0x0000a55a  A12     : 0x00000001  A13     : 0x00000022  
A14     : 0x00001600  A15     : 0x3ffb8360  SAR     : 0x00000013  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400900b7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40127ae8
0x400900b7: portENTER_CRITICAL_NESTED at /home/chegewara/esp/esp-idf/components/freertos/include/freertos/portmacro.h:324
 (inlined by) xPortInIsrContext at /home/chegewara/esp/esp-idf/components/freertos/port.c:338

0x40127ae8: r_platform_reset at ??:?

ELF file SHA256: ad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0ba

Backtrace: 0x40127ae5:0x3ffbe210 0x400466c3:0x3ffbe240 |<-CORRUPTED
0x40127ae5: r_get_stack_usage at ??:?
```

chegewara, 2019-08-17:
Hi @Weijian-Espressif 
i think i found something useful that can help track the cause of this issue:
- when i set this somewhere at beggining of my code `	esp_log_level_set("*", ESP_LOG_ERROR);
` there is no longer crash,
- when i set to `esp_log_level_set("*", ESP_LOG_WARN);` there is crash like described above,
- with `esp_log_level_set("*", ESP_LOG_ERROR);` i can see this warn message `W (647) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration` then it crash, without that warn log there is no crash

mvonflotow, 2019-08-29:
Hi @chegewara 
Were you able to solve this? I am experiencing a very similar problem when trying to use either the SPI flash API or OTA APIs in a separate task from an active BLE server. 

I can stop the crash from happening by calling these APIs from my BLE server callback function, such that the callback does not return before the APIs are called.

chegewara, 2019-08-29:
Hi @mvonflotow 
like i described above, when i change log level then i dont see crashes anymore, also it takes 2 or 3 crashes on restart (only right after OTA update) and then runs normally.

mvonflotow, 2019-08-30:
Thanks for the input, @chegewara. Unfortunately my crash still occurs even with the log verbosity set to "error," as you suggested above.

This seems to be related to the coexistence of WiFi and BLE in my project. Can you tell me if you had both of these protocols active when your crashes occurred? What is your setting in menuconfig -> component config -> Wi-Fi -> Software controls WiFi/Bluetooth coexistence?

chegewara, 2019-08-30:
https://github.com/chegewara/esp32-OTA-over-BLE

mvonflotow, 2019-09-17:
I was able to stop this bug from occurring by setting CONFIG_FREERTOS_UNICORE in make menuconfig ("Run Freertos only on first core").

Not a long-term solution, but certainly a work around.

chegewara, 2019-10-18:
Yes, i can confirm that unicore works as workaround.

AxelLin, 2022-03-03:
@Alvin1Zhang 

Why do you close this issue?
The espressif-bot added "Resolution: Done", so what is the resolution?
Note, CONFIG_FREERTOS_UNICORE=y is not a resolution at all, it's just a workaround with sacrifice of hw performance.

AxelLin, 2023-03-06:
@Alvin1Zhang @blueMoodBHD 
How is the progress of this issue since it's marked as "In Progress" for 1 year?

xiewenxiang, 2023-03-15:
Currently, a fix has been made for ESP32 that maybe solve this issue. You can test it with newer versions of IDF, such as the master branch.

You need to open this configuration item:
`menuconfig -> Component config -> Bluetooth -> Controller Options -> [*] High level interrupt`

xiewenxiang, 2023-03-16:
The Bluetooth library was rebuilt based on IDF version(v4.0-dev-1443-g39f090a).
You can replace the library and retest it.
[libbtdm_app.a.zip](https://github.com/espressif/esp-idf/files/10986259/libbtdm_app.a.zip)

jack0c, 2023-04-12:
@AxelLin I think this was closed since it was consider same as https://github.com/espressif/esp-idf/issues/7335, and closed by bot

AxelLin, 2023-04-12:
> @AxelLin I think this was closed since it was considered same as #7335, and closed by bot

https://github.com/espressif/esp-idf/issues/7335#issuecomment-1057536196 explictly mentioned BLE is not used in https://github.com/espressif/esp-idf/issues/7335,
so I don't think it's the same issue.