espressif/esp-idf #3924: NimBLE Max Connection limit

sportybj, 2019-08-14:
the NimBLE Max concurrent is too limit 1-10, compare to 32 on platforms  with less resources, say nrf52832.  My questions are:
is it possible now to increase it by modify  CONFIG_BT_NIMBLE_MAX_CONNECTIONS=32   in sdkconfig file?
is the limit of 10 due to hardware resources or just a temporary limit for this version?

dhrishi, 2019-08-14:
@sportybj  The limit is because the ESP32 controller supports maximum 9 connections (BTDM_CTRL_BLE_MAX_CONN). 

On a different note, we will update the value from 10 to 9.

sportybj, 2019-08-14:
> @sportybj The limit is because the ESP32 controller supports maximum 9 connections (BTDM_CTRL_BLE_MAX_CONN).
> 
> On a different note, we will update the value from 10 to 9.
thnx for your quick reply.  I will be pleased if you could let me know it is possible there is or will be soon any workaround or update to have more connections,  up to 32.

dhrishi, 2019-08-14:
That won't be possible as 9 is the maximum connections supported by the controller. Even if the host supports more (say 32) the same won't help.

sportybj, 2019-08-15:
> That won't be possible as 9 is the maximum connections supported by the controller. Even if the host supports more (say 32) the same won't help.
Thnx for ur quick reply

closed 2019-08-15 by sportybj