espressif/esp-idf #3925: Reset + boot mode not working

tatulea, 2019-08-14:
Hi,

I have a custom board with a ESP-WROVER-B. There is a CP2102N that has the reset and boot circuit implemented as it is in the ESP-DevkitC schematic (RTS, DTR, two transistors, EN and IO0).
The problem is that the board is not getting into "download mode" in order to flash it, read the MAC address or anything else that I can do using esptool.

I had a look with a logic analyzer for EN and IO0 and EN goes low for a bit, then it goes high and in the same time IO0 goes low for a bit and then high. Basically, they go low and high one after another. I think that IO0 should be low in the same time as EN and go high after EN goes high.

What can be wrong?

jeremyherbert, 2019-08-15:
Do you have a revision A02 CP2102N? The revisions before that have a critical bug which makes them useless on linux. See https://github.com/espressif/esptool/issues/399 

I have tested rev A02 and it seems to work fine.

projectgus, 2019-08-15:
Hi @tatulea ,

A better place to post this kind of question would be the https://esp32.com forums, as it isn't related to an issue in the ESP-IDF software framework.

Without seeing a schematic of your board and an image of the logic analyzer capture then it's not really possible to know what the issue is. If taking a logic analyzer capture then suggest also capturing the UART TXD0 pin, as this way you can see when the chip has started up.

The automatic reset circuit (with the two transistors) does mean that there can be a window when both EN & IO0 are released together, while DTR & RTS are asserted. This window should be short as RTS is then released and IO0 goes low, to start into bootloader mode.

On Windows sometimes this window is still too long, so recent Espressif development boards include a capacitor between EN & GND to slow the EN rising edge and work around this issue. Without the capacitor the board may fail to start up into bootloader mode reliably. See this issue for a lot of the details: https://github.com/espressif/esptool/issues/136 - or see any recent Espressif development board schematic for an example of the fix using a capacitor.

As mentioned above, if you post a thread on the esp32.com forums with some images then we'll be happy to help you there.

Angus

closed 2019-08-15 by projectgus