espressif/esp-idf #3926: a2dp source using adc dma (i2s) not working.

uched41, 2019-08-14:
## Environment

- Development Kit:     ESP32-DevKitC 
- Kit version : v1
- IDF version : v4.0-dev-1443-g39f090a4f, v3.2
- Build System:  CMake
- Compiler version: 5.2
- Operating System: Linux
- Power Supply:  USB

## Problem Description
I have been trying to get the a2dp source example to work with ADC DMA. I am able to run the basic example and send noise audio. But when I configure and connect my analog microphone ( which I have confirmed to have decent audio quality) I get bad and barely audible audio on my Bluetooth headset.

Problems
- Barely recognizable audio with I2S_ADC DMA.
- I  get a real rate which is half the requested rate when using apll.
```
APLL: Req RATE: 44100, real rate: 22049.980, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 5644795.000, SCLK: 705599.375000, diva: 1, divb: 0
```
But I suspect this is more of a debug problem, because if the real sample rate is less than the standard 44100 for a2dp, the data callback function will crash. 
- Also if I set use_apll to false, the a2dp data callback function crashes after a few calls.

### Steps to reproduce
1. Clone latest esp-idf SDK
2. Copy a2dp source example.
3. Add the init_microphone() function, and call it at the end of the app_main function.
4. Edit the bt_app_a2d_data_cb function to match below.

### Code to reproduce this issue

```cpp
void init_microphone(void)
{
	 int i2s_num = I2S_NUM_0;
	 i2s_config_t i2s_config = {
            .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
            .sample_rate =  44100,
            .bits_per_sample = 16,
	    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
	    .channel_format = I2S_CHANNEL_FMT_ALL_LEFT,
	    .intr_alloc_flags = 0,
	    .dma_buf_count = 4,
	    .dma_buf_len = 512,
	    .use_apll = true,
	 };

	 //install and start i2s driver
	 ESP_ERROR_CHECK(i2s_driver_install(i2s_num, &i2s_config, 0, NULL));

	 //init ADC pad
	 ESP_ERROR_CHECK(i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_7));
	 adc1_config_width(ADC_WIDTH_BIT_12);
   	 adc1_config_channel_atten(ADC1_CHANNEL_7, ADC_ATTEN_DB_11);

	 vTaskDelay(5000/portTICK_RATE_MS);	// Delay necessary for reliable start
	 i2s_adc_enable(I2S_NUM);

	 SYSCON.saradc_ctrl2.sar1_inv = 1; // Inversion, to get uninverted values from i2s
}

// a2dp data call back function
static int32_t bt_app_a2d_data_cb(uint8_t *data, int32_t len)
{
    if (len < 0 || data == NULL) {
        return 0;
    }

    size_t bytes_read;
    int16_t d1[len/2];
    i2s_read(I2S_NUM, (void*) d1, len, &bytes_read, portMAX_DELAY);    // read adc data from dma

    int countt = 0;
    int16_t raw = 0;
    for(int i=0; i<len/2; i++){
        raw = (d1[i] & 0xfff)*16 ;    // Scale 12 bit data to 16 bit
        d1[i] = raw;
    }
    memcpy(data, d1, len);
    return len;
}
```

It is possible that my problem may be in the way I process the data in the a2dp call back function. I know that a2dp supports 16 bit PCM data at 44100 Hz. 
People seem to be getting good results with older releases.
Could the problem I am having be a bug somewhere in newer releases.

jason-mao, 2019-08-23:
@uched41 How bad is the audio? Did you test ADC audio as follow : `ADC`->`File`?

uched41, 2019-08-23:
Hello, I have not tested with a file. The audio is really bad. You cannot decipher what is being said. I tried to send a sine wave and the audio was clear. I was able to get good results.
I suspect the problem could be with the i2s stream.
What do you think about my code for converting the analog signal to 16bit pcm data for a2dp?

```   
   i2s_read(I2S_NUM, (void*) d1, len, &bytes_read, portMAX_DELAY);    // read adc data from dma
   int countt = 0;
    int16_t raw = 0;
    for(int i=0; i<len/2; i++){
        raw = (d1[i] & 0xfff)*16 ;    // Scale 12 bit data to 16 bit
        d1[i] = raw;
    }
    memcpy(data, d1, len);
```

X-Ryl669, 2019-09-05:
Set apll to false, since in the TRM they recommend not to use it.

admiralmaggie, 2020-04-26:
@uched41 Were you ever able to get this working?

admiralmaggie, 2020-05-03:
I go the A2DP source working with DCA I2S ADC. Here is the code if anyone is interested:

https://github.com/admiralmaggie/ESP32_BT_SRC_ADC

rujha, 2020-09-10:
@admiralmaggie I am new to ESP. was trying to run your code, but seeing the issue while compilation on Arduino.
attached the error code.
[error_1.txt](https://github.com/espressif/esp-idf/files/5201276/error_1.txt)

admiralmaggie, 2020-09-10:
@rujha this code meant for ESP-IDF framework. You would need to port it / modify it for Arduino framework.

rujha, 2020-09-14:
@admiralmaggie I compiled on ESP-IDF framework and loaded on esp-32 module. Source esp-32 is able to connect to sink esp-32.
When connecting the MIC (MAX9814) to headphone, audio is clear, but when I provide the out of MAX9814 to PIN 36 (ADC channel 0) of source esp-32,  i am not able to hear any sound from the sink output ( using external DAC MAX98357). 
log message doesn,t show any output either on source esp-32.

13:53:54.302 -> [0;33mW (158595) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###[0m
13:53:54.302 -> [0;33mW (158595) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 4, 0[0m
13:53:54.337 -> [0;33mW (158615) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###[0m
13:53:54.337 -> [0;33mW (158625) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 6, 0[0m
13:53:54.407 -> [0;33mW (158675) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###[0m
13:53:54.407 -> [0;33mW (158685) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 3, 0[0m
13:53:54.440 -> [0;33mW (158715) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###[0m
13:53:54.440 -> [0;33mW (158715) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 5, 0[0m
13:53:54.440 -> [0;33mW (158735) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###[0m
13:53:54.474 -> [0;33mW (158745) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 7, 0[0m
13:53:54.474 -> [0;33mW (158775) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###[0

can you please point out if I am doing something wrong.

thanks,
Rupak
[no_data.txt](https://github.com/espressif/esp-idf/files/5216507/no_data.txt)

admiralmaggie, 2020-09-14:
@rujha That example was created to source ADC input to a Blue Tooth speaker sink and not a DAC.

rujha, 2020-09-14:
@admiralmaggie tried connecting with bluetooth speaker, but still there is no audio. 
any pointer will be helpful.

lakshminarayanaks79, 2021-01-08:
Hi everyone, I need some help with i2s could you be able to answer me i am doing a project to read an ECG signal and sen it via Bluetooth but I need to Send the raw value from the sensor to DMA via i2s my signal is analog can it be possible to do it bcoz I didn't find a way to integrate ADC with DMA.

please do help me with my question.

Thanks in advance

ESP-YTGerd, 2023-09-22:
Hi, @uched41 
It is recommended to use **i2s_read** in a non-blocking way, or just set (portMAX_DELAY = 0), since call function **bt_app_a2d_data_cb** will be passed to the btc layer and called back. So when **i2s_read** is blcoked, the **bluedroid btc task**  also will be blcoked.
[image]

Indastri, 2024-01-09:
Since there was no response from the user closing this issue. If the issue still persists feel free to reopen this or create a new issue. Cheers.

closed 2024-01-09 by Indastri