espressif/esp-idf #3927: android provisioning app building fails

work-media, 2019-08-15:
source code : [provisioning-android](https://github.com/espressif/esp-idf-provisioning-android)
building output tip: Directory '/Users/sagetsang/Destop/esp-idf-provisioning-android-master/app/build/extracted-include-protos/main' specified for property '$3' does not exist

Alvin1Zhang, 2019-08-15:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @work-media

work-media, 2019-08-15:
Could you provide an apk so that I can test wifi provisioning directly?

FayeY, 2019-08-15:
https://github.com/espressif/esp-idf-provisioning-android/releases

closed 2019-08-16 by work-media