espressif/esp-idf #3928: ESP32 - I2S - DMA - Questions

jhinklejr, 2019-08-15:
I posted these questions on the ESP32 forum but no one from Espressif answered them so I thought I would try you folks since my questions are about hardware and drivers.  

I am using igrr's Camera demo code as a base line for my questions but they are not about his implementation so I did not post the questions on his github site.

My questions are on DMA operations and how fields in the DMA descriptors affect DMA operation.

I hope someone here will reply:

I'm trying to fully understand the CAM demo by igrr - but it seems the demo code does not agree with ESP32 documentation - hence these questions.

From The ESP32 Tech ref manual on DMA - descriptors:

descriptor parameter EOF set to identify the end of the DMA link list (also NEXT = 0)
descriptor parameter SIZE is equal to size of buffer
descriptor parameter LENGTH is size of data to be acquired by this descriptor.

Q1: The tech manual states that if actual data count is less than LENGTH - descriptor is complete and DMA moves to next in link list. Does DMA descriptor return any information as to the actual data count acquired?

igrr's demo code sets every descriptor's EOF to 1 which appears to contradict the information in the tech document. I know the igrr's implementation uses a circular array of descriptors instead of of a fixed link-list size.

Q2: why does igrr's code set each DMA descriptor's EOF to 1? (contrary to technical documentation) - this is realyya DMA descriptor relate question.

Q3: If no circular link-list is used but a set of descriptors whose size is the number of camera lines - is only the last descriptor's EOF one?

The tech document on I2S camera processing states that data is passed when HREF is 1 (given the other required are also 1).
I believe the design goal is to process one DMA descriptor for each camera line of data (HREF == 1)

Q4: Since HREF is just a requirement for I2S to pass data to the DMA - what happens if less data is passed than identified in the DMA descriptor's LENGTH parameter? 

Q5: Does the DMA descriptor's stay active until enough data from the next camera line is acquired from the next camera line until LENGTH is acquired? 

Q6: If NOT -- then what terminates the current DMA descriptor? HREF going to 0?

Thanks for any answers or comments.

igrr, 2019-08-19:
Hi @jhinklejr,
Sorry, the code you are referring to was written before the TRM chapter existed, so there quite possibly are inconsistencies there. I'll answer based on what I remember about that code, but please keep in mind that the officially supported repository is now at https://github.com/espressif/esp32-camera. For further i2s-camera related issues, please use that one.

> Q2: why does igrr's code set each DMA descriptor's EOF to 1? (contrary to technical documentation) - this is realyya DMA descriptor relate question.

I think this is so that an IN_DONE interrupt is generated after every DMA block, and not at the end of the chain (which is circular, so there's no end).

> Q3: If no circular link-list is used but a set of descriptors whose size is the number of camera lines - is only the last descriptor's EOF one?

The list is indeed circular, refer to https://github.com/igrr/esp32-cam-demo/blob/14b6170060d03847ece370013be7448b25be1f19/components/camera/camera.c#L482.

> Q4: Since HREF is just a requirement for I2S to pass data to the DMA - what happens if less data is passed than identified in the DMA descriptor's LENGTH parameter?

The DMA engine will time out if no data comes for some time, i think, the timeout is configured in I2S_LC_HUNG_CONF_REG. The camera driver doesn't touch that register so default values apply.

> Q5: Does the DMA descriptor's stay active until enough data from the next camera line is acquired from the next camera line until LENGTH is acquired?

If the DMA doesn't time out before the next line, it will stay active. In practice it times out, given the typical timespan between lines.

> Q6: If NOT -- then what terminates the current DMA descriptor? HREF going to 0?

No, DMA engine works more or less independently from the rest of the I2S module. I2S is responsible for acquiring the data (in serial or parallel format), arranging it in a certain way, and pushing words into the FIFO. The DMA engine reads from FIFO and moves values into RAM.

closed 2019-08-19 by igrr

giPeteR, 2020-12-24:
Sorry for bumping an old thread, but I just wanted to **thank both of you** for a very good problem description and also excellent answers!
I've been struggling for years trying to get **I2S multichannel ADC DMA** working. This post wasn't the solution, but it really helped me to understand the circular buffers used in the linked list descriptors. And the DMA now works, but that's another [story](https://github.com/espressif/esp-idf/pull/1991#issuecomment-541061718).