espressif/esp-idf #3929: Compliance with standard IEC 60335-1

asweeney99, 2019-08-15:
In Europe, CE marking is a certification mark that indicates conformity with health, safety, and environmental protection standards for commercial products.

One of these standards, IEC 60335-1 covers the safety and security of household electronic appliances. IEC 60335-1 describes the requirements for “software-related controls” for automatic electrical controls for household and similar uses. The requirements relate to both the microcontroller and the application running on it and are summarised in the following table:

[image]

Microcontrollers such as ESP32 that are a component of an automatic electrical control must comply with these requirements if they want to pass 60335-1 testing and receive CE marking. 

Would it be possible to create a test library to meet the 60335-1 standard? Please correct me if I'm wrong, but I don't think this is something we can do because some of the requirements require very low-level programming. Compliance with this standard would increase the marketability of esp32 and make it more attractive to product developers. Thank you!

Alvin1Zhang, 2019-08-20:
@asweeney99 Thanks for reporting this. IEC 60335-1 is not applicable for modules, since it is the CE standard for home applicance. The Wi-Fi standard for modules is EN300328. Thanks.

negativekelvin, 2019-08-20:
@Alvin1Zhang ok but you miss the point, he is asking for support in developing a framework that will allow a IEC 60335-1 compliant product to be built using esp32.

asweeney99, 2019-08-20:
EN 300328 covers wideband data systems operating in the 2.4GHz ISM band, including WiFi, Bluetooth, and Zigbee. This is a harmonised standard against which manufacturers must issue their Declaration of Conformity for the Radio and Telecommunications Terminal Equipment (R&TTE) Directive. This allows Espressif to market and sell ESP32.

But users buy ESP32 to make stuff. So if I use ESP32 to build a device, say a smart heating control, and I like it so much I want to market it and sell it, I must submit my device for testing. In Europe, my device must comply with three separate standards:

	1. Electromagnetic radiation (EMC Directive 2014/30/EU)
	2. Radio Directive (EU Directive 2014/53/EU)
	3. Product safety (Low Voltage Directive 2014/35/EU)

Showing the EN300328 certificate is useful for showing compliance with requirement 2. But what about number 3 - product safety? The low voltage directive ensures that electrical equipment within certain voltage limits is safe. It refers to standards EN 60335-1 and EN 60730. By the way, EN is just the European standard which is based on IEC, the international standard.

Specifically for microcontrollers, Annexe H of IEC 60730 sets out the requirements for Electronic Controls. These include test and diagnostic methods to ensure that the operation of embedded control hardware and software for appliances is safe.

Alvin1Zhang, 2019-08-20:
@asweeney99 Thanks for the information, will relay the information to our internal team for evaluation. Thanks.

negativekelvin, 2019-08-20:
It may be difficult to show that an internet connected mcu running a network stack and multi threaded os can provide the appropriate safety model which is why it is common to have another mcu for that.

asweeney99, 2019-08-20:
A very good technical explanation of this topic is given [here](https://www.nxp.com/files-static/training_pdf/WBNR_FTF10_ENT_F0714.pdf).

igrr, 2019-08-20:
Hi @asweeney99, this requirement indeed did come up a few times.

It is not that complicated to provide a test library for the purposes of the IEC 60730 Annex H. 
However additional questions arise when it comes to building a real product which requires Wi-Fi/Bluetooth protocol stacks and the OS to run on the ESP32, in addition to the (safety-critical) control functions. For the entire application, there are two options:
1) partition the application so that it can be shown that the safety-critical part can not be affected by the OS and protocol stacks, and certify the safety-critical part, and
2) certify the entire application.

Option 2) is not viable due to the amount of code involved in the operation of the OS and protocol stacks. Section H.11.12.3 "Measures to avoid errors" lists many requirements which are not satisfied for the majority of the code in the SDK.

Option 1) would have been viable if we had a way to isolate the safety-critical "control" part of the application from the rest of it. We can perform some amount of isolation by running these two parts on different CPUs, but unfortunately in the ESP32 we lack a feature to isolate the IRAM/DRAM of the safety-critical part from corruption by the rest of the application. ESP32 does have a PID controller but it serves the opposite purpose: to isolate a small untrusted bit of code from accessing the resources of the application.

The only remaining option, as @negativekelvin points out, is to run the safety critical part on an entirely separate MCU. This MCU can also be an ESP32, in which case the test function library would be required.

Last thing to note, the measures needed to achieve IEC 60730 compliance often depend on the evaluator, and their interpretation of the requirements. If you find that you don't need to demonstrate full isolation between the safety critical part and the rest of the application due to some other measures taken, then such solution may still be viable. For instance, if it can be shown that an additional external watchdog is a sufficient safeguard against a fault caused by memory corruption of the safety critical part, then the isolation may not be required. We can't recommend this as a general practice for Class B/C products, though.

asweeney99, 2019-08-21:
So this is doable. That's very good news. It means we have the cost of an extra chip, but the chip prices are low enough to make it workable. But what about the resources required to write the test library? This is normally something done by the MCU manufacturer. Could you help with that?

negativekelvin, 2019-08-21:
esp32 is pretty cheap, but is it really cheaper than another low end mcu that is capable of filling the need and already has the support?

asweeney99, 2019-08-21:
Ok thanks for the advice and suggestions guys, I will see if I can find an inexpensive mcu with the necessary support.

closed 2019-09-09 by asweeney99

rftafas, 2024-09-20:
For the sake of helping people getting here because of a web search, we now have [ESP-BIST](https://github.com/espressif/esp-bist).

5ami, 2025-07-05:
Is the esp32-s3 world controller and permission controller peripherals suitable for this kind of applications. Would it provide the necessary separation between the general application (in one core) and the safety critical part in another core ?
Is the BIST library available for the esp32s3 ?