espressif/esp-idf #3930: Cannot run "idf.py menuconfig" in HelloWorld.

piotrkundu, 2019-08-16:
Running "idf.py menuconfig" in esp-idf/examples/get-started/hello_world an latest master

`
idf.py menuconfig
Checking Python dependencies...
Python requirements from /Users/piotrk/_git/esp-idf/requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory /Users/piotrk/_git/esp-idf/examples/get-started/hello_world/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /Users/piotrk/_git/esp-idf/examples/get-started/hello_world"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.20.1 (Apple Git-117)")
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0-dev-1443-g39f090a4f-dirty
-- Building ESP-IDF components for target esp32
CMake Warning at /Users/piotrk/_git/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain
  /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc version 5.2.0
  is not the supported version 8.2.0.  Check Getting Started documentation or
  proceed at own risk.
Call Stack (most recent call first):
  /Users/piotrk/_git/esp-idf/components/esp32/project_include.cmake:20 (gcc_version_check)
  /Users/piotrk/_git/esp-idf/tools/cmake/build.cmake:287 (include)
  /Users/piotrk/_git/esp-idf/tools/cmake/build.cmake:440 (__build_process_project_includes)
  /Users/piotrk/_git/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Warning at /Users/piotrk/_git/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain
  /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc crosstool-ng
  version crosstool-ng-1.22.0-80-g6c4433a doesn't match supported version
  esp32-2019r1.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  /Users/piotrk/_git/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  /Users/piotrk/_git/esp-idf/tools/cmake/build.cmake:287 (include)
  /Users/piotrk/_git/esp-idf/tools/cmake/build.cmake:440 (__build_process_project_includes)
  /Users/piotrk/_git/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

cmake failed with exit code -11
`

`git log
commit 39f090a4f1dee4e325f8109d880bf3627034d839 (HEAD -> master, origin/master, origin/HEAD, origin/4.0)
Merge: 6517c5033 26fc858ea
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Aug 6 11:16:17 2019 +0800

    Merge branch 'deprecate/spi_common_claim_macros' into 'master'
    spi_common: remove deprecated spi_common_periph_claim macros
    See merge request espressif/esp-idf!5677`

piotrkundu, 2019-08-16:
```
git log
commit 39f090a4f1dee4e325f8109d880bf3627034d839 (HEAD -> master, origin/master, origin/HEAD, origin/4.0)
Merge: 6517c5033 26fc858ea
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Aug 6 11:16:17 2019 +0800

    Merge branch 'deprecate/spi_common_claim_macros' into 'master'
    spi_common: remove deprecated spi_common_periph_claim macros
    See merge request espressif/esp-idf!5677
```

piotrkundu, 2019-08-16:
Looks like this issue -> https://github.com/espressif/esp-idf/issues/3712

piotrkundu, 2019-08-17:
I managed to run `idf.py menuconfig` but I still get the warning:
```
CMake Warning at /Users/piotrk/_git/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain
  /Users/piotrk/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc crosstool-ng
  version crosstool-ng-1.22.0-80-g6c4433a doesn't match supported version
  esp32-2019r1.  Check Getting Started documentation or proceed at own risk.
```

What am I doing wrong here?

negativekelvin, 2019-08-17:
There is a newer toolchain if you follow the latest getting started guide https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html

piotrkundu, 2019-08-19:
Thank you @negativekelvin that solved the issue

closed 2019-08-19 by piotrkundu