espressif/esp-idf #3931: Regression: sntp_init() crashes when network not yet initialized.

vonnieda, 2019-08-16:
Using ESP-IDF 3.2 (from today) and calling sntp_init() before WiFi is initialized, there is a crash:

```
/Users/jason/esp/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40086f05 on core 0
0x40086f05: xQueueGenericReceive at /Users/jason/esp/esp-idf/components/freertos/queue.c:2038

Backtrace: 0x40089a18:0x3ffb9e20 0x40089c45:0x3ffb9e40 0x40086f05:0x3ffb9e60 0x401112b6:0x3ffb9ea0 0x4011132e:0x3ffb9ec0 0x40111648:0x3ffb9ee0 0x401116b1:0x3ffb9f00 0x4010ca51:0x3ffb9f20 0x4010ca72:0x3ffb9f40 0x4010619b:0x3ffb9f60 0x400d25ef:0x3ffb9f80 0x400d25f7:0x3ffb9fa0 0x400d26c0:0x3ffba000 0x400d0d72:0x3ffba090 0x4008668d:0x3ffba0b0
0x40089a18: invoke_abort at /Users/jason/esp/esp-idf/components/esp32/panic.c:707

0x40089c45: abort at /Users/jason/esp/esp-idf/components/esp32/panic.c:707

0x40086f05: xQueueGenericReceive at /Users/jason/esp/esp-idf/components/freertos/queue.c:2038

0x401112b6: sys_mutex_lock at /Users/jason/esp/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:444

0x4011132e: sys_arch_protect at /Users/jason/esp/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:469

0x40111648: do_memp_malloc_pool at /Users/jason/esp/esp-idf/components/lwip/lwip/src/core/memp.c:459

0x401116b1: memp_malloc at /Users/jason/esp/esp-idf/components/lwip/lwip/src/core/memp.c:459

0x4010ca51: udp_new at /Users/jason/esp/esp-idf/components/lwip/lwip/src/core/udp.c:1078

0x4010ca72: udp_new_ip_type at /Users/jason/esp/esp-idf/components/lwip/lwip/src/core/udp.c:1078

0x4010619b: sntp_init at /Users/jason/esp/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c:693

0x400d25ef: initialize_sntp at /Users/jason/esp/esp-idf/examples/protocols/sntp/main/sntp_example_main.c:122

0x400d25f7: obtain_time at /Users/jason/esp/esp-idf/examples/protocols/sntp/main/sntp_example_main.c:100

0x400d26c0: app_main at /Users/jason/esp/esp-idf/examples/protocols/sntp/main/sntp_example_main.c:69 (discriminator 9)

0x400d0d72: main_task at /Users/jason/esp/esp-idf/components/esp32/cpu_start.c:500

0x4008668d: vPortTaskWrapper at /Users/jason/esp/esp-idf/components/freertos/port.c:403
```

I am using the example at https://github.com/espressif/esp-idf/tree/release/v3.2/examples/protocols/sntp

The crash can be triggered simply by commenting out lines 96-99 which initializes WiFi.

I see in #944 that others have experienced this, and came to the conclusion that networking must be initialized, but this was not the case in ESP-IDF 3.1 or 3.0. It only happens in 3.2.

Alvin1Zhang, 2019-08-17:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @vonnieda

vonnieda, 2019-08-19:
Thanks @Alvin1Zhang, I have updated the issue with the template data.

closed 2020-02-18 by espressif-bot