espressif/esp-idf #3933: NimBLE: Can it be used with the v3.2.2 stable release?

bfriedkin, 2019-08-16:
Hello -

We've done a considerable amount of NimBLE testing using the feature/nimble_preview branch. NimBLE works quite well there and we would like to migrate to NimBLE in our projects. That said, for customer work we use the ESP-IDF v3.2.2 stable release branch, which we clone following the instructions provided in the Getting Started documentation:

```
cd ~/esp
git clone -b v3.2.2 --recursive https://github.com/espressif/esp-idf.git
```

Unfortunately for us, NimBLE is not available in this stable release. We did try a different method to download a v3.2.2 tag and found that NimBLE sources appear to be included:

```
cd ~/esp32
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout v3.2.2
git submodule update
```

Comparing the two trees, it seems that the v3.2.2 checkout is essentially identical to the branch checkout, but adds new modules, including NimBLE. Is there a "safe" way to integrate NimBLE with the stable release branch? We prefer to use a stable branch that won't be updated by Espressif.

Regards,
Brian

dhrishi, 2019-08-19:
@bfriedkin Ideally, neither release/v3.2 nor v3.2.2 tag should add NimBLE submodule (and hence sources) as this support has been added from 4.0 onwards. Can you please check if you see nimble folder as GIT controlled using  `git status`?

> That said, for customer work we use the ESP-IDF v3.2.2 stable release branch,

What are the timelines for your customer release? Also, are you going to move to the next stable 3.x version (when released)?

bfriedkin, 2019-08-19:
@dhrishi -

I tried checking out the v3.2.2 tag again into a new location using the following commands:

```
cd ~/esp32
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout v3.2.2
git submodule update --init
```

After the checkout I see nimble files in my ESP-IDF. For example:

```
Brians-MacBook-Air:src brianfriedkin$ pwd
/Users/brianfriedkin/esp32-nimble/esp-idf/components/bt/host/nimble/nimble/nimble/host/src
Brians-MacBook-Air:src brianfriedkin$ git log ble_gap.c
commit 936e538ffb14998326c45a32f2e5a32fa73b973b
Author: Hrishikesh Dhayagude <hrishi@espressif.com>
Date:   Tue Jun 4 22:16:02 2019 +0800

    NimBLE-component: Add support to disable/deinit NimBLE host and port
...
```

How does this method for checking out the v3.2.2 version differ from the recommended `git clone` of the v3.2.2 branch? I see the nimble files in one but not the other. I can try building our BLE apps against this v3.2.2 version later this week to see if things work as expected.

As for our timeline, we completed our nimble integration a few months ago using the `feature/nimble-preview` branch. We are ready now to migrate to an ESP-IDF that includes nimble in a stable release. Performance seems good and nimble requires less RAM, which is important to us. We just updated our customer projects to 3.2.2 from 3.1. Jumping to 4.0 would be challenging, but we had looked briefly at 3.3 before settling on the 3.2.2 stable release.

Regards,
Brian

dhrishi, 2019-08-19:
@bfriedkin Thanks for the overall feedback.

Shouldn't you be looking at:
`/Users/brianfriedkin/esp32/esp-idf/`
instead of
`/Users/brianfriedkin/esp32-nimble/esp-idf/`

You won't find NimBLE there :-)

I cannot assure you that NimBLE will be added in v3.2.2. Although, we will try to include it to a latest ESP-IDF stable release (something like 3.3.x). Again, this is not confirmed and I do not have concrete timelines for that.

bfriedkin, 2019-08-20:
@dhrishi -

Thank you for the follow. Sorry for the confusion. I created a `esp32-nimble` directory just to contain a fresh checkout of the v3.2.2 tag. When I build I rename that directory to `esp32`.

Getting back to my original question, I see NimBLE sources when I checkout the v3.2.2 tag but not the branch. I will try to build from the tag to see if it works. It looks like the 3.2.2 branch is a subset of the 3.2.2 tag. Because the 3.2.2 branch is a stable release and the 3.2.2 tag adds NimBLE, is it "safe" to assume that the tag is also stable but additionally includes the NimBLE component we require? Are the NimBLE sources different comparing what is included in the 3.2.2 tag to the 4.0 release(s)?

Good to know that NimBLE is targeted for the 4.x releases, but it could potentially be included in the next 3.3.x stable release. We would definitely appreciate that.

Regards,
Brian

dhrishi, 2019-08-20:
@bfriedkin As I mentioned before, 3.2.2 tag won't add NimBLE as it has been added after that. I tried the steps as below:
```
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout v3.2.2
git submodule update --init
```
After this, NimBLE sources will still be seen but they are not GIT controlled.

1. They will be seen as they were applicable when you first cloned the esp-idf (as by default it clones the master branch). `git status` will show `components/bt/host/` as untracked, `components/bt/host/nimble/nimble` being its part. 

2. `git log components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c` in esp-idf folder won't give you any output. Doing `git log` inside the NimBLE component (as you did above) won't be the right thing as the submodule (part of master branch) will still show you the GIT history but it would be untracked for the current tag.

3. You can manually remove `components/bt/host` folder and see `git diff`. There won't be any diff

bfriedkin, 2019-08-21:
@dhrishi -

Thank you for taking the time to provide additional details. Now it makes sense to me.

Regards,
Brian

closed 2019-08-21 by bfriedkin

dhrishi, 2019-08-21:
@bfriedkin 

>  but it could potentially be included in the next 3.3.x stable release. We would definitely appreciate that.

I will let you know if (and when) this happens :-)

dhrishi, 2019-10-25:
@bfriedkin We have ported NimBLE to v3.3. The release v3.3.1 should be out in a few weeks.
For the time being, you can use the latest `release/v3.3` branch to test it out.

bfriedkin, 2019-10-25:
@dhrishi -

Excellent. Thank you for porting NimBLE to v3.3. Just to confirm, the code would be available and GIT controlled if we check out the branch as follows?

```
git clone -b v3.3 --recursive https://github.com/espressif/esp-idf.git
```

Thanks,
Brian

dhrishi, 2019-10-25:
@bfriedkin Yes. That's right.

```
$  git clone -b release/v3.3 --recursive https://github.com/espressif/esp-idf.git
$ cd esp-idf
$ git submodule (You will see NimBLE submodule at components/nimble/nimble)
--
--
 8326807c5e580c3bb8ad13d1e80819aa0029004e components/nimble/nimble (remotes/origin/nimble-1.1.0-idf-v3.3)
--
--
```