espressif/esp-idf #3937: The CMAKE_C_COMPILER: xtensa-esp32-elf-gcc is not a full path and was not found in the PATH.

zexanana, 2019-08-18:
## Environment

- Development Kit:      ESP32-PICO-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-PICO-D4
- IDF version:  v4.0-dev-1443-g39f090a4f
- Build System:         CMake
- Compiler version: (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:  Linux - Ubuntu 18.04.03 LTS
- Power Supply:         USB

## Problem Description

I'm trying to compile a project for the first time. Followed all the steps in the Get Started guide. When i get to step "7 - Configure", I get the error message below. I am using the command `sudo python2 $(which idf.py) menuconfig` since the command `sudo idf.py menuconfig` returns command not found.

```
fulano@fulano:~/esp/hello_world$ sudo python2 $(which idf.py) menuconfig
Setting IDF_PATH environment variable: /home/fulano/esp/esp-idf
Checking Python dependencies...
Python requirements from /home/fulano/esp/esp-idf/requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory /home/fulano/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /home/fulano/esp/hello_world"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at /home/fulano/esp/esp-idf/tools/cmake/project.cmake:259 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

CMake Error at /home/fulano/esp/esp-idf/tools/cmake/project.cmake:259 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

CMake Error at /home/fulano/esp/esp-idf/tools/cmake/project.cmake:259 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:9 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/fulano/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "/home/fulano/esp/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```
Output of `printenv PATH`:
```
/home/fulano/esp/esp-idf/components/esptool_py/esptool:/home/fulano/esp/esp-idf/components/espcoredump:/home/fulano/esp/esp-idf/components/partition_table/:/home/fulano/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:/home/fulano/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/home/fulano/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin:/home/fulano/.espressif/python_env/idf4.0_py2.7_env/bin:/home/fulano/esp/esp-idf/tools:/home/fulano/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```
Installed CMake version:
`cmake version 3.10.2`

## What I tried

Tried adding the line
```
set(CMAKE_C_COMPILER /home/fulano/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin)
```
with multiple variations (path in quotes, add the compiler at the end of the path) to multiple files (`build.cmake`, `project.cmake` at /esp/esp-idf/tools/cmake, and to `CMakeLists.txt` at /esp/hello_world). Did not resolve the first error.

Tried running the command
`sudo python2 $(which idf.py) -DCMAKE_C_COMPILER:PATH=/home/fulano/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin menuconfig`
also with some variations, to no avail.

This seemed like an easy problem to fix but I have been scouring the internet for a solution and have not found one yet. Help is much appreciated.

projectgus, 2019-08-19:
Hi @zexanana ,

Probably running as `sudo` is resetting some of the paths, as the path is there for your user in the `printenv1 output.

Do you need to run `idf.py` as root (ie with `sudo`? Generally we would not recommend this.

Angus

projectgus, 2019-08-19:
(You can try using `sudo -E`  which tells sudo to preserve environment variables, including PATH, when running the target. But overall it's best to find a way to run `idf.py` without sudo at all, if you can.)

zexanana, 2019-08-19:
Thank you for the prompt response!! `sudo -E` did not work but I followed your suggestion. I was using sudo because CMake could not edit some of the files in the `/home/user/esp` folder. By giving all permissions to this and all subfolders (using the command below) I was able to run `python2 $(which idf.py) menuconfig` without sudo.

`sudo chmod -R 777 /home/user/esp`

Working like a charm. Thank you.

closed 2019-08-19 by zexanana

phypy, 2019-09-11:
Hi，I get this error on Windows 10,it's not help to  run python idf.py as a super manager  in windows power shell.

harshkr16, 2020-04-08:
I'm getting the same error and even after changing the mode as told by @zexanana , my issue persists.

projectgus, 2020-04-09:
@harshkr16 Are you also running idf.py with `sudo`? Can you give any other details about what is happening? What version of ESP-IDF do you have?

If you're not using `sudo` then please open a new issue and provide all the information that is asked for in the issue template, as it may be that you have the same error message but a different root cause.

mazorvanunu, 2020-09-09:
getting this error on windows 10, windows 7, ubuntu.
i am using an extension to visualstduio (visualgdb) which is at best - sketchy but at least can operate with it..
tried to came around to this via eclipse but could not config eclipse either.. why is this idf.py making so much problems? i cannot seem to make this work on any kind of platform i tried.

projectgus, 2020-09-14:
Hi @mazorvanunu,

Sorry to hear it's not working for you.

For problems that only occur with VisualGDB then I'm afraid you'll need to report them to the VisualGDB authors as we don't produce or support that plugin ourselves.

Are you using the [IDF Eclipse Plugin](https://github.com/espressif/idf-eclipse-plugin/)? Please open a new issue either on the ESP-IDF repo or on the Eclipse Plugin repo and supply all the requested details including versions, the full output when it tries to build, etc. Then we can help you debug this.

Angus

mazorvanunu, 2020-09-14:
Hi Angus, thank you for the fast reply
I got it working alright, feel much more comfortable after swimming through the ocean of configs...
One thing maybe others have faced on windows:

When you cannot seem to find idf.py, pull up a command line, and recall export.bat from the version (v4.1/4.2 etc) of the idf you want to subscribe to the *currently active* cmd window, then call idf.py from the project directory you wish to config.

trying to put idf.py directly into the path will cause various problems, one of which is recurring due to the fact that different environments of python and python_env (some kind of encapsulation for the esp-idf python environment) and then you might see missing packages you just know you already installed (but not in the right python-env since the PATH is all screwey after we forced our idf.py into it..

all n all i must say, idf is pretty great, even with all the hassle!
cheers

projectgus, 2020-09-14:
Hi @mazorvanunu,

I'm glad you got it working. Did you see some instructions in our current docs about editing the PATH? If so, please let me know and we'll fix it. You are right that `export.bat` and similar options set up quite a few options which aren't covered if you set the PATH only.

The Eclipse Plugin should handle this part, as well.

Angus

Kavish26, 2024-02-25:
hi, 
do one thing, make sure the xtensa compiler present in .espressif/tools/xtensa...whatever, is set as environment path in basrc, just chatgpt how to set it, its like, export PATH=$PATH:/path/to/directory.....

do the same for python3 under python_env under .espressif folder, 

and make sure IDF_PATH is exported, and like, one last path for tools folder present under esp-idf.

i had faced the same error, I am currently on windows, otherwise would have given you the commands. i don't know if someone is still facing the same issues or not, but I just solved this error like a few hours back, and make sure when you open vs code (hopefully you're working in it) (since I was), you scan for kits in the cmake extension, then select the xtensa one. that should make sure no more errors occur. make sure to have all 3 paths, and IDF_PATH set in user bashrc.

you wont face same issues again.