espressif/esp-idf #3938: after update.....

pilnikov, 2019-08-19:
Hallo all!
After updating the IDF, I got a few problems. Details below.

//////////////////////////////////////////////////////////////////////////////
Problem

In file included from Z:/msys32/home/pilnikov/examples/Ka-Radio32/components/audio_renderer/audio_renderer.c:17:
Z:/esp-idf/components/driver/include/driver/dac.h:43:53: error: unknown type name 'gpio_num_t'

//////////////////////////////////////////////////////////////////////////////
Solutions

I had to add

#include "driver/gpio.h"      
 

//////////////////////////////////////////////////////////////////////////////
Problem

Z:/esp-idf/components/driver/rtc_module.c: In function 'dac_pad_get_io_num':
Z:/esp-idf/components/driver/rtc_module.c:1773:21: error: 'DAC_CHANNEL_1_GPIO_NUM' undeclared (first use in this function); did you mean 'ADC1_CHANNEL_1_GPIO_NUM'?
         *gpio_num = DAC_CHANNEL_1_GPIO_NUM;
                     ^~~~~~~~~~~~~~~~~~~~~~
                     ADC1_CHANNEL_1_GPIO_NUM
Z:/esp-idf/components/driver/rtc_module.c:1773:21: note: each undeclared identifier is reported only once for each function it appears in
Z:/esp-idf/components/driver/rtc_module.c:1776:21: error: 'DAC_CHANNEL_2_GPIO_NUM' undeclared (first use in this function); did you mean 'ADC2_CHANNEL_2_GPIO_NUM'?
         *gpio_num = DAC_CHANNEL_2_GPIO_NUM;
                     ^~~~~~~~~~~~~~~~~~~~~~
                     ADC2_CHANNEL_2_GPIO_NUM

//////////////////////////////////////////////////////////////////////////////
Solutions

I had to add

#include "soc/dac_channel.h" 

  
//////////////////////////////////////////////////////////////////////////////
Prpoblem

Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/app_main.c: In function 'start_network':
Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/app_main.c:568:4: error: implicit declaration of function 'IPADDR2_COPY'; did you mean 'IPADDR_NONE'? [-Werror=implicit-function-declaration]
    IPADDR2_COPY(&gate,&ipAddr);
    ^~~~~~~~~~~~
    IPADDR_NONE

//////////////////////////////////////////////////////////////////////////////
Solutions

I had to add

#ifndef IPADDR2_COPY
#define IPADDR2_COPY(dest, src) SMEMCPY(dest, src, 4)
#endif

//////////////////////////////////////////////////////////////////////////////
Problem
  
In file included from Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/servers.c:14:
Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/include/websocket.h:18:10: fatal error: crypto/sha1_i.h: No such file or directory
 #include "crypto/sha1_i.h"
          ^~~~~~~~~~~~~~~~~
		  
//////////////////////////////////////////////////////////////////////////////
Solutions
?????
The file is located at  z:\esp-idf\components\wpa_supplicant\src\crypto\

Alvin1Zhang, 2019-08-19:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @pilnikov

pilnikov, 2019-08-19:
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc.exe (crosstool-NG esp32-2019r1) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ git describe --tags
v4.0-dev-1443-g39f090a4f

Build System:  Make

Operating System: Windows

projectgus, 2019-08-19:
Hi @pilnikov ,

Thanks for the report. We'll fix the two missing header includes in the driver headers, ASAP.

Regarding the other two issues:

> Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/app_main.c: In function 'start_network':
> Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/app_main.c:568:4: error: implicit declaration of > function [...]

This was an internal LWIP macro which has probably been removed when we upgraded to a newer LWIP version. You can change the calling code to something standard like `memcpy(&gate, &ipAddr, sizeof(ipAddr)` instead.

> In file included from Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/servers.c:14:
> Z:/msys32/home/pilnikov/examples/Ka-Radio32/main/include/websocket.h:18:10: fatal error: 
> crypto/sha1_i.h: No such file or directory
> #include "crypto/sha1_i.h"

This was an internal wpa_supplicant header that was never intended to be part of the public interface, so it's been removed in ESP-IDF V4.0.

Suggest refactoring the code to `#include "mbedtls/sha.h"` and call the mbedTLS APIs for SHA funcitonality, instead.

pilnikov, 2019-08-23:
after serial update ..... ((((👎 👎👎👎👎👎

$ make -j4 app
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from Z:/esp-idf\requirements.txt are satisfied.
GENCONFIG
Traceback (most recent call last):
  File "Z:/esp-idf/tools/kconfig_new/confgen.py", line 548, in <module>
    main()
  File "Z:/esp-idf/tools/kconfig_new/confgen.py", line 236, in main
    deprecated_options = DeprecatedOptions(config.config_prefix, path_rename_files=sdkconfig_renames)
  File "Z:/esp-idf/tools/kconfig_new/confgen.py", line 51, in __init__
    self.r_dic, self.rev_r_dic = self._parse_replacements(path_rename_files)
  File "Z:/esp-idf/tools/kconfig_new/confgen.py", line 67, in _parse_replacements
    with open(rep_path) as f_rep:
IOError: [Errno 2] No such file or directory: '/z/esp-idf/components/bootloader/sdkconfig.rename'

Such a situation with any code  in "make" build system (even with examples) the file sdkconfig.rename exists. Examples from "cmake" is working.

+++++++++++++

$ git submodule update --init --recursive
error: Server does not allow request for unadvertised object 7f8c86e501e690301630029fa9bae22424adf618
Fetched in submodule path 'components/coap/libcoap/ext/tinydtls', but it did not contain 7f8c86e501e690301630029fa9bae22424adf618. Direct fetching of that commit failed.
Failed to recurse into submodule path 'components/coap/libcoap'

pilnikov, 2019-08-23:
$ git describe --tags
v4.1-dev-58-g02c7c3885

projectgus, 2019-08-23:
Hi @pilnikov ,

Thanks for reporting these as well.

> Such a situation with any code in "make" build system (even with examples).

Are you using a "MINGW32" MSYS2 shell window, or something else? You can tell if it's the correct shell as the prompt should say "MINGW32".

If yes, what is the output of `which python` in your shell?

> Fetched in submodule path 'components/coap/libcoap/ext/tinydtls', but it did not contain
> 7f8c86e501e690301630029fa9bae22424adf618. Direct fetching of that commit failed.

This is a shortcoming in git, it doesn't handle things well when the URL of a submodule changes and the libcoap project has changed the URL of the tinydtl submodule (so it's a nested submodule inside a submodule of IDF.)

There are 3 options:

1. Ignore it if you're not using libcoap, as it probably won't matter.
2. Clean clone of ESP-IDF to a new working directory (sorry for the inconvenience).
3. Run these commands to fix it in place:
    > cd $IDF_PATH/components/coap/libcoap/ext/tinydtl
    > git remote set-url origin https://github.com/eclipse/tinydtls.git

Any of these 3 should work.

negativekelvin, 2019-08-23:
https://github.com/espressif/esp-idf/issues/3950

pilnikov, 2019-08-23:
If I understand correctly, I need to add a line

`COMPONENT_SDKCONFIG_RENAMES := $(shell cygpath -w (COMPONENT_SDKCONFIG_RENAMES))`

to the file \ esp-idf \ make \ project_config.mk after line No. 11?

pilnikov, 2019-08-23:
> #3950

++++
TNX!!! ALL WORK

closed 2019-09-08 by pilnikov