espressif/esp-idf #3941: CONFIG_BTU_TASK_STACK_SIZE no longer works

zacwbond, 2019-08-19:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.0-dev-1443-g39f090a4f
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux|]
- Power Supply:         [external 3.3V]

## Problem Description

In order to turn on Bluetooth logging, in the past I have had to increase the config option CONFIG_BTU_TASK_STACK_SIZE.   It looks like this config option is no longer used in recent versions of the code and the BTU stack size has been hard-coded to 4096.  I think this options should be added again; without it, I receive 

```***ERROR*** A stack overflow in task btuT has been detected.```

whenever my BLE peripheral connects.  Also, it seems like all the other tasks have configurable stack sizes, and this option is even mentioned in documentation. I think its removal was probably an accident.

### Expected Behavior

CONFIG_BTU_TASK_STACK_SIZE changes the btuT stack size.

### Actual Behavior

CONFIG_BTU_TASK_STACK_SIZE doesn't change the btuT stack size.

### Steps to reproduce

1. Change CONFIG_BTU_TASK_STACK_SIZE.
2. Rebuild.
3. Discover that the btuT stack size is unchanged.

negativekelvin, 2019-08-19:
Relevant commit
https://github.com/espressif/esp-idf/commit/faf23df19ac936710b12d36d56683d0bf0468189

blueMoodBHD, 2019-09-16:
Hi, This bug have been fixed, refer to commit 07bd6d4

Alvin1Zhang, 2019-09-16:
@zacwbond Hi, would you please help try the fix commit and check whether your issue has been fixed? Thanks.

zacwbond, 2019-09-17:
Looks like this has been resolved!  Thanks.

closed 2019-09-17 by zacwbond