espressif/esp-idf #3942: wpa_supplicant looses connection for a short time with Soft-AP when ESP32 is connecting to an access point

farizadp, 2019-08-20:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-PICO-D4
- IDF version: v3.2.2
- Build System:         Make
- Compiler version :    crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         external 3.3V

## Problem Description
### Hardware setup:
<ol>
<li>ESP32 (MAC 50:4d:35(ap), 50:4d:34(sta))</li>
It runs Soft-AP application. It hosts HTTP server on the AP interface
<li>PC (MAC e4:8b:21)</li>
It runs wpa_supplicant to connect to ESP32 AP. When it's connected to the ESP32 AP, it instructs the ESP32 to connect to Router (through HTTP request)
<li>Router (MAC 2c:9c:7c)</li>
It provides broad internet. Encryption type is WPA2.
</ol>

### Problem:
When ESP32 is making connection to the Router, the ```wpa_supplicant``` catches disconnect event. After few moments ```wpa_supplicant``` is connected again to the ESP32 access point
```
wlp2s0: CTRL-EVENT-DISCONNECTED bssid=00:0f:11:50:4d:35 reason=0
wlp2s0: Trying to associate with 00:0f:11:50:4d:35 (SSID='P1Dongle-4D34' freq=2422 MHz)
Failed to add supported operating classes IE
wlp2s0: Associated with 00:0f:11:50:4d:35
wlp2s0: WPA: Key negotiation completed with 00:0f:11:50:4d:35 [PTK=CCMP GTK=TKIP]
wlp2s0: CTRL-EVENT-CONNECTED - Connection to 00:0f:11:50:4d:35 completed [id=0 id_str=]
```
However, there is no disconnect event at ESP32 side (i.e. ```EVENT_AP_STADISCONNECTED```) but there is another ```SYSTEM_EVENT_AP_STACONNECTED``` event at ESP32 side.

Below are functions which were called when ESP32 is connecting to Router:
```
esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_sta_config);
// set Router's ssid and password
esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_sta_config);
esp_wifi_connect();
```

Attached is the air traffic logging, when the ESP32 is making connection to the Router.
[disconnect_sniff.zip](https://github.com/espressif/esp-idf/files/3521146/disconnect_sniff.zip)

Alvin1Zhang, 2019-08-21:
@farizadp Thanks for reporting the issue, we will look into this and provide feedback later. Thanks.

negativekelvin, 2019-08-21:
If esp32 has to change wifi channel to connect to the router then station will have to reconnect to esp32 ap on new channel.  Most likely no disconnection event because station wasn't disconnected long enough to trigger timeout.

sagb2015, 2019-08-21:
@farizadp - What @negativekelvin has said is correct.  The disconnection is because of change in the channel. What confused me earlier was the capture. You changing channel from 1 to 3 and these are not orthogonal channels. So you could see some packets from channel 3 on 1. I thought you have the same channels set for both.

farizadp, 2019-08-21:
But I don't configure the ```channel``` member neither in ```wifi_ap_config_t``` struct nor in ```wifi_sta_config_t```.

For the ```wifi_ap_config_t```, I memset it and configure the ```ap.ssid```, ```ap.password```, ```ap.authmode```, ```ap.max_connection```.
For the ```wifi_sta_config_t```, I call ```esp_wifi_get_config``` and then configure only the ```sta.ssid``` and ```sta.password```

Now my question is, how does ESP32 select the channel for its AP at beginning? and then how ESP32 chooses/switches the channel when connecting to router?

negativekelvin, 2019-08-21:
Well it probably chooses channel 1 by default and then when connecting to a router it will do a scan to find which channel the router is on and switch to it.

farizadp, 2019-08-21:
@negativekelvin : Alright, so this kind of behavior will always happen in our application?

negativekelvin, 2019-08-21:
Yes, if you want you could do a scan first and set the channel based on most likely router (highest RSSI) which may minimize the chances of it happening.

farizadp, 2019-08-21:
Perhaps it's good to mention this behavior somewhere in the esp-idf documentation.

To me, it means that the AP and the STA uses the same channel number although the ```wifi_sta_config_t``` ```wifi_ap_config_t``` each has member ```channel```

sagb2015, 2019-08-21:
> @negativekelvin @sagb2015
> But I don't configure the `channel` member neither in `wifi_ap_config_t` struct nor in `wifi_sta_config_t`.
> 
> For the `wifi_ap_config_t`, I memset it and configure the `ap.ssid`, `ap.password`, `ap.authmode`, `ap.max_connection`.
> For the `wifi_sta_config_t`, I call `esp_wifi_get_config` and then configure only the `sta.ssid` and `sta.password`
> 
> Now my question is, how does ESP32 select the channel for its AP at beginning? and then how ESP32 chooses/switches the channel when connecting to router?

The channel is set by default to start_channel of the regulatory domain (country). Other channels can be set by populating channel element in wifi_ap_config_t. However, after STA connects the channel will be changed to exAP channel by design.  To avoid the channel change, you can set SoftAP channel to required beforehand as suggested by @negativekelvin. You can use bssid field in wifi_sta_config_t to tell which AP to connect to, if there are multiple APs with same ESSID.

farizadp, 2019-08-21:
> 
> 
> Perhaps it's good to mention this behavior somewhere in the esp-idf documentation.
> 
> To me, it means that the AP and the STA uses the same channel number whn all are connected at the same time although the `wifi_sta_config_t` `wifi_ap_config_t` each has member `channel`

sagb2015, 2019-08-21:
> Perhaps it's good to mention this behavior somewhere in the esp-idf documentation.
> 
> To me, it means that the AP and the STA uses the same channel number whn all are connected at the same time although the `wifi_sta_config_t` `wifi_ap_config_t` each has member `channel`
It is mentioned here:- 
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-mode

farizadp, 2019-08-21:
@sagb2015 : I see, then I've probably missed this tiny detail.

closed 2019-08-21 by sagb2015

sagb2015, 2019-08-21:
We will make this more explicit going forward. For 3.2.2, it is mentioned in more details [here](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/wifi.html#home-channel)

farizadp, 2019-08-21:
> 
> 
> We will make this more explicit going forward. For 3.2.2, it is mentioned in more details [here](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/wifi.html#home-channel)

Maybe also mention that device connected to ESP32's AP in Soft-AP might detect disconnection. Also put a suggestion how to minimize the chance of this behavior from happening.

sagb2015, 2019-08-21:
Noted. Will create a different tracker for this as I noticed some inconsistencies across the version in documentation.