espressif/esp-idf #3944: "BT_HCI: command_timed_out hci layer timeout waiting for response" message on attempt to disconnect

zacwbond, 2019-08-20:
## Environment
- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-dev-1443-g39f090a4f
- Build System:         [CMake]
- Operating System:     [Ubuntu Linux docker image]
- Power Supply:         [external 3.3V]

## Problem Description

We have an ESP32-WROOM-32D on custom hardware, and it operates as a central.  We also have a BLE peripheral device running on an NRF52832.  The central and peripheral are able to connect and interact without a problem.

The problem arises when I want the ESP32 to force disconnect from the peripheral.  The goal is to both a) disconnect from the peripheral and b) remove any bonds associated with that peripheral.  The sequence of events I follow is:

1. Call `esp_ble_gattc_close()` for the correct server and connection ID.
2. Wait to receive`ESP_GATTC_DISCONNECT_EVT`.
3. Verify that the disconnect reason is `0x16`; if it is, then call  `esp_ble_remove_bond_device(params->disconnect.remote_bda)`

This sequence works fine about half the time.  The other half of the time, the BLE stack seems to get stuck in some kind of weird, half-connected state.  I have provided some annotated logs below.  

I have also tried some other sequences (such as calling `esp_ble_remove_bond_device()` immediately after `esp_ble_gattc_close()` without waiting for `ESP_GATTC_DISCONNECT_EVT.`  I get the same results whatever I try.

I should also note the (quite slow) connection parameters we are using, which could play a role in the problem:

```min_int = 60
max_int = 120
conn_int = 120
latency = 90
timeout = 3000
```

### Expected Behavior

I would expect that we would disconnect with reason 0x16 after `esp_ble_remove_bond_device()`, and that after the next connection interval (around 13 seconds with the parameters above) the other BLE device will notice the disconnect with a reason code of 0x13. 

About half the time, that's exactly what happens.

### Actual Behavior

All of the function calls and events return without any error codes, but I receive the following error in the log:
```BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x406```

Opcode 0x406 is apparently the opcode for "disconnect."

Afterward, if I try to use the external device, it seems to think it is still connected to the central and I see the following message multiple times:

```BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0```

### Steps to reproduce

1. Set up the ESP32 as a central and connect to it with a peripheral, then request the connection parameters specified above.
2. Call esp_ble_gattc_close() to disconnect from the peripheral.
3. If it succeeds, reconnect and try again until you experience the failure.  For me, it's a failure rate of about 50%.

## Debug Logs

I have included some comments in this log.  BT_GATT and BT_L2CAP debug-level logging is enabled.  Let me know if other debug would be useful.

```
# We receive the command to remove bonding and first call esp_ble_gattc_close()
E (226567) Network: Got remove bonding command
E (226568) Ble Driver: Remove device 36:AD:F5:78:30:CC
I (226570) BT_GATT: GATT_Disconnect conn_id=3
D (226570) BT_GATT: gatt_update_app_use_link_flag  is_add=0 chk_link=0
ZWB esp_ble_gattc_close: 0
D (226570) BT_GATT: gatt_update_app_hold_link_status found=1[1-found] idx=0 gatt_if=3 is_add=0
D (226572) BT_GATT: gatt_num_apps_hold_link   num=0
D (226572) BT_GATT: gatt_disconnect
D (226573) BT_GATT: gatt_get_ch_state: ch_state=4
I (226573) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0004  BDA: cc3078f5ad36
D (226574) BT_L2CAP: l2cu_release_ccb: cid 0x0004  in_use: 1
D (226575) BT_L2CAP: l2cu_no_dynamic_ccbs() IDLE timer 0, disconnecting link
D (226576) BT_GATT: GATT   ATT protocol channel with BDA: cc3078f5ad36 is disconnected
D (226577) BT_GATT: gatt_is_bda_in_the_srv_chg_clt_list :cc-30-78-f5-ad-36
D (226578) BT_GATT: bda is in the srv chg clt list
D (226578) BT_GATT: gatt_cleanup_upon_disc
D (226579) BT_GATT: found p_tcb
D (226579) BT_GATT: gatt_set_ch_state: old=4 new=0
D (226580) BT_GATT: gatt_free_pending_ind
D (226580) BT_GATT: gatt_free_pending_enc_queue
D (226581) BT_GATT: gatt_free_pending_prepare_write_queue
D (226582) BT_GATT: found p_reg tcb_idx=0 gatt_if=1  conn_id=0x1
D (226582) BT_GATT: gatt_connect_cback: from cc3078f5ad36 connected:0 conn_id=1 reason = 0x0016
D (226583) BT_GATT: gatt_get_ch_state: ch_state=0
I (226584) BT_GATT: GATT_GetConnIdIfConnected status=0
D (226585) BT_GATT: found p_reg tcb_idx=0 gatt_if=2  conn_id=0x2
D (226586) BT_GATT: found p_reg tcb_idx=0 gatt_if=3  conn_id=0x3
W (226587) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
D (226587) BT_GATT: found p_reg tcb_idx=0 gatt_if=4  conn_id=0x4
D (226588) BT_GATT: gatt_delete_dev_from_srv_chg_clt_list
D (226589) BT_GATT: gatt_is_bda_in_the_srv_chg_clt_list :cc-30-78-f5-ad-36
D (226590) BT_GATT: bda is in the srv chg clt list
D (226590) BT_GATT: ATT disconnected
D (226591) BT_L2CAP: l2cu_no_dynamic_ccbs() starting IDLE timeout: 30
E (226592) Esp Ble Client: gatt server event: ESP_GATTC_CLOSE_EVT on interface: 3
E (226593) Esp Ble Client: gatt server event: ESP_GATTC_DISCONNECT_EVT on interface: 3
E (226594) Esp Ble Client: remote disconnected!

# At this point, or callback has received ESP_GATTC_DISCONNECT_EVT 
# In response, we call esp_ble_remove_bond_device(), and which returns success.
# We also restart scanning to potentially connect with a new device (there are none nearby in this test, so none try to connect).

ZWB esp_ble_remove_bond_device: 0
E (226595) Esp Ble Client: Restart scanning due to disconnect
E (226595) Esp Ble Client: Start scanning!
E (226605) Ble Driver: GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
E (226605) Ble Driver: GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT

# I don't know what timed out here--was it the internal BT stack, or the external device?  
# 8 seconds seems too early for the device to have responded given the connection parameters.

E (234605) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x406
D (256592) BT_L2CAP: L2CAP - l2c_link_timeout() link state 5 first CCB 0 is_bonding:0
I (256593) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
E (256606) Ble Driver: GAP event: ESP_GAP_BLE_REMOVE_BOND_DEV_COMPLETE_EVT

# The same error message again.

E (264596) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x406

# At this point, I try to send some notifies from the peripheral to the central.
# It looks like the connection ID is now invalid so L2CAP doesn't know what to do with it.

E (281096) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (281696) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (282146) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (282746) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (283196) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (283796) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (284096) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (287547) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (288146) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (288596) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (289196) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
E (289646) BT_L2CAP: L2CAP - rcvd ACL for unknown handle:0 ls:0 cid:4 opcode:27 cur count:0
W (305930) BT_BTM: btm_find_dev_by_identity_addr find pseudo->random match with diff addr type: 0 vs 1

# We continue in this state until I power-cycle the central.
```

Alvin1Zhang, 2019-08-21:
@zacwbond Thanks for reporting the issue, we will look into this. Thanks.

albertmbt, 2019-09-16:
I have this same issue. And also, after a disconnection, I'm not able to do a new scan. It's just that ESP_GAP_BLE_SCAN_RESULT_EVT is not receiving interrupts any more. I have to reset the chip to solve it.

gengyuchao, 2019-09-26:
Hello, have you added any functions which take much time in esp_gap_cb ?

Alvin1Zhang, 2019-10-11:
@zacwbond Thanks for reporting, would you please help share if any more details could be shared for the questions by @gengyuchao ? Thanks.

zacwbond, 2019-10-11:
@gengyuchao @Alvin1Zhang 

Sorry for my slow reply.  How long is too much time?  I can annotate the log I provided to show the GATTC and GAP callback events that we received and what our code does in response.

```
# We receive the command to remove bonding and first call esp_ble_gattc_close()
E (226567) Network: Got remove bonding command

E (226592) Esp Ble Client: gatt server event: ESP_GATTC_CLOSE_EVT on interface: 3
# This event is ignored.

E (226593) Esp Ble Client: gatt server event: ESP_GATTC_DISCONNECT_EVT on interface: 3
# In response to this event, we do the following:
# 1. Delete an entry in a table of connected devices.  Right now we only support one.
# 2. Create a message to send to the host device indicating that a device was removed and post that message to a FreeRTOS queue.  
# 3. Log the following message:
E (226594) Esp Ble Client: remote disconnected!
# 4. Call esp_ble_remove_bond_device() for the device in question.
# 5. Call esp_ble_gap_start_scanning()
# 6. Print the following messages
E (226595) Esp Ble Client: Restart scanning due to disconnect
E (226595) Esp Ble Client: Start scanning!

E (226605) Ble Driver: GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
# This event is ignored.  It was probably triggered by our host micro sending a command.

E (226605) Ble Driver: GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
# This event is ignored.

# Here's the error, which happened 8 seconds after the last GAP event that we saw.
E (234605) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x406

E (256606) Ble Driver: GAP event: ESP_GAP_BLE_REMOVE_BOND_DEV_COMPLETE_EVT
# This event is ignored.

# The same error message again, another 30s later
E (264596) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x406

# We continue in this state until I power-cycle the central.
```

Hope this helps.

lucastcox, 2020-01-06:
@zacwbond , did you ever get this fixed? I meant to reply a long time ago, but completely forgot. Since leaving baf, I've learned that one potential cause of issues can be the log statements in the ble callbacks. The esp32 uart tx buffer is 128 bytes. 

In a period of 2 milliseconds, the following message are logged: 

```
E (226593) Esp Ble Client: gatt server event: ESP_GATTC_DISCONNECT_EVT on interface: 3
E (226594) Esp Ble Client: remote disconnected!
E (226595) Esp Ble Client: Restart scanning due to disconnect
E (226595) Esp Ble Client: Start scanning!
```
Unless I misunderstood, or something has changed esp_log statements are blocking. If there's enough room in the fifo, it will be "instant," but if there isn't, it'll poll until the entire message is sent. 

The total character count of those messages is 236. Not sure that this is the issue, but if the baud rate is 115200, that means the ble stack has the potential to be blocked for (1s/115200baud)*10baud/char * (236-128 chars) = 9.4ms.

Alvin1Zhang, 2020-01-08:
@zacwbond @lucastcox Thanks for the updates. We would look into and get back to you if any updates. Thanks.

Pragalf, 2022-10-11:
> @zacwbond , did you ever get this fixed? I meant to reply a long time ago, but completely forgot. Since leaving baf, I've learned that one potential cause of issues can be "too many" log statements in the ble callbacks. The esp32 uart tx buffer is 128 bytes.
> 
> In a period of 2 milliseconds, the following messages are logged:
> 
> ```
> E (226593) Esp Ble Client: gatt server event: ESP_GATTC_DISCONNECT_EVT on interface: 3
> E (226594) Esp Ble Client: remote disconnected!
> E (226595) Esp Ble Client: Restart scanning due to disconnect
> E (226595) Esp Ble Client: Start scanning!
> ```
> 
> Unless I misunderstood, or something has changed, esp_log statements are blocking. If there's enough room in the fifo, it will be "instant," but if there isn't, it'll poll until the entire message is sent.
> 
> The total character count of those messages is 236. Not sure that this is the issue, but if the baud rate is 115200, that means the ble stack has the potential to be blocked for (1s/115200baud)*10baud/char * (236-128 chars) = 9.4ms.
> 
> @gengyuchao @Alvin1Zhang, what do you think?

Hai,
 I am also having a similar issue with Bluetooth. After sending data through BT for sometime, BT will be disconnected and am not able to connect to device. Strange enough, Bluetooth is not shown in available devices when checked with a phone. 

`W (4526463) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 800, status 0x0
I (12:26:26.063) SPP_ACCEPTOR_DEMO: ESP_BT_GAP_MODE_CHG_EVT mode:2 bda:[94:3c:c6:c1:72:ba]
loop:58380
loop:58380
W (4534743) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x8
W (4534743) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:1 state 2 closed: Peer connection failed (res: 16)
I (12:26:34.347) SPP_ACCEPTOR_DEMO: ESP_SPP_CLOSE_EVT
I (12:26:34.348) SPP_ACCEPTOR_DEMO: BT_connected:0
loop:62240
loop:62372
W (4550633) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (4550633) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x8
E (4550643) BT_HCI: CC evt: op=0x80d, status=0x2
E (4550643) BT_HCI: Ctlr H/w error event - code:0x0

loop:62176
loop:62176
E (4558673) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x80d
loop:62308
loop:62176
E (4575373) BT_BTC: btc_spp_write unable to find RFCOMM slot!
I (12:27:14.979) SPP_ACCEPTOR_DEMO: ESP_SPP_WRITE_EVT`

These is the last events occurred. What would be the problem?
I am using IDF 4.4.2.

boblane1, 2023-04-12:
Hi @Pragalf, 

"BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x80d"  shows the Bluetooth Host and Controller is out of sync, since a hardware error has occurred in the Controller("BT_HCI: Ctlr H/w error event - code:0x0"). Can you always reproduce the issue? Would you mind opening a new issue to describe it in detail? Thanks.