espressif/esp-idf #3946: Secure boot continuous reboot issue - Importance of MTDI pin for secure boot

AjiGopal, 2019-08-20:
Hi all

While enabling Secure boot in re-flashable mode it showsthe below error.
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8664
load:0x40078000,len:17352
load:0x40080400,len:6700
csum err:0x9d!=0xdf
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8664
load:0x40078000,len:17352
load:0x40080400,len:6700
csum err:0x9d!=0xdf
ets_main.c 371 

Here I am using ESP WROVER B board for the development in LINUX OS With ESP IDF.
I am following the below mentioned links 
https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#bootloader-size
https://docs.pycom.io/advance/encryption/

I have worked as per the instructions given in the above mentioned links,
The key is successfully generated and burned to the module but while uploading the app and the partition  table it shows  the above mentioned reboot issue.

While referring some other links one of them says that the GPIO12 (MTDI) pin has some effect in this process and it is in floating state it shows this kind of an error. Does this  statement has any validity.

Please respond as soon as possible.

Regards
Aji Gopal

negativekelvin, 2019-08-20:
https://github.com/espressif/esp-idf/issues/3212

Alvin1Zhang, 2019-08-23:
@negativekelvin Thanks for the information, @AjiGopal would you please help check whether your issue is same/similar with [3212](https://github.com/espressif/esp-idf/issues/3212), and has your issue been resolved? Thanks.

AjiGopal, 2019-08-23:
Thanks for the replays. 
Its fixed now

closed 2019-08-23 by AjiGopal