espressif/esp-idf #3949: [BLE Mesh] How can I get some status of node?

zhanlej, 2019-08-21:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): [v4.0-dev-301-gc7c419da4]
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): [1.22.0-80-g6c4433a]
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

我如何才能获取到node的一些状态，比如：

- 手机给node添加key的时候
- 手机给vendor model绑定app key的时候
- 手机给vendor model订阅group address的时候
- 等等

实际上我已经在``ble_mesh_fast_prov_server``这个demo中找到了``esp_ble_mesh_register_config_server_callback()``这个函数去注册config server的回调函数，并且在回调函数中也发现了``ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD``这种情况的处理，但是当我想要扩展的时候却发现了一些问题：

1. 当我将``esp_ble_mesh_register_config_server_callback()``这个函数移植到``ble_mesh_node``这个demo中后发现回调函数并没有被调用，我对比了两个demo的``sdkconfig.defaults``文件，发现主要区别在于有没有``CONFIG_BLE_MESH_FAST_PROV=y``这个选项，然后再``ble_mesh_node``中添加了这个选项后config server的回调函数才能按照预期的开始工作。为什么一定要添加``CONFIG_BLE_MESH_FAST_PROV=y``这个选项？
2. 但是当我参照``esp_ble_mesh_defs.h``里面定义的关于config model的宏，添加了很多代码想要实现更多的提示后发现除了``ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD``之外并没有任何新的提示，是你们的代码还没有实现其他的功能么，还是我想要实现的功能并不是通过这种方式实现而另外有其他的函数可以实现？代码如下：
```c
static void example_ble_mesh_config_server_cb(esp_ble_mesh_cfg_server_cb_event_t event,
        esp_ble_mesh_cfg_server_cb_param_t *param)
{
    esp_err_t err;

    ESP_LOGI(TAG, "%s, event = 0x%02x, opcode = 0x%04x, addr: 0x%04x",
             __func__, event, param->ctx.recv_op, param->ctx.addr);

    switch (event) {
    case ESP_BLE_MESH_CFG_SERVER_RECV_MSG_EVT:
        switch (param->ctx.recv_op) {
        case ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD app_idx:0x%04x", param->status_cb.app_key_add.app_idx);
            break;
        case ESP_BLE_MESH_MODEL_OP_MODEL_SUB_ADD:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_MODEL_SUB_ADD");
            break;
        case ESP_BLE_MESH_MODEL_OP_MODEL_SUB_DELETE:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_MODEL_SUB_DELETE");
            break;
        case ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND");
            break;
        case ESP_BLE_MESH_MODEL_OP_NODE_RESET:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_NODE_RESET");
            break;
        case ESP_BLE_MESH_MODEL_OP_NET_KEY_UPDATE:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_NET_KEY_UPDATE");
            break;
        case ESP_BLE_MESH_MODEL_OP_NET_KEY_DELETE:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_NET_KEY_DELETE");
            break;
        case ESP_BLE_MESH_MODEL_OP_APP_KEY_UPDATE:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_APP_KEY_UPDATE");
            break;
        case ESP_BLE_MESH_MODEL_OP_APP_KEY_DELETE:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_APP_KEY_DELETE");
            break;
        default:
            break;
        }
        break;
    default:
        return;
    }
}
```

WCCWCC, 2019-08-21:
Hi zhanlej ,
   
    `example_ble_mesh_config_server_cb` 负责处理 节点( config server model ) 配置相关的回调函数。
    比如： ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD 事件和 Config AppKey Add 消息对应。如果事件没有触发则表示没有收到该消息。该事件表示 provisioner 给 node 添加 Appkey。

    如果你想要通过 provisioner 发送该消息，消息的 opcode 是   `ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD`

``` c
        cfg_set.app_key_add.net_idx
        cfg_set.app_key_add.app_idx
        
        err = esp_ble_mesh_config_client_set_state(&msg_common, &cfg_set);
        if (err) {
            ESP_LOGE(TAG, "Failed to send Config AppKey Add");
            break;
        }
```

Alvin1Zhang, 2019-08-22:
@zhanlej Thanks for reporting the this. For "how" questions, you could raise a post in our forum at https://esp32.com, thanks.

closed 2019-08-22 by Alvin1Zhang

zhanlej, 2019-08-22:
Hi, WCCWCC,

```
`example_ble_mesh_config_server_cb` 大部分事件都是底层自动完成的，没有通知到应用层。操作成功与不成功是通过 provisioner 端判断的。
```

那为什么`example_ble_mesh_config_server_cb` 中能响应`ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD`这个事件，而其他事件不能被响应，是因为你们对app_key_add事件的底层代码有修改么？

我现在想实现设备知道app key bind事件应该如何实现？

```
`CONFIG_BLE_MESH_FAST_PROV=y` 这个是为了快速配网特性做的特殊处理。
```

现在的问题是当我在ble_mesh_node demo中实现example_ble_mesh_config_server_cb后发现该回调函数并没有执行，我必须在sdkconfig中添加`CONFIG_BLE_MESH_FAST_PROV=y`这个选项之后config server的回调函数才能正常工作，这是为什么？

```
补充： 如果你想要通过 provisioner 发送该消息，消息的 opcode 是   `ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD`
```

我使用nRF Mesh APP作为provisioner控制设备，已经能在设备的打印中看到`ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD`事件的处理消息。

WCCWCC, 2019-08-22:
Hi zhanlej,

     节点在收到 Config AppKey Add 消息后，会调用 cfg_srv.c 文件中的 app_key_add 函数，该函数负责处理该消息。  请注意 函数中条件编译 CONFIG_BLE_MESH_FAST_PROV，如果这个宏打开，该事件就会通知应用层，如果没有打开就不会通知到应用层。

    CONFIG_BLE_MESH_FAST_PROV 这个配置项是为 快速配网 feature 而实现的，如果你不使能该配置项且需要收到该事件，把条件编译部分注释掉就可以了。

zhanlej, 2019-08-22:
Hi WCCWCC,

简单看了一下你们的源码，发现目前仅对app_key_add事件做了通知应用层的处理，对config server其他事件通知应用层的功能是否有规划？

WCCWCC, 2019-08-22:
Hi zhanlej，

    目前对于节点配网相关的消息，底层自动处理的，没有通知应用层的事件。 你暂时只能通过 provisioner 端来获取节点的配置状态。

zhanlej, 2019-08-22:
@WCCWCC  谢谢