espressif/esp-idf #3950: Latest confgen.py fails on Windows/msys32

negativekelvin, 2019-08-21:
## Environment

- IDF version (run ``git describe --tags`` to find it): v4.1-dev-59-g147417251
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp32-2019r1
- Operating System:     Windows/msys32

## Problem Description

Build fails using make in msys32 environment

Reverting https://github.com/espressif/esp-idf/commit/bd219609556b8ee016d2ae27c25c9fb2acb92adf#diff-3f54e34000271b834b839d614f008208 and https://github.com/espressif/esp-idf/commit/423fb9573bf6c7516536b7c14cc488312166550d#diff-3f54e34000271b834b839d614f008208 allows it to work again

### Code to reproduce this issue

station example

## Debug Logs
```
Python requirements from C:/msys32/home/dev/esp32/esp-idf/requirements.txt are satisfied.
GENCONFIG
Traceback (most recent call last):
  File "C:/msys32/home/dev/esp32/esp-idf/tools/kconfig_new/confgen.py", line 548, in <module>
    main()
  File "C:/msys32/home/dev/esp32/esp-idf/tools/kconfig_new/confgen.py", line 236, in main
    deprecated_options = DeprecatedOptions(config.config_prefix, path_rename_files=sdkconfig_renames)
  File "C:/msys32/home/dev/esp32/esp-idf/tools/kconfig_new/confgen.py", line 51, in __init__
    self.r_dic, self.rev_r_dic = self._parse_replacements(path_rename_files)
  File "C:/msys32/home/dev/esp32/esp-idf/tools/kconfig_new/confgen.py", line 67, in _parse_replacements
    with open(rep_path) as f_rep:
IOError: [Errno 2] No such file or directory: '/home/dev/esp32/esp-idf/components/bootloader/sdkconfig.rename'
```

ROYMURLI, 2019-08-21:
I am also facing same issue with IDF version - v4.1-dev-58-g02c7c3885
OS- Windows 10 (msys32)
xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

Exactly same log debug logs I got.

dobairoland, 2019-08-21:
I'm sorry for the inconvenience and thanks for reporting this.

I believe that adding 
```COMPONENT_SDKCONFIG_RENAMES := $(shell cygpath -w $(COMPONENT_SDKCONFIG_RENAMES))```
after line https://github.com/espressif/esp-idf/blob/master/make/project_config.mk#L11 should fix the problem.

patrykkrz, 2019-08-22:
I have the same bug.
@dobairoland your patch is good, it works.

dobairoland, 2019-08-22:
Thanks @patrykkrz for the feedback.

ROYMURLI, 2019-08-22:
Thanks @dobairoland . It works.

ROYMURLI, 2019-08-24:
Still facing some issue while compiling. It works fine with IDF 3.3 but when I am trying with present 4.1 with same sdkconfig and above patch it solved my above mentioned issue but few more errors are:-

D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c: In function 'esp_secure_boot_verify_signature_block':
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:96:5: error: unknown type name 'mbedtls_ecdsa_context'
     mbedtls_ecdsa_context ecdsa_context;
     ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:97:5: error: implicit declaration of function 'mbedtls_ecdsa_init' [-Werror=implicit-function-declaration]
     mbedtls_ecdsa_init(&ecdsa_context);
     ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:5: error: implicit declaration of function 'mbedtls_ecp_group_load' [-Werror=implicit-function-declaration]
     mbedtls_ecp_group_load(&ecdsa_context.grp, MBEDTLS_ECP_DP_SECP256R1);
     ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:42: error: request for member 'grp' in something not a structure or union
     mbedtls_ecp_group_load(&ecdsa_context.grp, MBEDTLS_ECP_DP_SECP256R1);
                                          ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:48: error: 'MBEDTLS_ECP_DP_SECP256R1' undeclared (first use in this function)
     mbedtls_ecp_group_load(&ecdsa_context.grp, MBEDTLS_ECP_DP_SECP256R1);
                                                ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:48: note: each undeclared identifier is reported only once for each function it appears in
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:100:50: error: request for member 'grp' in something not a structure or union
     size_t plen = mbedtls_mpi_size(&ecdsa_context.grp.P);
                                                  ^
In file included from D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/asn1.h:36:0,
                 from D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/x509.h:33,
                 from D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:22:
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:108:59: error: request for member 'Q' in something not a structure or union
     MBEDTLS_MPI_CHK(mbedtls_mpi_read_binary(&ecdsa_context.Q.X, signature_verification_key_start, plen));
                                                           ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/bignum.h:52:22: note: in definition of macro 'MBEDTLS_MPI_CHK'
         if( ( ret = (f) ) != 0 ) \
                      ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:109:59: error: request for member 'Q' in something not a structure or union
     MBEDTLS_MPI_CHK(mbedtls_mpi_read_binary(&ecdsa_context.Q.Y, signature_verification_key_start + plen, plen));
                                                           ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/bignum.h:52:22: note: in definition of macro 'MBEDTLS_MPI_CHK'
         if( ( ret = (f) ) != 0 ) \
                      ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:110:52: error: request for member 'Q' in something not a structure or union
     MBEDTLS_MPI_CHK(mbedtls_mpi_lset(&ecdsa_context.Q.Z, 1));
                                                    ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/bignum.h:52:22: note: in definition of macro 'MBEDTLS_MPI_CHK'
         if( ( ret = (f) ) != 0 ) \
                      ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:112:11: error: implicit declaration of function 'mbedtls_ecdsa_verify' [-Werror=implicit-function-declaration]
     ret = mbedtls_ecdsa_verify(&ecdsa_context.grp, image_digest, DIGEST_LEN, &ecdsa_context.Q, &r, &s);
           ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:112:46: error: request for member 'grp' in something not a structure or union
     ret = mbedtls_ecdsa_verify(&ecdsa_context.grp, image_digest, DIGEST_LEN, &ecdsa_context.Q, &r, &s);
                                              ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:112:92: error: request for member 'Q' in something not a structure or union
     ret = mbedtls_ecdsa_verify(&ecdsa_context.grp, image_digest, DIGEST_LEN, &ecdsa_context.Q, &r, &s);
                                                                                            ^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:118:5: error: implicit declaration of function 'mbedtls_ecdsa_free' [-Werror=implicit-function-declaration]
     mbedtls_ecdsa_free(&ecdsa_context);

ROYMURLI, 2019-08-26:
Any suggestion @dobairoland

dobairoland, 2019-08-26:
Hi @ROYMURLI. IDF v3.3 and `master` are very different so please make sure you remove the build directory (`rm -r build`) after switching between them and make sure that the submodules are updated. If you still facing the issue then please open a new issue for this because this will be closed soon and the two things are unrelated. Thank you!

closed 2019-08-27 by igrr