espressif/esp-idf #3951: Where can I access the data transmitted via Bluetooth in the A2DP-Sink example ?

philison, 2019-08-21:
Hello,

I am trying to access the data packages that are being send from my phone when I connect to the ESP32, which is running the [ESP-IDF A2DP-SINK demo](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink). The example is working great.
But I can't identify the pointer or variable that is pointing to the place where the data is stored. Or is it stored in some kind of queue or directly send to the audio output ?

(PS: I had no luck getting an answer on the esp32-forum so far)

Many thanks in advance,
Phil

avsheth, 2019-08-21:
Hi @philison 
You can look at bt_app_a2d_data_cb in bt_app_av.c.
This will be a SBC-decoded PCM data.

philison, 2019-08-21:
Thank you very much @avsheth !
I'll have a look at it.

philison, 2019-08-21:
I think I found what I was looking for @avsheth . The Data-Stream is written to the **RingBuffer** ringbuf_i2s of type **Byte buffer** (defined in the bt_app_core.c file) by the _write_ringbuf_ function, which is called in the _bt_app_a2d_data_cb_ function. I think I can now modify the task that is normaly used to get the data from the **RingBuffer** ringbuf_i2s into the i2s interface.

But two things I did not understand:
1. Where is the CallbackFunction _bt_app_a2d_data_cb_ actually called with actuall parameters ?
2. In which format is the data (used by the _bt_app_a2d_data_cb_ function as a parameter) that is written to the **RingBuffer** ringbuf_i2s ? Is it still encoded as SBC-signal or decoded to what? just PCM?

I want to do a correlation with the streamed audio signal and a prestored signal. 

Thx in advance,
Phil

avsheth, 2019-08-26:
Hi @philison 
Sorry for the delayed response.
For 1. bt_app_a2d_data_cb is actually called from bluedroid stack, (btc_a2dp_sink.c file), whenever data from the source device is received.
For 2. Data received from the bluedroid stack is PCM data. Decoding from SBC to PCM is done by the stack and is passed to the application via this callback.

Hope this helps.

Thanks

Alvin1Zhang, 2019-10-11:
@philison Thanks for reporting, would you please help share if any updates for your issue, has it been resolved? Thanks.

safaeis, 2019-11-10:
@philison, second that, working on a similar project that needs access to the A2DP sink data stream here.

Any insights would be appreciated.

Thanks,
Sam

closed 2021-09-07 by Alvin1Zhang