espressif/esp-idf #3955: EXT1 All Low / Any High documentation mismatch.

nevercast, 2019-08-23:
Hello, I'm having trouble getting EXT1 wakeup working, and I would like to get a clarification on the ALL_LOW vs ANY_HIGH because header file and C source comments do not match.

*esp_sleep.h*
```
/**
 * @brief Logic function used for EXT1 wakeup mode.
 */
typedef enum {
    ESP_EXT1_WAKEUP_ALL_LOW = 0,    //!< Wake the chip when all selected GPIOs go low
    ESP_EXT1_WAKEUP_ANY_HIGH = 1    //!< Wake the chip when any of the selected GPIOs go high
} esp_sleep_ext1_wakeup_mode_t;
```

*Technical Reference Manual V4.0*
```
RTC_CNTL_EXT_WAKEUP1_LV 0: external wake-up at low level, 1: external wake-up at high level.
```

*sleep_modes.c*
```
    // Set logic function (any low, all high)
    SET_PERI_REG_BITS(RTC_CNTL_EXT_WAKEUP_CONF_REG, 0x1,
            s_config.ext1_trigger_mode, RTC_CNTL_EXT_WAKEUP1_LV_S);
```

> // Set logic function (any low, all high)
does not match `esp_sleep_ext1_wakeup_mode_t` definition.

Can I get a clarification on the real behaviour of the EXT1 interrupt and the required logical states for wakeup. Further, are the RTC pins buffered with a schmitt trigger, will slow rise matter when using levels?

negativekelvin, 2019-08-23:
Comment is wrong
https://github.com/espressif/esp-idf/blob/02c7c3885e10c75772a3d0984c1cdbb6ad9484f3/components/esp32/sleep_modes.c#L554

surekhaibot, 2019-12-27:
Hi,
I am implementing deepsleep by EXT1. It is working fine on both  GPIO34, GPIO35(pull down with 10K resistor, wakeup when any high detects)  with ESP_EXT1_WAKEUP_ANY_HIGH, i need to implement with ANY LOW logic(pull up with 10K resistor, wakeup when any low detects), is it possible in esp32? if it's possible what should i do?

chegewara, 2019-12-27:
`ESP_EXT1_WAKEUP_ALL_LOW` with just 1 GPIO setup.

surekhaibot, 2019-12-27:
By using ESP_EXT1_WAKEUP_ALL_LOW it will wakeup when both gpio pins are low, but in my case when any gpio low detect it needs to wakeup from deepsleep. we need solution for any low detection when implements on multiple external gpio

chegewara, 2019-12-27:
You have only 2 options:
```
typedef enum {
    ESP_EXT1_WAKEUP_ALL_LOW = 0,    //!< Wake the chip when all selected GPIOs go low
    ESP_EXT1_WAKEUP_ANY_HIGH = 1    //!< Wake the chip when any of the selected GPIOs go high
} esp_sleep_ext1_wakeup_mode_t;
```

chegewara, 2019-12-27:
There is one option i can think of, write ULP code that is running in deep sleep, check pins value and wake up when value is 0.

igrr, 2019-12-27:
> pull up with 10K resistor, wakeup when any low detects

@surekhaibot for example, you can use a single input, pulled up with a resistor, and connect a couple of transistors between the input and the ground. Each base (or gate, in case of a FET) can be driven by a separate signal. This will result in "any low" logic with only a couple of additional external components.

surekhaibot, 2020-01-10:
Hi,
 can i use both ext0 on GPIO27 with active low(ESP_EXT1_WAKEUP_ALL_LOW ) and ext1 with multiple pins (GPIO34 and GPIO35) with active high(ESP_EXT1_WAKEUP_ANY_HIGH ) at a time for deepsleep.