espressif/esp-idf #3956: Wi-Fi connection fails to Cisco enterprise network with 802.11b rates disabled

mikaelkanstrup, 2019-08-23:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-58-g02c7c3885
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp32-2019r1
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description

With latest ESP-IDF from master Wi-Fi connections to Cisco based corporate wifi networks fails most of the time. The network uses WPA2 Enterprise (EAP-PEAP-MSCHAPv2). 

Bisecting the problem shows that the problem was introduced in the following commit:

```
commit 45dd6175cddc5a9103a89566df995a522cbe6703
Author: liu zhifu <liuzhifu@espressif.com>
Date:   Tue Jul 16 10:17:48 2019 +0800

    esp_wifi: optimize wifi rx
```

//Detailed problem description goes here.

When checking sniffer logs it seems that with the above mentioned patch ESP32 ACKs all management frames with 802.11b 1Mbit/s data rate.

The network is configured with the lower datarates disabled and it appears the then AP simply ignores all 1Mbit/s ACKs. In sniffer log this show as AP continuously retransmit the management frames.

ESP32 knows that AP does not support the 1Mbit/s data rate through beacons, probe responses and even acknowledge this by sending assoc requests with correct supported rates tag (though strangely the frame is sent with a 1Mb/s data rate).

### Expected Behavior

ESP32 should only send frames with negotiated supported data rates.

### Actual Behavior

ESP32 with the "esp_wifi: optimize wifi rx" patch sends ACKs and management frames with 1Mb/s data rate.

ESP32 without the "esp_wifi: optimize wifi rx" patch sends ACKs and management frames with a mix of 1Mb/s and 6Mb/s data rates. Even though this appears to be accepted by the AP and a connection can be established, ESP32 should ideally only send frames with the negotiated data rates.

### Steps to repropduce

1. Configure ESP32 with proper wifi network settings
2. Try to connect to Cisco wifi enterprise network with 802.1b (and more) data rates disabled.
3. Notice that connection fails nearly 100% of the times.

## Debug Logs

Sniffer logs attached

mikaelkanstrup, 2019-08-23:
Sniffer logs attached
[rx-optimized-fail.tar.gz](https://github.com/espressif/esp-idf/files/3533690/rx-optimized-fail.tar.gz)

liuzfesp, 2019-08-28:
Hi @mikaelkanstrup it's new introduced by commit 45dd6175cddc5a9103a89566df995a522cbe6703, we will fix it soon.

mikaelkanstrup, 2019-08-28:
> Hi @mikaelkanstrup it's new introduced by commit [45dd617](https://github.com/espressif/esp-idf/commit/45dd6175cddc5a9103a89566df995a522cbe6703), we will fix it soon.

@liuzfesp great! Thanks!

Alvin1Zhang, 2019-10-10:
@mikaelkanstrup Would you please help try with the commit and see whether your issue is resolved? Thanks.

kanstrup, 2019-10-10:
@esp-daiwei @Alvin1Zhang Most likely that will solve the issue, it's a revert of the commit causing the issue after all. But it's not available on master branch. Only available on v4.0.

liuzfesp, 2019-11-15:
Hi @kanstrup master fix commit: b7334b03, we already merged the fix. However, it may take several days before you can view it on the github IDF master branch.

mikaelkanstrup, 2019-11-20:
I've done some initial testing and still see that ESP32 use non-negotiated data rates. However ESP32 has stopped ACKing with only 1Mbit data rate which indicates whatever was done in 45dd617 is nowreverted. I am able to connect again so problem appears to be solved.

@liuzfesp how is b7334b0 related to 45dd617 that introduced the problem? Under our config we end up with default  ESP32_WIFI_IRAM_OPT=y and ESP32_WIFI_RX_IRAM_OPT=y so don't how see b7334b0 can be the solution to this problem.

negativekelvin, 2019-11-20:
Probably the fix is included in the library update in https://github.com/espressif/esp-idf/commit/a9c1970c031301f232d0c8473dcacdb7c9c12f7e

mikaelkanstrup, 2019-11-20:
@negativekelvin OK thanks. I'll do some more testing before closing the issue.

liuzfesp, 2019-11-20:
@mikaelkanstrup，b7334b0 has no relationship with this problem. b7334b0 actually involves several WiFi changes which is related to WiFi receiving, one of these change is to revert 45dd617.

mikaelkanstrup, 2019-11-20:
@liuzfesp Thanks for info and thanks for solving this issue. After some more testing done I can confirm this is fixed.

closed 2019-11-20 by mikaelkanstrup