espressif/esp-idf #3957: websockets-client cannot handle received messages faster than 1/second

mangodan2003, 2019-08-23:
## Environment

- Development Kit:      [OLIMEX/ESP32-POE]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.0-dev-1443-g39f090a4f
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

Note I am using the OLIMEX/ESP32-POE which requires some trivial changes to enable power to the PHY and set the Address to 0, however this should be irrelevant for the purposes of this report.

There is a hard coded WEBSOCKET_SEND_EVENT_TIMEOUT_MS set to 1 second in esp_websocket_client.c. This in conjunction with the implementation of the loop in esp_websocket_client_task() restricts receiving messages to this rate. 

### Expected Behavior

Be able to receive and handle messages as fast as dispatched by the server  - hardware limitations permitting.

### Actual Behavior

The websocket-client as it stands is unable to receive messages faster than a  rate of one per second.

### Steps to repropduce

1. Copy the websocket_example demo
2. Make minor changes such that it sends messages every 100mS and waits for 10 seconds before cleaning up:

### Code to reproduce this issue

change

```
    while (i < 10) {
        if (esp_websocket_client_is_connected(client)) {
            int len = sprintf(data, "hello %04d", i++);
            ESP_LOGI(TAG, "Sending %s", data);
            esp_websocket_client_send(client, data, len, portMAX_DELAY);
        }
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
```
to

```
    while (i < 10) {
        if (esp_websocket_client_is_connected(client)) {
            int len = sprintf(data, "hello %04d", i++);
            ESP_LOGI(TAG, "Sending %s", data);
            esp_websocket_client_send(client, data, len, portMAX_DELAY);
        }
        vTaskDelay(100 / portTICK_RATE_MS);
    }
    vTaskDelay(10000 / portTICK_RATE_MS);
```

```
(8445) WEBSOCKET: Connectiong to ws://echo.websocket.org...
I (8795) WEBSOCKET: WEBSOCKET_EVENT_CONNECTED
I (8855) WEBSOCKET: Sending hello 0000
I (8955) WEBSOCKET: Sending hello 0001
I (9055) WEBSOCKET: Sending hello 0002
I (9155) WEBSOCKET: Sending hello 0003
I (9255) WEBSOCKET: Sending hello 0004
I (9355) WEBSOCKET: Sending hello 0005
I (9455) WEBSOCKET: Sending hello 0006
I (9555) WEBSOCKET: Sending hello 0007
I (9655) WEBSOCKET: Sending hello 0008
I (9755) WEBSOCKET: Sending hello 0009
I (9815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (9815) WEBSOCKET: Received=hello 0000

I (10815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (10815) WEBSOCKET: Received=hello 0001

I (11815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (11815) WEBSOCKET: Received=hello 0002

I (12815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (12815) WEBSOCKET: Received=hello 0003

I (13815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (13815) WEBSOCKET: Received=hello 0004

I (14815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (14815) WEBSOCKET: Received=hello 0005

I (15815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (15815) WEBSOCKET: Received=hello 0006

I (16815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (16815) WEBSOCKET: Received=hello 0007

I (17815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (17815) WEBSOCKET: Received=hello 0008

I (18815) WEBSOCKET: WEBSOCKET_EVENT_DATA
W (18815) WEBSOCKET: Received=hello 0009

I (20815) WEBSOCKET: Websocket Stopped
```

negativekelvin, 2019-08-23:
The problem is this call is forcing the event loop to run for 1 second
https://github.com/espressif/esp-idf/blob/6517c5033aab27afe491e433d6beeba93dfb1aa7/components/esp_websocket_client/esp_websocket_client.c#L117

If you change WEBSOCKET_SEND_EVENT_TIMEOUT_MS to 0 it will work.  I'm not sure about the concept of forcing the event loop to run for a period of time even if the queue is empty.

mangodan2003, 2019-08-24:
Yes I had come to the same conclusion, I had not actually tried setting it to 0 but set to 100mS. The example code would then run as expected. But I do not yet  know if this will lead to problems elsewhere, or why this call exists with a 1 second timeout in the first place.

negativekelvin, 2019-08-24:
Bad design? The task that posts to the event loop is the only thing that runs the event loop so really it is not doing anything and by having the 1 second timeout it just blocks itself.  Changing to 0 just turns it into a regular synchronous callback.

closed 2019-10-17 by igrr