espressif/esp-idf #3959: Unknown CMake command "idf_build_set_property".

jiapei100, 2019-08-23:
How to build **esp-idf** from scratch?

```console
-- Configuring incomplete, errors occurred!
See also "/opt/mcu/esp-idf/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
longervision-GT72-6QE% idf.py clean
Checking Python dependencies...
Python requirements from /opt/mcu/esp-idf/requirements.txt are satisfied.
Executing action: clean
Running cmake in directory /opt/mcu/esp-idf/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /opt/mcu/esp-idf"...
Warn about uninitialized values.
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:58 (idf_build_set_property):
  Unknown CMake command "idf_build_set_property".

-- Configuring incomplete, errors occurred!
See also "/opt/mcu/esp-idf/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
...
```

HenriquePereiraRosa, 2019-08-23:
same Problem here.

`The system is: Windows - 10.0.17134 - AMD64`
`Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
The system cannot find the file specified

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
The system cannot find the file specified

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -Aa 

The output was:
The system cannot find the file specified

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -D__CLASSIC_C__ 

The output was:
The system cannot find the file specified

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
The system cannot find the file specified

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
The system cannot find the file specified

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -Aa 

The output was:
The system cannot find the file specified

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: CMAKE_C_COMPILER-NOTFOUND 
Build flags: 
Id flags: -D__CLASSIC_C__ 

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --c++ 

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --ec++ 

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --c++ 

The output was:
The system cannot find the file specified

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --ec++ 

The output was:
The system cannot find the file specified`

renzbagaporo, 2019-08-26:
@jiapei100 Are you executing `idf.py` from IDF_PATH? If so please execute the build from a project directory such as `examples/get-started/hello_world`.

@HenriquePereiraRosa This looks like a different problem from @jiapei100's. Is `xtensa-esp32-elf-gcc` in your PATH?

HenriquePereiraRosa, 2019-08-29:
@renzbagaporo You are Right. My mistake. A reinstaled the esp-tool but forgot to start another CMD instance. Solved here.

renzbagaporo, 2019-09-02:
Closing this issue @jiapei100, then.

closed 2019-09-02 by renzbagaporo