espressif/esp-idf #3960: Failed to recurse into submodule path 'components/coap/libcoap'

hero, 2019-08-24:
## Problem Description
Today(2019-08-24 09:30), I want to update my esp-idf, I execute the following `git` commands:
```
git pull
git submodule sync
git submodule update --init --recursive
```
I got a error message: 
```bash
error: Server does not allow request for unadvertised object 7f8c86e501e690301630029fa9bae22424adf618
Fetched in submodule path 'components/coap/libcoap/ext/tinydtls', but it did not contain 7f8c86e501e690301630029fa9bae22424adf618. Direct fetching of that commit failed.
Failed to recurse into submodule path 'components/coap/libcoap'
```
How to solve this problem？ Any suggestion will be appreciated
BTW, It is highly recommended that you make a synchronized mirror on gitee.com

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2019-08-24:
https://github.com/espressif/esp-idf/issues/3938#issuecomment-524183334

mahavirj, 2019-10-07:
@hero Did you get chance to try suggestion from above comment? Do you still have an issue or we can close this?

mahavirj, 2019-10-09:
Closing, per suggestions in above comment. Please feel free to open if issue still exists.

closed 2019-10-09 by mahavirj