espressif/esp-idf #3964: cant login on forum

chegewara, 2019-08-26:
Yes, i am lame, i know, but what is the answer:
```
Name one of the wireless protocols Espressifs products are famous for supporting.:
This question is a means of preventing automated form submissions by spambots.
```

closed 2019-08-26 by chegewara

guiglass, 2019-12-28:
What is the answer?

chegewara, 2019-12-28:
`wifi` IIRC.

nathanRamaNoodles, 2020-03-12:
I thought it was `ESPNOW`, but the answer is `wifi`
XD

threeguys, 2020-03-22:
This has got to be one of the worst captcha implementations I have ever seen in my entire life, you folks should be ashamed of yourselves. Are you actively trying to keep people from being able to post on the forums? This is laughably bad. At least give folks a hint of the answer, freaking "wifi", you have got to be kidding me. This actually makes me want to drop your products completely.

chegewara, 2020-03-22:
I feel better when i know im not the only lame ;)

projectgus, 2020-03-23:
Sorry for the confusion about this question.

I've added a hint to it now.

The list of acceptable answers is actually longer than "wifi": it includes Wi-Fi, wifi, "Wi Fi", 802.11, 802.11b, 802.11g, Bluetooth, blue tooth, bt, etc. Basically, trying to anticipate anything someone who was not a bot might type in here.

I've just  added ESP-NOW & ESPNOW, BLE-MESH and BLEMESH to the accepted list. If you think I'm missing another wireless protocol, let me know.

chegewara, 2020-03-23:
A year ago i thought it may be ble or espnow, but none worked. Dont know if that has been changed.

threeguys, 2020-03-24:
Ok, so apparently I fail at puzzles. For future reference, I plugged in everything I could think of (which idk if you're checking case, none of them worked, tcp/ip, ip, 802.11) and then went to the "Protocols" page under the documentation and tried plugging all of those in: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/index.html

Sorry for the tone of my earlier comment, I am trying to be a good citizen but was just really pissed off after wasting so much time before searching for it (which in hindsight I should have done in the first place :).