espressif/esp-idf #3965: idf.py missing target defconfig

ShellAddicted, 2019-08-26:
## Environment
- Hardware: Not relevant
- IDF version: latest master 
v4.1-dev-58-g02c7c3885 
(HEAD: 02c7c3885e10c75772a3d0984c1cdbb6ad9484f3)
- Build System: cmake idf.py
- Compiler version: esp32-2019r1 8.2.0 (not relevant)
- Operating System:  Linux

## Problem Description
I'm switching to cmake, I need an equivalent of `make defconfig` for `idf.py`
Docs (https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html)
shows the `idf.py defconfig` command.
but 
an `Error: No such command "defconfig"` occurs.

### Expected Behavior
same as `make defconfig`, sdkconfig.h should be generated

### Actual Behavior
an Error occurs, and file is not created

### Steps to repropduce

1. use get-started/hello_world as template
2. execute `idf.py defconfig`

kumekay, 2019-08-28:
Hi @ShellAddicted ,
Thank you for the issue,
`defconfig` is not implemented in CMake build system. If you need to generate `sdkconfig.h` you can use `idf.py reconfigure` command as the closest equivalent.

We will update documentation to reflect current state.

Alvin1Zhang, 2020-01-15:
@ShellAddicted Thanks for reporting. The fix has been available, feel free to reopen if the issue still happens. Thanks.

closed 2020-01-15 by Alvin1Zhang