espressif/esp-idf #3966: [wifi mesh]  do not work  when the router does not exist.

FFtust, 2019-08-26:
#3401  Environment

- Development Kit:      [None]
- Kit version (for WroverKit/PicoKit/DevKitC):
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-beta3-6b3da6b1882f3b72e904cc90be67e9c4e3f369a9
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

I update esp-idf to the version v3.3-beta3, from version V3.1-a18f92b. 
In the old IDF, the wifi mesh can work without a router avaliable, but the new IDF do not work. 

I use the same config for wifi mesh, the main cofig is:

          tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP);
    	  tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA);
		
		  /*  wifi initialization */
		  wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT();
		  esp_wifi_init(&config);
		  esp_wifi_set_storage(WIFI_STORAGE_FLASH);
		  esp_wifi_start();
		
		  /*  mesh initialization */
		  esp_mesh_init();
		  esp_mesh_set_max_layer(CONFIG_MESH_MAX_LAYER);
		  esp_mesh_set_vote_percentage(1);
		  esp_mesh_set_ap_assoc_expire(10);
		
		  esp_mesh_fix_root(true);
          ......
          esp_mesh_set_self_organized(1, 0);

For the new IDF, mesh only work when the router is truly existed(I used a mobile hotspot), when I closed the mobile hotspot, the node can not find any root device.And then I opened the mobile hotspot, it worked well.

The problem is that it performed differently in the old version, and our product havs been designed to be working without a router avaliable. Is there any solution to work without a router.Thank you very mush for your help.

### Expected Behavior
WIFI mesh in IDF(v3.3-beta3-6b3da6b1882f3b72e904cc90be67e9c4e3f369a9) can work without a router.

### Actual Behavior

### Steps to repropduce

### Code to reproduce this issue

## Debug Logs

## Other items if possible

dongdong004, 2019-08-27:
There's an example in ESP-MDF. 
[esp-mdf get_started example](https://github.com/espressif/esp-mdf/tree/master/examples/get-started)
```
Introduction
It introduces a quick way to build an ESP-MESH network without a router. 
```
In short, you need the following configurations to build a no-router network.
```
#ifdef ROOT
        esp_mesh_set_type(MESH_ROOT);
#else /*NON_ROOT*/
        esp_mesh_fix_root(1);
#endif /*ifdef ROOT*/
    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
    //.....
    // do not need router set here
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
```

FFtust, 2019-08-27:
@dongdong004
     Thanks very mush for your help, I have successfully started a no-router network. It seems the problem is that I have configed the router. I removed all the codes related to the router, and then it did work.
     Thanks again for your prompt reply.

closed 2019-08-27 by FFtust