espressif/esp-idf #3971: Problem with mbedtls component in IDF v4.1-dev-58-g02c7c3885

ROYMURLI, 2019-08-27:
I was working with IDF v3.3 and developed my whole project in this version only. Now, I am facing some problem related to wifi connectivity because of BLE and BT involvement. I downloaded new IDF v4.1-dev-58-g02c7c3885 and tried after removing my old
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3544652/sdkconfig.txt)

 "build" folder from project. I am getting some mbedtls undefined error which seems to be some library and command line declaration problem. 

OS- Windows 10 (msys32)
xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

Things I did:- 
-Cloned the latest IDF and moved my project to example folder.
-Added "COMPONENT_SDKCONFIG_RENAMES := $(shell cygpath -w $(COMPONENT_SDKCONFIG_RENAMES))"
after line https://github.com/espressif/esp-idf/blob/master/make/project_config.mk#L11 
Because I was facing problem with "no such file and directory .............. sdkconfig_rename" which is solved using above solution.
- rm -r build did for build directory.
- sdkconfig are attached below.
Debug logs are given below:-

D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c: In function 'esp_secure_boot_verify_signature_block':
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:96:5: error: unknown type name 'mbedtls_ecdsa_context'
mbedtls_ecdsa_context ecdsa_context;
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:97:5: error: implicit declaration of function 'mbedtls_ecdsa_init' [-Werror=implicit-function-declaration]
mbedtls_ecdsa_init(&ecdsa_context);
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:5: error: implicit declaration of function 'mbedtls_ecp_group_load' [-Werror=implicit-function-declaration]
mbedtls_ecp_group_load(&ecdsa_context.grp, MBEDTLS_ECP_DP_SECP256R1);
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:42: error: request for member 'grp' in something not a structure or union
mbedtls_ecp_group_load(&ecdsa_context.grp, MBEDTLS_ECP_DP_SECP256R1);
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:48: error: 'MBEDTLS_ECP_DP_SECP256R1' undeclared (first use in this function)
mbedtls_ecp_group_load(&ecdsa_context.grp, MBEDTLS_ECP_DP_SECP256R1);
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:99:48: note: each undeclared identifier is reported only once for each function it appears in
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c💯50: error: request for member 'grp' in something not a structure or union
size_t plen = mbedtls_mpi_size(&ecdsa_context.grp.P);
^
In file included from D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/asn1.h:36:0,
from D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/x509.h:33,
from D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:22:
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:108:59: error: request for member 'Q' in something not a structure or union
MBEDTLS_MPI_CHK(mbedtls_mpi_read_binary(&ecdsa_context.Q.X, signature_verification_key_start, plen));
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/bignum.h:52:22: note: in definition of macro 'MBEDTLS_MPI_CHK'
if( ( ret = (f) ) != 0 ) 
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:109:59: error: request for member 'Q' in something not a structure or union
MBEDTLS_MPI_CHK(mbedtls_mpi_read_binary(&ecdsa_context.Q.Y, signature_verification_key_start + plen, plen));
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/bignum.h:52:22: note: in definition of macro 'MBEDTLS_MPI_CHK'
if( ( ret = (f) ) != 0 ) 
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:110:52: error: request for member 'Q' in something not a structure or union
MBEDTLS_MPI_CHK(mbedtls_mpi_lset(&ecdsa_context.Q.Z, 1));
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/mbedtls/mbedtls/include/mbedtls/bignum.h:52:22: note: in definition of macro 'MBEDTLS_MPI_CHK'
if( ( ret = (f) ) != 0 ) 
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:112:11: error: implicit declaration of function 'mbedtls_ecdsa_verify' [-Werror=implicit-function-declaration]
ret = mbedtls_ecdsa_verify(&ecdsa_context.grp, image_digest, DIGEST_LEN, &ecdsa_context.Q, &r, &s);
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:112:46: error: request for member 'grp' in something not a structure or union
ret = mbedtls_ecdsa_verify(&ecdsa_context.grp, image_digest, DIGEST_LEN, &ecdsa_context.Q, &r, &s);
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:112:92: error: request for member 'Q' in something not a structure or union
ret = mbedtls_ecdsa_verify(&ecdsa_context.grp, image_digest, DIGEST_LEN, &ecdsa_context.Q, &r, &s);
^
D:/msys32/home/MurlidharRoy/Newfolder/esp-idf/components/bootloader_support/src/idf/secure_boot_signatures.c:118:5: error: implicit declaration of function 'mbedtls_ecdsa_free' [-Werror=implicit-function-declaration]
mbedtls_ecdsa_free(&ecdsa_context);

mahavirj, 2019-08-27:
@ROYMURLI Could you please enable https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L486 and see if that fixes your problem?

ROYMURLI, 2019-08-27:
It is already enabled. Kconfig file attached here.
[Kconfig - Copy.txt](https://github.com/espressif/esp-idf/files/3545402/Kconfig.-.Copy.txt)

mahavirj, 2019-08-27:
@ROYMURLI If you check your `sdkconfig` you will find line `# CONFIG_MBEDTLS_ECP_C is not set` which means particular option is disabled for your project. You can enable by selecting option at, `make menuconfig -> component config -> mbedtls -> Elliptic Curve Ciphers`

ROYMURLI, 2019-08-27:
Still getting following error:-
SDKCONFIG is attached here-
[sdkconfig - Copy.txt](https://github.com/espressif/esp-idf/files/3545662/sdkconfig.-.Copy.txt)

D:/msys32/home/MurlidharRoy/NewFolder/esp-idf/components/coap/port/coap_mbedtls.c: In function 'setup_server_ssl_session':
D:/msys32/home/MurlidharRoy/NewFolder/esp-idf/components/coap/port/coap_mbedtls.c:897:3: error: implicit declaration of function 'mbedtls_ssl_conf_dtls_cookies' [-Werror=implicit-function-declaration]
   mbedtls_ssl_conf_dtls_cookies(&m_env->conf, mbedtls_ssl_cookie_write,
   ^
D:/msys32/home/MurlidharRoy/NewFolder/esp-idf/components/coap/port/coap_mbedtls.c: In function 'coap_dtls_hello':
D:/msys32/home/MurlidharRoy/NewFolder/esp-idf/components/coap/port/coap_mbedtls.c:1646:15: error: implicit declaration of function 'mbedtls_ssl_set_client_transport_id' [-Werror=implicit-function-declaration]
     if((ret = mbedtls_ssl_set_client_transport_id(&m_env->ssl,

mahavirj, 2019-08-27:
@ROYMURLI This is known issue (https://github.com/espressif/esp-idf/issues/3961) and we are working on a fix. Interim you may select, `make menuconfig -> component config -> mbedtls -> TLS Protocol Role -> Server & Client` as a workaround for this.

ROYMURLI, 2019-08-27:
Tried your above suggestion and it worked. But, Let me know when this problem will fixed.

enricop, 2019-09-03:
In our project as well we are using only MbedTLS Client role so we disabled the server part.  We would like to continue to do so in master branch.

mahavirj, 2019-09-03:
@ROYMURLI @enricop Fix is available at https://github.com/espressif/esp-idf/pull/3977, it will merge soon to IDF

ROYMURLI, 2019-09-03:
Great. Thanks @mahavirj .

closed 2019-09-05 by ROYMURLI