espressif/esp-idf #3973: Uncorrect ADC iniziatization with I2S

GianlucaLoi, 2019-08-27:
#3401  Environment

- Development Kit:     ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):8.2.0
- Operating System:     Windows
- Power Supply:         external 3.3V

## Problem Description

I'm developing a system on ESP32 in order to sample a signal and I need an high sample rate (around 51200Hz). Fort this reason I'm using the I2S-ADC mode.
The input signal is composed by a dc value that is generated by an ESP's dac pin and an ac signal from outside.
So, at the adc pin, at least is possible to measure the dc value due to the dac pin.

Sometimes, when the board start for the first time and I initialize the I2S - ADC section, I'm not able to read the data from the dma. To be more specific, I read a zero value buffer. 
When I have this problem, I try a re-init of I2S-ADC section but I continue to see this problem. The only solution is to turn off the board, wait some seconds and the turn on. This solution, clearly, is not acceptable.

### Expected Behavior

I read no-zero values on the dma buffer

### Actual Behavior

Sometimes, after the I2S init in ADC Mode, I read zero value buffer from the dma. Re-init the I2S after the zero value read does not change the situation. Only a power off can be useful to restore the correct working mode.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.

static void adc_i2s_init(void)
{
	esp_err_t ret = ESP_OK;

	i2s_config_t adc_i2s_config =
	{
			.mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
			.sample_rate = SAMPLE_RATE,
			.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
			.communication_format = I2S_COMM_FORMAT_I2S_MSB,
			.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,					
			.intr_alloc_flags = 0,
			.dma_buf_count = DMA_BUF_COUNT,									
			.dma_buf_len = DMA_BUF_LEN,										
			.use_apll = false,
			.tx_desc_auto_clear = true										
	};

	//install and start i2s driver
	ret = i2s_driver_install(I2S_UNIT, &adc_i2s_config, 1, &adc_i2s_event_queue);
	if (ret  != ESP_OK)
	{
		ESP_LOGE(TAG, "i2s_driver_install operation error = %i", ret);
	}

	//init ADC pad
	ret = i2s_set_adc_mode_custom(ADC_UNIT, ADC_VOLTAGE_CHANNEL, ADC_VOLTAGE_WIDTH, ADC_VOLTAGE_ATTEN_DB, ADC_DATA_INV);
	if (ret  != ESP_OK)
	{
		ESP_LOGE(TAG, "i2s_set_adc_mode_custom operation error = %i", ret);
	}

	vTaskDelay(2000);
}

static void adc_i2s_enable(void)
{
	esp_err_t ret = ESP_OK;

	ESP_ERROR_CHECK(i2s_zero_dma_buffer(I2S_UNIT));

	ret = i2s_adc_enable_custom(I2S_UNIT, ADC_VOLTAGE_CHANNEL, ADC_VOLTAGE_WIDTH, ADC_VOLTAGE_ATTEN_DB, ADC_DATA_INV);
	if (ret  != ESP_OK)
	{
		ESP_LOGE(TAG, "i2s_adc_enable_custom operation error = %i", ret);
	}
}

static void adc_i2s_disable(void)
{
	i2s_adc_disable(I2S_UNIT);
}

static void adc_i2s_restore_sampling(void)
{
	adc_i2s_disable();
	adc_disable_function();
	vTaskDelay(100);
	adc_power_on();
	adc_i2s_init();
	adc_i2s_enable();
}

static void adc_task(void *pvParameter)
{
	esp_err_t ret = ESP_OK;
	i2s_event_t adc_i2s_evt;
	size_t bytes_read;

	/* Reset ADC */
	adc_disable_function();
	/*********************/

	adc_i2s_init();
	adc_i2s_enable();

	while ( 1 )
	{
		if(xQueueReceive(adc_i2s_event_queue, (void * )&adc_i2s_evt, (portTickType)portMAX_DELAY))
		{
			if(adc_i2s_evt.type == I2S_EVENT_RX_DONE)
			{
				ret = i2s_read(I2S_UNIT, (char*)adc_data_buf, NUM_SAMPLES*2, &bytes_read, portMAX_DELAY);
				if (ret  != ESP_OK)
				{
					ESP_LOGE(TAG, "i2s_read operation error = %i", ret);
				}

				if(((adc_data_buf[0] & 0xFFF) == 0) && ((adc_data_buf[1] & 0xFFF) == 0) && ((adc_data_buf[2] & 0xFFF) == 0))
				{
					// If the adc reads zero values, it means it hasn't been started correctly. Restart it again to fix this problem.
					adc_i2s_restore_sampling();
				}
				else
				{
					// Does something
				}
			}
		}
	}
	adc_i2s_disable(); // Unreachable
}

```

GianlucaLoi, 2019-08-27:
> Using WiFi?
> 
> #3714

Hello @negativekelvin,

In my project I will use it, but this problem appears also when the WiFi is not initialized.

Regards,
Gianluca.

GianlucaLoi, 2019-08-28:
@Alvin1Zhang @negativekelvin 
Further information:
I put some logs after the i2s_read.
```cpp
				ret = i2s_read_custom(I2S_UNIT, (char*)adc_data_buf, NUM_SAMPLES*2, &bytes_read, portMAX_DELAY);
				if (ret  != ESP_OK)
				{
					ESP_LOGE(TAG, "i2s_read operation error = %i", ret);
				}

				if(((adc_data_buf[0]) == 0) && ((adc_data_buf[1]) == 0) && ((adc_data_buf[2]) == 0))
				{
					printf("adc_data_buf[0] = %d adc_data_buf[1] = %d adc_data_buf[2] = %d\n",adc_data_buf[0],adc_data_buf[1],adc_data_buf[2]);
					// If the adc reads zero values, it means it hasn't been started correctly. Restart it again to fix this problem.
					adc_i2s_restore_sampling();
				}
				else
				{
					printf("adc_data_buf[0] = %d adc_data_buf[1] = %d adc_data_buf[2] = %d\n",adc_data_buf[0],adc_data_buf[1],adc_data_buf[2]);
					ESP_LOGW(TAG, "RESTARTING");
					vTaskDelay(1000);
                                        // Restart is done to reproduce the reading problem
					esp_restart();
				}
```
When it works correctly I read:
I (247) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=2
I (248) I2S: PLL_D2: Req RATE: 51600, real rate: 3333.000, BITS: 16, CLKM: 25, BCK: 60, MCLK: 25.840, SCLK: 106656.000000, diva: 64, divb: 53
I (2249) I2S: PLL_D2: Req RATE: 51600, real rate: 3333.000, BITS: 16, CLKM: 25, BCK: 60, MCLK: 25.840, SCLK: 106656.000000, diva: 64, divb: 53
adc_data_buf[0] = 30847 adc_data_buf[1] = 30861 adc_data_buf[2] = 30871
W (2250) ADC_TASK: RESTARTING

Instead, in wrong case:
I (247) I2S: queue free spaces: 1
I (247) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=2
I (248) I2S: PLL_D2: Req RATE: 51600, real rate: 3333.000, BITS: 16, CLKM: 25, BCK: 60, MCLK: 25.840, SCLK: 106656.000000, diva: 64, divb: 53
I (2249) I2S: PLL_D2: Req RATE: 51600, real rate: 3333.000, BITS: 16, CLKM: 25, BCK: 60, MCLK: 25.840, SCLK: 106656.000000, diva: 64, divb: 53
adc_data_buf[0] = 0 adc_data_buf[1] = 0 adc_data_buf[2] = 0

X-Ryl669, 2019-09-05:
Can you check you still have a WS on the output pin with an oscilloscope ?
```
// Output WS signal to gpio18(only for debug mode)
void i2s_adc_check_clk(void)
{
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[18], PIN_FUNC_GPIO);
    gpio_set_direction(18, GPIO_MODE_DEF_OUTPUT);
    gpio_matrix_out(18, I2S0I_WS_OUT_IDX, 0, 0);

    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[21], PIN_FUNC_GPIO);
    gpio_set_direction(21, GPIO_MODE_DEF_OUTPUT);
    gpio_set_level(21, 0);
}
```

It's just to make sure the I2S is still running (or if the SAR is still running).

GianlucaLoi, 2019-09-05:
> Can you check you still have a WS on the output pin with an oscilloscope ?
>
> ```
> // Output WS signal to gpio18(only for debug mode)
> void i2s_adc_check_clk(void)
> { [...]

Hello,

I've already done this test and during the fault the ws clock is still running.

Regards,

Gianluca.

projectgus, 2019-11-05:
Hi @GianlucaLoi ,

Thanks for the bug report. The bootloader_random_disable() function which runs during early boot can accidentally leave the I2S peripheral in a bad state. A fix has been tested and will be merged to master branch soon.

Angus

GianlucaLoi, 2019-11-05:
Hi @projectgus,

Thanks for your reply. I'm glad that this problem has been found and fixed.
Could you tell me when the fix is ready?

Regards,

Gianluca.

closed 2019-11-14 by espressif-bot