espressif/esp-idf #3974: ENC28J60 initialise phy device timeout

Mikegyver, 2019-08-27:
hi everyone. i'm facing the same issue #2331 but with different ethernet shield.

FYI, i'm using ENC28J60 and following are the error appear on arduino serial monitor.

`E (1033) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0x0000`
`E (2033) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0x0000`
`E (2033) emac: Initialise PHY device Timeout`

i did the test using example code in the library from http://www.fisch.lu/junk/ESP32-Ethernet.zip

hope the details above will provide enough information regarding this issue. regards.

suda-morris, 2019-08-28:
Hi @Mikegyver 

1. ENC28J60 is a SPI to Ethernet solution, IDF 3.x doesn't support this kind of Ethernet module. But 4.x does support, we already have a support for [DM9051](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_phy_dm9051.c) (mostly similar to ENC28J60, but more powerful)
2. Your code is for LAN8720, not ENC28J60.

Alvin1Zhang, 2019-08-30:
@Mikegyver Thanks for the reporting this. Feel free to open another ticket if you have more issues, thanks.

closed 2019-08-30 by Alvin1Zhang

Mikegyver, 2019-08-30:
> Hi @Mikegyver
>
> 1. ENC28J60 is a SPI to Ethernet solution, IDF 3.x doesn't support this kind of Ethernet module. But 4.x does support, we already have a support for [...]

@suda-morris & @Alvin1Zhang. I just wonder why esp32 not support ethernet based spi protocol? I have try on lan8720. But to be honest, it is pain in the butt. Instead of using 6 pins for enc28j60 / w5500, i wasted 12 pins only for lan8720. And for lan8720, the ethernet shield cannot be just plug and play, but instead need additional circuit for pullup & pull down circuit.

suda-morris, 2019-08-30:
Sorry @Mikegyver You're right, ESP32 provides many choices for Ethernet connections, but the most "painful" part could be the RMII clock and GPIO0 in ESP32 because ESP32 allows us to do some tricks to reduce BOM cost and power consumption.
Circuit hack is common when you're using a module that not exactly designed for your MCU. 
BTW, if you want to try SPI based Ethernet, now you have two choices:
1. Switch to Arduino library, so you can use ENC28j60 and w5500 easily
2. Switch to IDF4.x, where has a support for a SPI Ethernet module `DM9051`, you can write another similar driver for `ENC28J60`, take the DM9051 as a reference

Mikegyver, 2019-08-30:
@suda-morris i think even using arduino IDE, for both ENC28J60 & W5500 still have any working functional library. I look into espressif forum, still seing few posts raising the issue.