espressif/esp-idf #3975: clang: error: unknown argument: '-n'

Allineer, 2019-08-27:
#  Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.1-dev-104-gaa087667d
- Build System:        CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     macOS
- Power Supply:         USB|external 3.3V

## Problem Description

```
$ idf.py menuconfig

Checking Python dependencies...
Python requirements from /Users/X/esp/esp-idf/requirements.txt are satisfied.
Executing action: menuconfig
Running ninja in directory /Users/X/project/build
Executing "ninja menuconfig"...
[1/4] Performing build step for 'mconf-idf'
FAILED: mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build kconfig_bin/mconf-idf 
cd /Users/X/project/build/kconfig_bin && rm -f /Users/X/esp/esp-idf/tools/kconfig/zconf.lex.c /Users/X/esp/esp-idf/tools/kconfig/zconf.hash.c && make -f /Users/X/esp/esp-idf/tools/kconfig/Makefile mconf-idf && /usr/local/Cellar/cmake/3.15.2/bin/cmake -E touch /Users/X/project/build/mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build
cc -c  -DCURSES_LOC="<curses.h>" -n -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MMD -MP -I "." -I "/Users/X/esp/esp-idf/tools/kconfig"  /Users/X/project/esp/esp-idf/tools/kconfig/mconf.c -o mconf.o
clang: error: unknown argument: '-n'
make: *** [mconf.o] Error 1
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

dobairoland, 2019-08-28:
I'm sorry for the inconvenience. We are reverting back part of https://github.com/espressif/esp-idf/commit/d296aad2a99f16978031b1561a64927d771f76be#diff-1c6ecb5835068bd596b18b14d3b98df6 in commit f2a4131c241fa1959283e903bc884bc63b948af5. It will be pushed to GH ASAP.

Temporary workaround is to change `#!/bin/sh` back to `#!/bin/bash` in `tools/kconfig/lxdialog/check-lxdialog.sh`.

Related PR: https://github.com/espressif/esp-idf/pull/3972

dobairoland, 2019-09-02:
Closed by f2a4131c241fa1959283e903bc884bc63b948af5. Please report if there is any subsequent issue.

closed 2019-09-02 by dobairoland