espressif/esp-idf #3978: Issue with building esp32 firmware (missing files apparently)

rudecerik, 2019-08-28:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-104-gaa087667d
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB|]

## Problem Description
Can't build esp32 firmware, says it's missing files (and when I navigate to there it indeed is missing) tried adding them to the directory manually but it doesn't work.
IDF_PATH is set correctly, and xtensa-esp32-elf/bin is added to the path

> CC /home/srsbr/esp/esp-idf/components/bootloader_support/src/bootloader_clock.c
/home/srsbr/esp/esp-idf/components/bootloader_support/src/bootloader_clock.c:14:10: fatal error: esp32/rom/uart.h: No such file or directory
 #include "esp32/rom/uart.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:678: recipe for target 'build-GENERIC/bootloader//home/srsbr/esp/esp-idf/components/bootloader_support/src/bootloader_clock.o' failed
make: *** [build-GENERIC/bootloader//home/srsbr/esp/esp-idf/components/bootloader_support/src/bootloader_clock.o] Error 1

stevenj, 2019-10-22:
I just struck this problem now, myself.

projectgus, 2020-04-30:
Hi @rudecerik ,

I'm sincerely sorry noone got back to you about this earlier. I'm honestly not sure what this problem was, but I'm assuming it was transient in some way to do with paths.

If the issue still occurs on the latest release/v4.1 branch, please leave a comment and we'll reopen.

closed 2020-04-30 by projectgus