espressif/esp-idf #3980: Please, explain me how to `esp_register_freertos_idle_hook_for_cpu` works?

Allineer, 2019-08-28:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.2.2
- Build System:        CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     macOS
- Power Supply:         USB|external 3.3V

## Problem Description

Hello!
Please, look in this sample:

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_freertos_hooks.h"
#include "esp_log.h"

typedef struct {
    uint32_t total;
    uint32_t idle;
} cpu_core_usage_t;

cpu_core_usage_t cpu_usage[portNUM_PROCESSORS];

static const char* TAG = "main";

//==============================================================================

void cpu_core_tick_callback() {
    cpu_usage[xPortGetCoreID()].total++;
}

bool cpu_core_idle_callback() {
    cpu_usage[xPortGetCoreID()].idle++;

    // The callback should return true if it should be called by the idle hook
    // once per interrupt (or FreeRTOS tick)
    return true;
}

void cpu_usage_task(void *pvParameter)
{
  for( ;; ) {
    ESP_LOGE(TAG, "%d/%d - %d/%d", cpu_usage[0].idle, cpu_usage[0].total, cpu_usage[1].idle, cpu_usage[1].total);
    vTaskDelay(pdMS_TO_TICKS(1000));
  }
}

void app_main()
{
  // CPU usage hooks (total & idle ticks counters)
  for (int core = 0; core < portNUM_PROCESSORS; core++) {
      esp_register_freertos_tick_hook_for_cpu(cpu_core_tick_callback, core);
      esp_register_freertos_idle_hook_for_cpu(cpu_core_idle_callback, core);
  }

  xTaskCreate(
    &cpu_usage_task,
    "cpu_usage_task",
    2048,
    NULL,
    tskIDLE_PRIORITY + 5,
    NULL
  );
}
```

I have tick & idle hooks. I have one running task.

### Actual Behavior && Expected Behavior

I expected to see sometimes than total ticks count is greater than IDLE ticks count.
But I see that IDLE ticks count is always equals to total count:

```
E (0) main: 0/0 - 0/0
E (2239) main: 100/100 - 100/100
E (3239) main: 200/200 - 200/200
E (4239) main: 300/300 - 300/300
E (5239) main: 400/400 - 400/400
E (6239) main: 500/500 - 500/500
E (7239) main: 600/600 - 600/600
E (8239) main: 700/700 - 700/700
E (9239) main: 800/800 - 800/800
E (10239) main: 900/900 - 900/900
```

I see this output for vanilla FreeRTOS hooks too.
What's wrong with this sample? Why it equals when app is running?
Thanks.

igrr, 2019-08-28:
Can you please explain, why you expect the idle task to run more than once per tick, in this example?
If you add an additional task which would be woken up inbetween two ticks, then the idle task would run twice per tick: before the higher priority task runs, and after it yields.

igrr, 2019-08-28:
Your task finishes it's work (printf) much faster than an interval between two ticks, and once it finishes, the idle task runs. This means that the idle task runs once every tick, in your case. Note that in general, the idle task may run multiple times during a tick, if other tasks are unblocked from ISR and then yield again. So in your code, "total" may be less than "idle".

In general, this may be an XY problem here. Perhaps if you explain what you are trying to do, someone may suggest a way to do this. Also it would be more appropriate to have such discussions at esp32.com, unless you believe there is some bug in ESP-IDF.

Allineer, 2019-08-28:
@igrr, look:

- tick_hook is counting all ticks (100 ticks per second in my case), right?
- some part of this ticks don't do any work and will be counted by idle_hook, right?
- my sample has a task that shows the results of counting ticks, i.e. her work takes some ticks, right?

Looks logical, but why the number of idle ticks match the total number for both CPU cores?

Allineer, 2019-08-28:
@igrr,
> Your task finishes it's work (printf) much faster than an interval between two ticks, and once it finishes, the idle task runs.

Yahoo! I didn’t think that in one tick can work out a my tasks and the idle task together.

Спасибо, Иван!

closed 2019-08-28 by Allineer