espressif/esp-idf #3981: Issue with building my firmvare for esp32DevKir 1.1

snikul, 2019-08-28:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     ESP32-DevKitC|
- Kit version (for DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version :  v4.1-dev-104-gaa087667d
- Build System:         [CMake]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Windows]
- Power Supply:         [external 5V]

## Problem Description
C:\Users\GreenOnion\esp\blink>idf.py clean
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt are satisfied.
Executing action: clean
Running cmake in directory c:\users\greenonion\esp\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\greenonion\esp\blink"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.22.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- broken
CMake Error at C:/Users/GreenOnion/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeTmp

    Run Build Command:"C:/Users/GreenOnion/.espressif/tools/ninja/1.9.0/ninja.exe" "cmTC_dcfa7"
    [1/2] Building C object CMakeFiles/cmTC_dcfa7.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_dcfa7
    FAILED: cmTC_dcfa7
    cmd.exe /C "cd . && C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/cmTC_dcfa7.dir/testCCompiler.c.obj  -o cmTC_dcfa7   && cd ."
    xtensa-esp32-elf-gcc.exe: error: CMakeFiles/cmTC_dcfa7.dir/testCCompiler.c.obj: No such file or directory
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:257 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

C:\Users\GreenOnion\esp\blink>

----------------------------------------
The same probleme with idf.py menuconfig and idf.build

What does it mean:
-- Check for working C compiler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- broken

I would appreciate your frienfly reply
Regards
Sergei Nikulin

renzbagaporo, 2019-08-30:
Hi @snikul can you post the logs mentioned in the build?

```
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeError.log".
```

snikul, 2019-08-30:
Dear Renz,
please, find my log files in the attachment.
I have just changed the toolchain's folder from
C:/Users/GreenOnion/esp/.expressif
to C:/Users/GreenOnion/Program Files/.expressif.
The result is the same.

Env. variables for current user GreenOnion:
IDF_TOOLS_PATH/ C:\Program Files\.espressif

C:\Users\GreenOnion>path
PATH=C:\Python27\;C:\Python27\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program
Files\Java\jdk-12.0.1\bin;C:\Program Files\dotnet\;C:\Program
Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL
Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Git\cmd;C:\Program
Files\PuTTY\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK
Utility\ST-LINK Utility;C:\Program Files (x86)\Bitvise SSH
Client;C:\Program Files (x86)\Atmel\sam-ba_2.18\drv\;C:\Program Files
(x86)\Atmel\sam-ba_2.18;C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\;C:\Users\GreenOnion\AppData\Local\Microsoft\WindowsApps

C:\Users\GreenOnion>python -V
Python 2.7.16

C:\Users\GreenOnion>

Intel(R) i7-5500U CPU 2,4GHz / 8G W10/1903

Screen log --------------------------------------------------------

C:\Users\GreenOnion\esp\blink>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt
are satisfied.
Executing action: all (aliases: build)
Running cmake in directory c:\users\greenonion\esp\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1
--warn-uninitialized -DCCACHE_ENABLE=0 c:\users\greenonion\esp\blink"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version
"2.22.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program
Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program
Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program
Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- broken
CMake Error at C:/Program
Files/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52
(message):
  The C compiler

    "C:/Program
Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeTmp

    Run Build
Command:"C:/PROGRA~1/ESPRES~1/tools/ninja/196380~1.0/ninja.exe" "cmTC_097c7"
    [1/2] Building C object CMakeFiles/cmTC_097c7.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_097c7
    FAILED: cmTC_097c7
    cmd.exe /C "cd . &&
C:\PROGRA~1\ESPRES~1\tools\XTENSA~1\ESP32-~1.0\XTENSA~1\bin\XT8994~1.EXE
-mlongcalls -Wno-frame-address  -nostdlib
CMakeFiles/cmTC_097c7.dir/testCCompiler.c.obj  -o cmTC_097c7   && cd ."
    XT8994~1.EXE: error: CMakeFiles/cmTC_097c7.dir/testCCompiler.c.obj: No
such file or directory
    XT8994~1.EXE: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:257 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

C:\Users\GreenOnion\esp\blink>
--------------------------------------------------------------------------

My Best Regards
Sergei Nikulin

snikul, 2019-08-30:
Dear Renz,
I have just replied to you mail with log files in the attachment. 
I post the copy of my mail here too.
----------------------
I have just changed the toolchain's folder from C:/Users/GreenOnion/esp/.expressif to C:/Users/GreenOnion/Program Files/.expressif.
The result is the same.  
 
Env. variables for current user GreenOnion:
IDF_TOOLS_PATH/ C:\Program Files\.espressif

C:\Users\GreenOnion>path
PATH=C:\Python27\;C:\Python27\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Java\jdk-12.0.1\bin;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files (x86)\Atmel\sam-ba_2.18\drv\;C:\Program Files (x86)\Atmel\sam-ba_2.18;C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\;C:\Users\GreenOnion\AppData\Local\Microsoft\WindowsApps

C:\Users\GreenOnion>python -V
Python 2.7.16

C:\Users\GreenOnion>

Intel(R) i7-5500U CPU 2,4GHz / 8G W10/1903

Screen log -----------------------------------------------
C:\Users\GreenOnion\esp\blink>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory c:\users\greenonion\esp\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\greenonion\esp\blink"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.22.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- broken
CMake Error at C:/Program Files/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "C:/Program Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeTmp

    Run Build Command:"C:/PROGRA~1/ESPRES~1/tools/ninja/196380~1.0/ninja.exe" "cmTC_097c7"
    [1/2] Building C object CMakeFiles/cmTC_097c7.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_097c7
    FAILED: cmTC_097c7
    cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\tools\XTENSA~1\ESP32-~1.0\XTENSA~1\bin\XT8994~1.EXE -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/cmTC_097c7.dir/testCCompiler.c.obj  -o cmTC_097c7   && cd ."
    XT8994~1.EXE: error: CMakeFiles/cmTC_097c7.dir/testCCompiler.c.obj: No such file or directory
    XT8994~1.EXE: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:257 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

C:\Users\GreenOnion\esp\blink>
CMakeError.log -----------------------------------------------------------
[... 28 lines cut ...]

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/3.13.4/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: C:/Program Files/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/3.13.4/CompilerIdCXX/a.out"

Checking whether the ASM compiler is GNU using "--version" matched "(GNU assembler)|(GCC)|(Free Software Foundation)":
xtensa-esp32-elf-gcc.exe (crosstool-NG esp32-2019r1) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
------------------------------------------------------
My Best Regards
Sergei Nikulin

snikul, 2019-08-31:
I have just found the same problem in an other context but no solution there:
https://www.esp32.com/viewtopic.php?f=2&t=10948&p=44769&hilit=xtensa+esp+elf+gcc+broken#p44769

renzbagaporo, 2019-09-02:
Hi @snikul, unfortunately I haven't been able to pin the problem down just yet. However, if this issue is blocking your work,  you can try adding the following line at the very top of the project CMakeLists.txt:

```
set(CMAKE_C_COMPILER_WORKS 1)
```

This is a temporary workaround and allows skipping of the compiler checks. Please remove the existing build folder and sdkconfig and retry the build.

snikul, 2019-09-02:
Hi, Renz,
The result with set(CMAKE_C_COMPILER_WORKS 1)  at the very top of the project CMakeLists.txt (in blink folder) and deleted build folder and sdkconfig.defaults  is different but still negative:
-------

C:\Users\GreenOnion\esp\blink>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory c:\users\greenonion\esp\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\greenonion\esp\blink"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.22.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Detecting C compile features
-- Detecting C compile features - failed
-- Check for working CXX compiler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- broken
CMake Error at C:/Users/GreenOnion/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCXXCompiler.cmake:45 (message):
  The C++ compiler

    "C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeTmp

    Run Build Command:"C:/Users/GreenOnion/.espressif/tools/ninja/1.9.0/ninja.exe" "cmTC_66679"
    [1/2] Building CXX object CMakeFiles/cmTC_66679.dir/testCXXCompiler.cxx.obj
    [2/2] Linking CXX executable cmTC_66679
    FAILED: cmTC_66679
    cmd.exe /C "cd . && C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/cmTC_66679.dir/testCXXCompiler.cxx.obj  -o cmTC_66679   && cd ."
    xtensa-esp32-elf-g++.exe: error: CMakeFiles/cmTC_66679.dir/testCXXCompiler.cxx.obj: No such file or directory
    xtensa-esp32-elf-g++.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:258 (__project)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/GreenOnion/esp/blink/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

C:\Users\GreenOnion\esp\blink>

My Best Redards
Sergei

renzbagaporo, 2019-09-03:
Please also add:

```
set(CMAKE_CXX_COMPILER_WORKS 1)
```

temporarily at the top of the project CMakeLists.txt. 

Have you tried re-downloading the toolchain?

snikul, 2019-09-05:
Hi, Renz,
Still the same story. I have re-downloaded the toolchain.
I can idf.py clean && idf.py menuconfig now. But not idf.py all:
--------
C:\Users\GreenOnion\esp\blink>idf.py all
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\users\greenonion\esp\blink\build
Executing "ninja all"...
[114/714] Performing configure step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure
cmd.exe /C "cd /D C:\Users\GreenOnion\esp\blink\build\bootloader && C:\Users\GreenOnion\.espressif\tools\cmake\3.13.4\bin\cmake.exe -DSDKCONFIG=C:/Users/GreenOnion/esp/blink/sdkconfig -DIDF_PATH=C:/Users/GreenOnion/esp/esp-idf -DIDF_TARGET=esp32 -DPYTHON_DEPS_CHECKED=1 -DEXTRA_COMPONENT_DIRS=C:/Users/GreenOnion/esp/esp-idf/components/bootloader -DLEGACY_INCLUDE_COMMON_HEADERS= -GNinja C:/Users/GreenOnion/esp/esp-idf/components/bootloader/subproject && C:\Users\GreenOnion\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E touch C:/Users/GreenOnion/esp/blink/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- broken
CMake Error at C:/Users/GreenOnion/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeTmp

    Run Build Command:"C:/Users/GreenOnion/.espressif/tools/ninja/1.9.0/ninja.exe" "cmTC_39f20"
    [1/2] Building C object CMakeFiles/cmTC_39f20.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_39f20
    FAILED: cmTC_39f20
    cmd.exe /C "cd . && C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/cmTC_39f20.dir/testCCompiler.c.obj  -o cmTC_39f20   && cd ."
    xtensa-esp32-elf-gcc.exe: error: CMakeFiles/cmTC_39f20.dir/testCCompiler.c.obj: No such file or directory
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:258 (__project)
  CMakeLists.txt:27 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeOutput.log".
See also "C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeError.log".
[119/714] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/emac_hal.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\GreenOnion\esp\blink>

snikul, 2019-09-05:
Hi, Renz,
Still the same story. I have re-downloaded the toolchain.
I can idf.py clean && idf.py menuconfig now. But not idf.py all:
(By the way, now problems with cMake esp-idf v4.1 master on VM Ubuntu)
--------
C:\Users\GreenOnion\esp\blink>idf.py all
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt
are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\users\greenonion\esp\blink\build
Executing "ninja all"...
[114/714] Performing configure step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure
cmd.exe /C "cd /D C:\Users\GreenOnion\esp\blink\build\bootloader &&
C:\Users\GreenOnion\.espressif\tools\cmake\3.13.4\bin\cmake.exe
-DSDKCONFIG=C:/Users/GreenOnion/esp/blink/sdkconfig
-DIDF_PATH=C:/Users/GreenOnion/esp/esp-idf -DIDF_TARGET=esp32
-DPYTHON_DEPS_CHECKED=1
-DEXTRA_COMPONENT_DIRS=C:/Users/GreenOnion/esp/esp-idf/components/bootloader
-DLEGACY_INCLUDE_COMMON_HEADERS= -GNinja
C:/Users/GreenOnion/esp/esp-idf/components/bootloader/subproject &&
C:\Users\GreenOnion\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E touch
C:/Users/GreenOnion/esp/blink/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- broken
CMake Error at
C:/Users/GreenOnion/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52
(message):
  The C compiler

"C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir:
C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeTmp

    Run Build
Command:"C:/Users/GreenOnion/.espressif/tools/ninja/1.9.0/ninja.exe"
"cmTC_39f20"
    [1/2] Building C object CMakeFiles/cmTC_39f20.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_39f20
    FAILED: cmTC_39f20
    cmd.exe /C "cd . &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe
-mlongcalls -Wno-frame-address  -nostdlib
CMakeFiles/cmTC_39f20.dir/testCCompiler.c.obj  -o cmTC_39f20   && cd ."
    xtensa-esp32-elf-gcc.exe: error:
CMakeFiles/cmTC_39f20.dir/testCCompiler.c.obj: No such file or directory
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:258 (__project)
  CMakeLists.txt:27 (project)

-- Configuring incomplete, errors occurred!
See also
"C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeOutput.log".
See also
"C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeError.log".
[119/714] Building C object
esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/emac_hal.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\GreenOnion\esp\blink>

Best Regards
Sergei Nikulin

snikul, 2019-09-08:
Hi, Renz,
I got something new. Log below. Read it, pl. Maybe you will find the
solution.
------------
C:\Users\GreenOnion\esp\blink>idf.py all
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt
are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\users\greenonion\esp\blink\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project is not inside a git repository, or git repository has no
commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script
C:/Users/GreenOnion/esp/blink/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/GreenOnion/esp/esp-idf/components/esp32/ld/
esp32.project.ld.in
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt
cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common
esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota
esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client
esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap
idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp
nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc
spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling
wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/GreenOnion/esp/esp-idf/components/app_trace
C:/Users/GreenOnion/esp/esp-idf/components/app_update
C:/Users/GreenOnion/esp/esp-idf/components/asio
C:/Users/GreenOnion/esp/esp-idf/components/bootloader
C:/Users/GreenOnion/esp/esp-idf/components/bootloader_support
C:/Users/GreenOnion/esp/esp-idf/components/bt
C:/Users/GreenOnion/esp/esp-idf/components/cbor
C:/Users/GreenOnion/esp/esp-idf/components/coap
C:/Users/GreenOnion/esp/esp-idf/components/console
C:/Users/GreenOnion/esp/esp-idf/components/cxx
C:/Users/GreenOnion/esp/esp-idf/components/driver
C:/Users/GreenOnion/esp/esp-idf/components/efuse
C:/Users/GreenOnion/esp/esp-idf/components/esp-tls
C:/Users/GreenOnion/esp/esp-idf/components/esp32
C:/Users/GreenOnion/esp/esp-idf/components/esp_adc_cal
C:/Users/GreenOnion/esp/esp-idf/components/esp_common
C:/Users/GreenOnion/esp/esp-idf/components/esp_eth
C:/Users/GreenOnion/esp/esp-idf/components/esp_event
C:/Users/GreenOnion/esp/esp-idf/components/esp_gdbstub
C:/Users/GreenOnion/esp/esp-idf/components/esp_http_client
C:/Users/GreenOnion/esp/esp-idf/components/esp_http_server
C:/Users/GreenOnion/esp/esp-idf/components/esp_https_ota
C:/Users/GreenOnion/esp/esp-idf/components/esp_https_server
C:/Users/GreenOnion/esp/esp-idf/components/esp_local_ctrl
C:/Users/GreenOnion/esp/esp-idf/components/esp_ringbuf
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom
C:/Users/GreenOnion/esp/esp-idf/components/esp_websocket_client
[... 91 lines cut ...]
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:258 (__project)
  CMakeLists.txt:27 (project)

-- Configuring incomplete, errors occurred!
See also
"C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeOutput.log".
See also
"C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeError.log".
[199/827] Building C object
esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/wps/wps_validate.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\GreenOnion\esp\blink>
-------------------------
Regards
Sergei

пт, 6 сент. 2019 г. в 02:16, Sergey Nikulin <serg.nikulin@gmail.com>:

renzbagaporo, 2019-09-09:
Please also apply this temporary workaround to the top of [bootloader/subproject/CMakeLists.txt](https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/CMakeLists.txt).

snikul, 2019-09-09:
Hi, Renz,
Sending you fresh screen log and log files you requested (I deleted build
folder and sdkconfig and sdkconfig.old files
before idf.py menuconfig  and idf.py.all):
------
C:\Users\GreenOnion\esp\blink>idf.py all
Checking Python dependencies...
Python requirements from C:\Users\GreenOnion\esp\esp-idf\requirements.txt
are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\users\greenonion\esp\blink\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project is not inside a git repository, or git repository has no
commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script
C:/Users/GreenOnion/esp/blink/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/GreenOnion/esp/esp-idf/components/esp32/ld/
esp32.project.ld.in
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt
cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common
esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota
esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client
esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap
idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp
nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc
spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling
wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/GreenOnion/esp/esp-idf/components/app_trace
C:/Users/GreenOnion/esp/esp-idf/components/app_update
C:/Users/GreenOnion/esp/esp-idf/components/asio
C:/Users/GreenOnion/esp/esp-idf/components/bootloader
C:/Users/GreenOnion/esp/esp-idf/components/bootloader_support
C:/Users/GreenOnion/esp/esp-idf/components/bt
C:/Users/GreenOnion/esp/esp-idf/components/cbor
C:/Users/GreenOnion/esp/esp-idf/components/coap
C:/Users/GreenOnion/esp/esp-idf/components/console
C:/Users/GreenOnion/esp/esp-idf/components/cxx
C:/Users/GreenOnion/esp/esp-idf/components/driver
C:/Users/GreenOnion/esp/esp-idf/components/efuse
C:/Users/GreenOnion/esp/esp-idf/components/esp-tls
C:/Users/GreenOnion/esp/esp-idf/components/esp32
C:/Users/GreenOnion/esp/esp-idf/components/esp_adc_cal
C:/Users/GreenOnion/esp/esp-idf/components/esp_common
C:/Users/GreenOnion/esp/esp-idf/components/esp_eth
C:/Users/GreenOnion/esp/esp-idf/components/esp_event
C:/Users/GreenOnion/esp/esp-idf/components/esp_gdbstub
C:/Users/GreenOnion/esp/esp-idf/components/esp_http_client
C:/Users/GreenOnion/esp/esp-idf/components/esp_http_server
C:/Users/GreenOnion/esp/esp-idf/components/esp_https_ota
C:/Users/GreenOnion/esp/esp-idf/components/esp_https_server
C:/Users/GreenOnion/esp/esp-idf/components/esp_local_ctrl
C:/Users/GreenOnion/esp/esp-idf/components/esp_ringbuf
C:/Users/GreenOnion/esp/esp-idf/components/esp_rom
[... 92 lines cut ...]
    xtensa-esp32-elf-gcc.exe: error:
CMakeFiles/cmTC_abe90.dir/testCCompiler.c.obj: No such file or directory
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:258 (__project)
  CMakeLists.txt:27 (project)

-- Configuring incomplete, errors occurred!
See also
"C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeOutput.log".
See also
"C:/Users/GreenOnion/esp/blink/build/bootloader/CMakeFiles/CMakeError.log".
[302/834] Building C object
esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/ipv4/etharp.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\GreenOnion\esp\blink>
----------
Best Regards
Sergei

renzbagaporo, 2019-12-18:
Hi @snikul, are you still experiencing this problem?

snikul, 2020-01-02:
Hi, Renz!
I was out of the game  the last month. Mama after the hospital.  Took care. I have just returned to my comp. Inform you a little bit later. Regards.

renzbagaporo, 2020-01-06:
No problem @snikul.

beNative, 2020-02-05:
Hi, I'm joining the discussion here. On my Windows system I did run into the exact same problems as Sergei described.
This is the output I'm getting:

`PS D:\development\platformio\esp-idf> python ./tools/idf.py menuconfig
Note: You are using Python 3.7.2. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Setting IDF_PATH environment variable: D:\development\platformio\esp-idf
Checking Python dependencies...
Python requirements from D:\development\platformio\esp-idf\requirements.txt are satisfied.
Running cmake in directory D:\development\platformio\esp-idf\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized D:\development\platformio\esp-idf"...
Warn about uninitialized values.
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- broken
CMake Error at D:/development/platformio/ESP-IDF tools/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/development/platformio/esp-idf/build/CMakeFiles/CMakeTmp

    Run Build Command:"D:/DEVELO~1/PLATFO~1/ESP-ID~1/tools/ninja/196380~1.0/ninja.exe" "cmTC_d12ac"
    [1/2] Building C object CMakeFiles/cmTC_d12ac.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_d12ac.exe
    FAILED: cmTC_d12ac.exe
    cmd.exe /C "cd . && D:\DEVELO~1\PLATFO~1\ESP-ID~1\tools\XTENSA~1\ESP32-~1.0\XTENSA~1\bin\XT8994~1.EXE    CMakeFiles/cmTC_d12ac.dir/testCCompiler.c.obj  -o cmTC_d12ac.exe -Wl,--out-implib,libcmTC_d12ac.dll.a -Wl,--major-image-version,0,--minor-image-version,0   && cd ."
    d:/develo~1/platfo~1/esp-id~1/tools/xtensa~1/esp32-~1.0/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: unrecognized option '--major-image-version'
    d:/develo~1/platfo~1/esp-id~1/tools/xtensa~1/esp32-~1.0/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: use the --help option for usage information
    collect2.exe: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "D:/development/platformio/esp-idf/build/CMakeFiles/CMakeOutput.log".
See also "D:/development/platformio/esp-idf/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
PS D:\development\platformio\esp-idf> python $env:IDF_PATH/tools/idf.py menuconfig
D:\development\python\python.exe: can't open file '/tools/idf.py': [Errno 2] No such file or directory
PS D:\development\platformio\esp-idf> python %IDF_PATH%/tools/idf.py menuconfig
D:\development\python\python.exe: can't open file '%IDF_PATH%/tools/idf.py': [Errno 2] No such file or directory
PS D:\development\platformio\esp-idf> python ./tools/idf.py menuconfig
Note: You are using Python 3.7.2. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Setting IDF_PATH environment variable: D:\development\platformio\esp-idf
Checking Python dependencies...
Python requirements from D:\development\platformio\esp-idf\requirements.txt are satisfied.
Running cmake in directory D:\development\platformio\esp-idf\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized D:\development\platformio\esp-idf"...
Warn about uninitialized values.
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- broken
CMake Error at D:/development/platformio/ESP-IDF tools/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "D:/development/platformio/ESP-IDF tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.
[... 41 lines cut ...]

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/development/platformio/esp-idf/build/CMakeFiles/CMakeTmp

[... cut ...]
    d:/develo~1/platfo~1/esp-id~1/tools/xtensa~1/esp32-~1.0/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: use the --help option for usage information
    collect2.exe: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "D:/development/platformio/esp-idf/build/CMakeFiles/CMakeOutput.log".
See also "D:/development/platformio/esp-idf/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1`

renzbagaporo, 2020-04-26:
Hi @beNative, are you still experiencing this problem? Sorry for the late reply.

snikul, 2020-04-27:
Yes, I have, Renz! I am back. A week ago I got my new Esp32 DevKitC V4 with
esp32-wroom-32D. The installation of the latest soft is OK.
And the same story...

renzbagaporo, 2020-04-28:
You're experiencing the same problem as @beNative, not the old problem?

snikul, 2020-04-28:
Hi, Renz,
Sending you

snikul, 2020-04-28:
Hi, Renz,

Sending you the info about my problem with idf.py menuconfig on my new
Esp32 DevKitC V4 with esp32-wroom-32D which I had with my old Esp32 DevKitC
V1 with esp32-wroom-32/
 Python 3.7.7 installed
User Environment Veriables
C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\Scripts\
C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\
System Environment Veriables
C:\Users\GreenOnion\AppData\Roaming\Python\Python37\Scripts

C:\Users\GreenOnion>python
Python 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64
bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()

C:\Users\GreenOnion>python --version
Python 3.7.7

C:\Users\GreenOnion\esp\esp-idf>git describe
v4.2-dev-1262-g4a558131e

C:\Users\GreenOnion\Downloads>dir
11.04.2020  02:58       351 306 304 esp-idf-tools-setup-2.3.exe

After installation:

C:\Users\GreenOnion\esp\esp-idf>path
PATH=C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin;
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32s2-elf\esp-2020r1-8.2.0\xtensa-esp32s2-elf\bin;
C:\Users\GreenOnion\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;
C:\Users\GreenOnion\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;
C:\Users\GreenOnion\.espressif\tools\cmake\3.16.4\bin;
C:\Users\GreenOnion\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200420\openocd-esp32\bin;
C:\Users\GreenOnion\.espressif\tools\ninja\1.10.0\;
C:\Users\GreenOnion\.espressif\tools\idf-exe\1.0.1\;
C:\Users\GreenOnion\.espressif\tools\ccache\3.7\;
C:\Users\GreenOnion\.espressif\python_env\idf4.2_py3.7_env\Scripts;
C:\Users\GreenOnion\esp\esp-idf\tools;
C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\;
C:\Program Files\Git\cmd\;
C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\;
C:\Program Files\Git\cmd\;
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;
C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;
C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;
C:\Program Files\Java\jdk-12.0.1\bin;C:\Program Files\dotnet\;
C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;
C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;
C:\Program Files\PuTTY\;C:\Program Files (x86)\STMicroelectronics\STM32
ST-LINK Utility\ST-LINK Utility;
C:\Program Files (x86)\Bitvise SSH Client;
C:\Program Files\cURL\bin;C:\Program Files (x86)\Windows Kits\10\Windows
Performance Toolkit\;
C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\x64\;
C:\mysql\bin;C:\Program Files\Zerynth\ztc\windows64;
C:\Program Files (x86)\Brackets\command;
C:\Program
Files\Git\cmd;C:\Users\GreenOnion\Documents\Arduino\hardware\espressif\esp32\tools\esptool;
C:\Users\GreenOnion\Documents\esp8266_NodeMCU\ESPlorer\ESPlorer;
C:\Users\GreenOnion\AppData\Roaming\Python\Python37\Scripts;
C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\Scripts\;
C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\

C:\Users\GreenOnion\esp\esp-idf>

GreenOnion User environment variables.
IDF_TOOLS_PATH = C:\Users\GreenOnion\.espressif
[... 71 lines cut ...]
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/esp/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also
"C:/Users/GreenOnion/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also
"C:/Users/GreenOnion/esp/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

I am seek and tide of it. Help me to find the solution, Renz!

Best Regards
Sergei

>
>> .
>>
>

snikul, 2020-05-29:
Hi, Rent,
I still hope you will help me with my error.
I installed toolchain successfully with
https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe
Using Python in C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\
Python 3.7.7
Using Git in C:\Program Files\Git\cmd\
git version 2.21.0.windows.1
Setting IDF_PATH: C:\esp-idf

Adding ESP-IDF tools to PATH...

C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin

C:\Users\GreenOnion\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Users\GreenOnion\.espressif\tools\cmake\3.13.4\bin

C:\Users\GreenOnion\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Users\GreenOnion\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Users\GreenOnion\.espressif\tools\ninja\1.9.0\
    C:\Users\GreenOnion\.espressif\tools\idf-exe\1.0.1\
    C:\Users\GreenOnion\.espressif\tools\ccache\3.7\
    C:\Users\GreenOnion\.espressif\python_env\idf4.0_py3.7_env\Scripts
    C:\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\esp-idf>
C:\esp-idf>cd /esp

C:\esp>xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world
C:\esp-idf\examples\get-started\hello_world\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\Makefile
C:\esp-idf\examples\get-started\hello_world\README.md
C:\esp-idf\examples\get-started\hello_world\main\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\main\component.mk
C:\esp-idf\examples\get-started\hello_world\main\hello_world_main.c
Скопировано файлов: 6.

C:\esp>cd hello_world
C:\esp\hello_world>idf.py menuconfig

C:\esp\hello_world>idf.py menuconfig
Checking Python dependencies...
Python requirements from C:\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1
--warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=1
c:\esp\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version
"2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- broken
CMake Error at
C:/Users/GreenOnion/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52
(message):
  The C compiler

"C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/esp/hello_world/build/CMakeFiles/CMakeTmp

    Run Build
Command:"C:/Users/GreenOnion/.espressif/tools/ninja/1.9.0/ninja.exe"
"cmTC_9821e"
    [1/2] Building C object CMakeFiles/cmTC_9821e.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_9821e
    FAILED: cmTC_9821e
    cmd.exe /C "cd . &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe
-mlongcalls -Wno-frame-address  -nostdlib
CMakeFiles/cmTC_9821e.dir/testCCompiler.c.obj  -o cmTC_9821e   && cd ."
    xtensa-esp32-elf-gcc.exe: error:
CMakeFiles/cmTC_9821e.dir/testCCompiler.c.obj: No such file or directory
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/project.cmake:269 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "C:/esp/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

C:\esp\hello_world>

*****************************************
Using Python 3.7:
      C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\python.exe

Using Git 2.21.0.windows.1:
      C:\Program Files\Git\cmd\git.exe

Using existing ESP-IDF copy:
      C:\esp-idf

IDF tools directory (IDF_TOOLS_PATH):
      C:\Users\GreenOnion\.espressif
*************************
I would appreciate your friendly reply!

Best regards
Sergei Nikulin

вт, 28 апр. 2020 г. в 23:45, Sergey Nikulin <serg.nikulin@gmail.com>:

> Hi, Renz,
>
> Sending you the info about my problem with idf.py menuconfig on my new
> Esp32 DevKitC V4 with esp32-wroom-32D which I had with my old Esp32 DevKitC
> V1 with esp32-wroom-32/
>  Python 3.7.7 [...]

snikul, 2020-05-29:
Sorry, Renz,

I excuse me, pl, I made mistake in your Name in my previous letter!

My Best Regards
Sergei Nikulin

пт, 29 мая 2020 г. в 13:55, Sergey Nikulin <serg.nikulin@gmail.com>:

> Hi, Rent,
> I still hope you will help me with my error.
> I installed toolchain successfully with
> https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe
> Using Python in [...]

renzbagaporo, 2020-06-01:
Hi @snikul, I am terribly sorry about my late replies. Just to be sure, have you tried:

- building from any other directory other than `C:\esp`
- putting `esp-idf` in another directory (not C:\esp-idf)  during install?

snikul, 2020-06-02:
Hi, Renz,
I downloaded esp-idf 4.0 release.
I copied hello_world to C:\Users\GreenOnion\Desktop\esp
and esp-idf to  C:\Users\GreenOnion\Desktop\esp-idf
then I jumped to  C:\Users\GreenOnion\Desktop\esp\hello_world started
idf.py menuconfig
and got the same result as before with error.
****
Using Python 3.7:
      C:\Users\GreenOnion\AppData\Local\Programs\Python\Python37\python.exe

Using Git 2.21.0.windows.1:
      C:\Program Files\Git\cmd\git.exe

Using existing ESP-IDF copy:
      C:\Users\GreenOnion\Desktop\esp-idf

IDF tools directory (IDF_TOOLS_PATH):
      C:\Users\GreenOnion\.espressif
****************
C:\Users\GreenOnion\Desktop\esp>cd hello_world

C:\Users\GreenOnion\Desktop\esp\hello_world>idf.py menuconfig
Checking Python dependencies...
Python requirements from
C:\Users\GreenOnion\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\greenonion\desktop\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1
--warn-uninitialized -DCCACHE_ENABLE=0
c:\users\greenonion\desktop\esp\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version
"2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler:
C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- broken
CMake Error at
C:/Users/GreenOnion/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/CMakeTestCCompiler.cmake:52
(message):
  The C compiler

"C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir:
C:/Users/GreenOnion/Desktop/esp/hello_world/build/CMakeFiles/CMakeTmp

    Run Build
Command:"C:/Users/GreenOnion/.espressif/tools/ninja/1.9.0/ninja.exe"
"cmTC_abdbd"
    [1/2] Building C object CMakeFiles/cmTC_abdbd.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_abdbd
    FAILED: cmTC_abdbd
    cmd.exe /C "cd . &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe
-mlongcalls -Wno-frame-address  -nostdlib
CMakeFiles/cmTC_abdbd.dir/testCCompiler.c.obj  -o cmTC_abdbd   && cd ."
    xtensa-esp32-elf-gcc.exe: error:
CMakeFiles/cmTC_abdbd.dir/testCCompiler.c.obj: No such file or directory
    xtensa-esp32-elf-gcc.exe: fatal error: no input files
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/GreenOnion/Desktop/esp-idf/tools/cmake/project.cmake:258
(__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also
"C:/Users/GreenOnion/Desktop/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also
"C:/Users/GreenOnion/Desktop/esp/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

C:\Users\GreenOnion\Desktop\esp\hello_world>

chegewara, 2020-06-02:
Its probably not the case, but maybe this path is too long:
> C:\Users\GreenOnion\Desktop\esp-idf

try:
`C:\esp-idf`

renzbagaporo, 2020-06-03:
Ok. This is difficult, as I really can't reproduce the issue you are having @snikul.

In the past, I've asked you to try skipping compiler check via setting `set(CMAKE_C_COMPILER_WORKS 1)` -- turns out this was the wrong way to do it (sorry about that). So I will be asking you again, this time you need to change the **entire** contents of the file https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-esp32.cmake into:

```
include (CMakeForceCompiler)
set(CMAKE_SYSTEM_NAME Generic)

CMAKE_FORCE_C_COMPILER   (xtensa-esp32-elf-gcc GNU)
CMAKE_FORCE_CXX_COMPILER (xtensa-esp32-elf-g++ GNU)

set(CMAKE_ASM_COMPILER xtensa-esp32-elf-gcc)

set(CMAKE_C_FLAGS "-mlongcalls -Wno-frame-address" CACHE STRING "C Compiler Base Flags")
set(CMAKE_CXX_FLAGS "-mlongcalls -Wno-frame-address" CACHE STRING "C++ Compiler Base Flags")
```

In the logs, you should see something like this:

```
-- ccache will be used for faster recompilation
CMake Deprecation Warning at C:/Users/renz/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/renz/Work/esp-idf/tools/cmake/toolchain-esp32.cmake:4 (CMAKE_FORCE_C_COMPILER)
  C:/Users/renz/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:93 (include)
  C:/Users/renz/Work/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

CMake Deprecation Warning at C:/Users/renz/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/renz/Work/esp-idf/tools/cmake/toolchain-esp32.cmake:5 (CMAKE_FORCE_CXX_COMPILER)
  C:/Users/renz/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:93 (include)
  C:/Users/renz/Work/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

CMake Deprecation Warning at C:/Users/renz/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/renz/Work/esp-idf/tools/cmake/toolchain-esp32.cmake:4 (CMAKE_FORCE_C_COMPILER)
  build/CMakeFiles/3.16.4/CMakeSystem.cmake:6 (include)
  C:/Users/renz/Work/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

CMake Deprecation Warning at C:/Users/renz/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/renz/Work/esp-idf/tools/cmake/toolchain-esp32.cmake:5 (CMAKE_FORCE_CXX_COMPILER)
  build/CMakeFiles/3.16.4/CMakeSystem.cmake:6 (include)
  C:/Users/renz/Work/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

```

These warnings are fine  -- this is supposed to skip the test compile which is where your build fails. I want to see if these tests are skipped, the builds at least go through. If not, I need to see where it next fails, which might offer up more clue on the current problem.

snikul, 2020-06-03:
Hi, Renz,

I put changes you advised me to do to  tools/cmake/toolchain-esp32.cmake
<https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-esp32.cmake>

The following is the result:
Using Python in C:\Users\GreenOnion\AppData\Local\Programs\Python\Python38\
Python 3.8.3
Using Git in C:\Program Files\Git\cmd\
git version 2.27.0.windows.1
Setting IDF_PATH: C:\esp-idf

Adding ESP-IDF tools to PATH...

C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin

C:\Users\GreenOnion\.espressif\tools\xtensa-esp32s2-elf\esp-2020r1-8.2.0\xtensa-esp32s2-elf\bin

C:\Users\GreenOnion\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin

C:\Users\GreenOnion\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\GreenOnion\.espressif\tools\cmake\3.16.4\bin

C:\Users\GreenOnion\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200420\openocd-esp32\bin
    C:\Users\GreenOnion\.espressif\tools\ninja\1.10.0\
    C:\Users\GreenOnion\.espressif\tools\idf-exe\1.0.1\
    C:\Users\GreenOnion\.espressif\tools\ccache\3.7\
    C:\Users\GreenOnion\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\GreenOnion\.espressif\python_env\idf4.2_py3.8_env\Scripts
    C:\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\esp-idf>cd ../esp

C:\esp>xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world
C:\esp-idf\examples\get-started\hello_world\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\Makefile
C:\esp-idf\examples\get-started\hello_world\README.md
C:\esp-idf\examples\get-started\hello_world\main\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\main\component.mk
C:\esp-idf\examples\get-started\hello_world\main\hello_world_main.c
Скопировано файлов: 6.

C:\esp>cd hello_world

C:\esp\hello_world>idf.py menuconfig

and I got menuconfig menu screen. Looks like everything went fine.
   <<
-----------------------------------------------------------------------------
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

The following is the screen output:
Executing action: menuconfig
Running cmake in directory c:\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1
--warn-uninitialized -DCCACHE_ENABLE=1 c:\esp\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version
"2.27.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
CMake Deprecation Warning at
[... 127 lines cut ...]
C:/esp-idf/components/spi_flash C:/esp-idf/components/spiffs
C:/esp-idf/components/tcp_transport C:/esp-idf/components/tcpip_adapter
C:/esp-idf/components/tinyusb C:/esp-idf/components/ulp
C:/esp-idf/components/unity C:/esp-idf/components/vfs
C:/esp-idf/components/wear_levelling
C:/esp-idf/components/wifi_provisioning
C:/esp-idf/components/wpa_supplicant C:/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/esp/hello_world/build
Running ninja in directory c:\esp\hello_world\build
Executing "ninja menuconfig"...
[0/1] cmd.exe /C "cd /D C:\esp\hello_world\build && C:\Use...TARGET=esp32
--output config C:/esp/hello_world/sdkconfig"
Loaded configuration 'C:/esp/hello_world/sdkconfig'
No changes to save (for 'C:/esp/hello_world/sdkconfig')

C:\esp\hello_world>idf.py menuconfig

******************************
So what was wrong? What is your opinion?
CMakeOutput.log in the attachment.
CMakeError.log not available.
Regards
Sergei

snikul, 2020-06-03:
Hi, Renz,

Then I run idf.py build
and got the following error:
C:\esp\hello_world>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp\hello_world\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
CMake Deprecation Warning at
C:/Users/GreenOnion/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:75
(message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/toolchain-esp32.cmake:4 (CMAKE_FORCE_C_COMPILER)
  build/CMakeFiles/3.16.4/CMakeSystem.cmake:6 (include)
  C:/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

CMake Deprecation Warning at
C:/Users/GreenOnion/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:89
(message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/toolchain-esp32.cmake:5 (CMAKE_FORCE_CXX_COMPILER)
  build/CMakeFiles/3.16.4/CMakeSystem.cmake:6 (include)
  C:/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

-- Project is not inside a git repository, or git repository has no
commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Warning at C:/esp-idf/tools/cmake/crosstool_version_check.cmake:8
(message):
  Xtensa toolchain

C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
  version is not the supported version 8.2.0.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  C:/esp-idf/components/esp32/project_include.cmake:30 (gcc_version_check)
  C:/esp-idf/tools/cmake/build.cmake:306 (include)
  C:/esp-idf/tools/cmake/build.cmake:451 (__build_process_project_includes)
  C:/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "hello-world" version: 1
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/esp/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt
cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common
esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server
esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom
esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi
[... 302 lines cut ...]
esp-idf/log/CMakeFiles/__idf_log.dir/log_buffers.c.obj
esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe
esp-idf\log\liblog.a && cd ."
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe:
esp-idf\log\liblog.a: No such file or directory
[73/100] Building C object
esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_utility.c.obj
[74/100] Building C object
esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_clock.c.obj
[75/100] Building C object
esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_flash.c.obj
[76/100] Building C object
esp-idf/micro-ecc/CMakeFiles/__idf_micro-ecc.dir/uECC_verify_antifault.c.obj
[77/100] Building C object
esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_common.c.obj
ninja: build stopped: subcommand failed.
[94/940] Building C object
esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/ssl_ciphersuites.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\esp\hello_world>

ср, 3 июн. 2020 г. в 17:53, Sergey Nikulin <serg.nikulin@gmail.com>:

snikul, 2020-06-03:
The same story with flash:

C:\esp\hello_world>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp\hello_world\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
CMake Deprecation Warning at
C:/Users/GreenOnion/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:75
(message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/toolchain-esp32.cmake:4 (CMAKE_FORCE_C_COMPILER)
  build/CMakeFiles/3.16.4/CMakeSystem.cmake:6 (include)
  C:/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

CMake Deprecation Warning at
C:/Users/GreenOnion/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:89
(message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/toolchain-esp32.cmake:5 (CMAKE_FORCE_CXX_COMPILER)
  build/CMakeFiles/3.16.4/CMakeSystem.cmake:6 (include)
  C:/esp-idf/tools/cmake/project.cmake:307 (__project)
  CMakeLists.txt:6 (project)

-- Project is not inside a git repository, or git repository has no
commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Warning at C:/esp-idf/tools/cmake/crosstool_version_check.cmake:8
(message):
  Xtensa toolchain

C:/Users/GreenOnion/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
  version is not the supported version 8.2.0.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  C:/esp-idf/components/esp32/project_include.cmake:30 (gcc_version_check)
  C:/esp-idf/tools/cmake/build.cmake:306 (include)
  C:/esp-idf/tools/cmake/build.cmake:451 (__build_process_project_includes)
  C:/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "hello-world" version: 1
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script
C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/esp/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt
cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common
esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server
esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom
esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi
espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn
json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash
[... 300 lines cut ...]
esp-idf/log/CMakeFiles/__idf_log.dir/log_buffers.c.obj
esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe
esp-idf\log\liblog.a && cd ."
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe:
esp-idf\log\liblog.a: No such file or directory
[73/100] Building C object
esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_utility.c.obj
[74/100] Building C object
esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_clock.c.obj
[75/100] Building C object
esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_flash.c.obj
[76/100] Building C object
esp-idf/micro-ecc/CMakeFiles/__idf_micro-ecc.dir/uECC_verify_antifault.c.obj
[77/100] Building C object
esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_common.c.obj
ninja: build stopped: subcommand failed.
[94/940] Building C object
esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/ssl_ciphersuites.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\esp\hello_world>

ср, 3 июн. 2020 г. в 19:11, Sergey Nikulin <serg.nikulin@gmail.com>:

renzbagaporo, 2020-06-04:
Can you try executing just an app build (`idf.py app`), see if that at least succeeds?

snikul, 2020-06-04:
> least succeeds? @snikul <https://github.com/snikul>

Here is the result:
***************
Using Python in C:\Users\GreenOnion\AppData\Local\Programs\Python\Python38\
Python 3.8.3
Using Git in C:\Program Files\Git\cmd\
git version 2.27.0.windows.1
Setting IDF_PATH: C:\esp-idf

Adding ESP-IDF tools to PATH...

C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin

C:\Users\GreenOnion\.espressif\tools\xtensa-esp32s2-elf\esp-2020r1-8.2.0\xtensa-esp32s2-elf\bin

C:\Users\GreenOnion\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin

C:\Users\GreenOnion\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\GreenOnion\.espressif\tools\cmake\3.16.4\bin

C:\Users\GreenOnion\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200420\openocd-esp32\bin
    C:\Users\GreenOnion\.espressif\tools\ninja\1.10.0\
    C:\Users\GreenOnion\.espressif\tools\idf-exe\1.0.1\
    C:\Users\GreenOnion\.espressif\tools\ccache\3.7\
    C:\Users\GreenOnion\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\GreenOnion\.espressif\python_env\idf4.2_py3.8_env\Scripts
    C:\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\esp-idf>cd ../esp

C:\esp>xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world
C:\esp-idf\examples\get-started\hello_world\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\Makefile
C:\esp-idf\examples\get-started\hello_world\README.md
C:\esp-idf\examples\get-started\hello_world\main\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\main\component.mk
C:\esp-idf\examples\get-started\hello_world\main\hello_world_main.c
Скопировано файлов: 6.

C:\esp>xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world
Заменить C:\esp\hello_world\CMakeLists.txt [Yes (да)/No (нет)/All (все)]? A
C:\esp-idf\examples\get-started\hello_world\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\Makefile
C:\esp-idf\examples\get-started\hello_world\README.md
C:\esp-idf\examples\get-started\hello_world\main\CMakeLists.txt
C:\esp-idf\examples\get-started\hello_world\main\component.mk
C:\esp-idf\examples\get-started\hello_world\main\hello_world_main.c
Скопировано файлов: 6.

C:\esp>cd hello_world

C:\esp\hello_world>idf.py app
Executing action: app
Running cmake in directory c:\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1
--warn-uninitialized -DCCACHE_ENABLE=1 c:\esp\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version
"2.27.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
CMake Deprecation Warning at
C:/Users/GreenOnion/.espressif/tools/cmake/3.16.4/share/cmake-3.16/Modules/CMakeForceCompiler.cmake:75
[... 129 lines cut ...]
C:/esp-idf/components/unity C:/esp-idf/components/vfs
C:/esp-idf/components/wear_levelling
C:/esp-idf/components/wifi_provisioning
C:/esp-idf/components/wpa_supplicant C:/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/esp/hello_world/build
Running ninja in directory c:\esp\hello_world\build
Executing "ninja app"...
[576/931] Linking CXX static library esp-idf\ulp\libulp.a
FAILED: esp-idf/ulp/libulp.a
cmd.exe /C "cd . &&
C:\Users\GreenOnion\.espressif\tools\cmake\3.16.4\bin\cmake.exe -E remove
esp-idf\ulp\libulp.a &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe
qc esp-idf\ulp\libulp.a  esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp.c.obj
esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp_macro.c.obj &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe
esp-idf\ulp\libulp.a && cd ."
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe:
esp-idf\ulp\libulp.a: No such file or directory
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\esp\hello_world>

renzbagaporo, 2020-06-04:
I see. So even that fails.

Removing the build directory before every run, does it fail at this part (`esp-idf\ulp\libulp.a`) every time?

```
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe
qc esp-idf\ulp\libulp.a  esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp.c.obj
esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp_macro.c.obj &&
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe
esp-idf\ulp\libulp.a && cd ."
C:\Users\GreenOnion\.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe:
esp-idf\ulp\libulp.a: No such file or directory
ninja: build stopped: subcommand failed.
```

snikul, 2020-06-05:
If I got your question correctly, I can run idf.py app or idf.py menuconfig
only once - right after installation. If I exit cmd I cannot start ESP-IDF
Command Prompt again using main menu cmd shortcut. It asks me to start from
esp-idf folder:
*************
This script must be invoked from ESP-IDF directory.

C:\Users\GreenOnion>
***************************
So I deindtall ESP-IDF Tools 2.3 then delete C:\esp\hello_world folder and
copy it again to esp folder
xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world
every time before  to start  ESP-IDF Tools 2.3.

So every time before to run idf.py app I have fresh copy of hello_world
without build folder.
Then I run idf.py app and get the error you seen.

I got the error 3 times. It fail at this part (esp-idf\ulp\libulp.a) every
time.
---------------
A question to you. How to clean installation cash? Or cash is clean after
deinstallation of ESP-IDF Tools 2.3 from installed apps list?
--------------
I tried again and got the same result
Executing "ninja app"...
[575/931] Linking CXX static library esp-idf\ulp\libulp.a
FAILED: esp-idf/ulp/libulp.a

snikul, 2020-06-06:
Hi, Renz,

I found the solution. Different one from the online manual. According to
https://www.youtube.com/watch?v=q0Kkl9qLwr0  step by step. Clever and
simple get-started guide.
Tell Igor to watch, pl. I am Russian too. Russians like simple and fast
solutions. I can move on now. But I still wonder what was that. That error.
Executing "ninja app"...
[575/931] Linking CXX static library esp-idf\ulp\libulp.a
FAILED: esp-idf/ulp/libulp.a

Regards
Sergei

пт, 5 июн. 2020 г. в 13:37, Sergey Nikulin <serg.nikulin@gmail.com>:

renzbagaporo, 2020-06-08:
Glad its working for you now @snikul. 

I've watched the video you linked and as far as I can tell, it's mostly similar apart from the somewhat redundant setting of environment variables, which should be unneeded if the `ESP-IDF Command Prompt` is used. Tagging @igrr if he knows something about this.

SureshDahiya, 2020-07-17:
Please help on this issue, if anyone has found reliable solution. I am struggling with this issue for a long time.
Settings:
1. Windows 10 Pro in C drive.
2. Other key software running: WSS Ubuntu 18.04 on windows, MySQL server etc.
3. Working Drive: D
4. System: Lenovo Z580 laptop

Findings are as follows:
1. Before using IDF v4.2 with CMAKE , I was using IDF V3.3 with MAKE where everything was OK including Eclipse.
2. I can use IDF V4.2 and associated tools on my other windows desktop using simple steps provided by Espressif but this windows is a fresh setup with a few other software and this is a light weight setup of windows 10.
3. All the logs on command prompt and in log files are same as mentioned in the above discussions.
4. All hit and trial have been done without positive result.
5. While comparing output on desktop and laptop one to one, only difference found was the setting DCCACHE_ENABLE=1 but I have not found where to configure this setting.

Help on two points will be greatly appreciated.
a). Information on where to configure DCCACHE_ENABLE=1
b). If someone may provide discrete commands to run and test the build process step by step.

Thanks
Suresh Dahiya

FJ-VEL, 2021-01-21:
Hi to everyone,

maybe it is a silly contribution but I found this thread searching the solution for the next problem, which more or less is similar to above comments, 

[image]

In my case the solution was really easy.
I sent the order to programming flash but I did not press the boot button on the PCB. So after a while the system shown me the above fail.

The button to press after launch flash programming is: 

[image]

I hope this can help to someone. 

Thanks

closed 2023-08-28 by dobairoland