espressif/esp-idf #3986: udp丢包的问题

hellosunvi, 2019-08-29:
编译sdk/examples/protocols/sockets/udp_server 例子程序，修改了代码，把发送回去的代码注释，只是记录上次接收数据包的时间和当前接收数据包的时间间隔，运行一个电脑软件，定时间隔30毫秒发送一包570字节的udp数据包到esp32模块，发现这个时间时间间隔和实际出入比较大，有时候间隔有150毫秒的延时。修改了lwip和wifi的缓冲区配置，效果没有什么变化，不知道大家有没有碰到这个问题？

Alvin1Zhang, 2019-08-29:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

hellosunvi, 2019-08-29:
> Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information [...]

硬件环境：ESP32-WROOM-32U，使用最新的sdk版本，udp_server 例子代码，用了默认的configuration，修改了wifi密码，当udp接收到数据时
                    if (sourceAddr.sin6_family == PF_INET) {
                    inet_ntoa_r(((struct sockaddr_in *)&sourceAddr)->sin_addr.s_addr, addr_str, sizeof(addr_str) - 1);
                } else if (sourceAddr.sin6_family == PF_INET6) {
                    inet6_ntoa_r(sourceAddr.sin6_addr, addr_str, sizeof(addr_str) - 1);
                }
                //修改部分
		int diff = system_get_time() - ticks;
		if (diff > 50000)
               {
		      printf("%d\n\r", diff);
                }
		ticks = system_get_time();
		//ESP_LOGI(TAG, "%s", rx_buffer);
                /*rx_buffer[len] = 0; // Null-terminate whatever we received and treat like a string...
                ESP_LOGI(TAG, "Received %d bytes from %s:", len, addr_str);
                ESP_LOGI(TAG, "%s", rx_buffer);

                int err = sendto(sock, rx_buffer, len, 0, (struct sockaddr *)&sourceAddr, sizeof(sourceAddr));
                if (err < 0) {
                    ESP_LOGE(TAG, "Error occured during sending: errno %d", errno);
                    break;
                }*/

hellosunvi, 2019-08-29:
硬件环境：ESP32-WROOM-32U，使用最新的sdk版本，udp_server 例子代码，用了默认的configuration，修改了wifi密码，当udp接收到数据时

int len = recvfrom(sock, rx_buffer, sizeof(rx_buffer) - 1, 0, (struct sockaddr *)&sourceAddr, &socklen);

            // Error occured during receiving
            if (len < 0) {
                ESP_LOGE(TAG, "recvfrom failed: errno %d", errno);
                break;
            }
            // Data received
            else {
                // Get the sender's ip address as string
                if (sourceAddr.sin6_family == PF_INET) {
                    inet_ntoa_r(((struct sockaddr_in *)&sourceAddr)->sin_addr.s_addr, addr_str, sizeof(addr_str) - 1);
                } else if (sourceAddr.sin6_family == PF_INET6) {
                    inet6_ntoa_r(sourceAddr.sin6_addr, addr_str, sizeof(addr_str) - 1);
                }
                //修改的代码
								int diff = system_get_time() - ticks;
								if (diff > 50000)
								{
									printf("%d\n\r", diff);
								}
								ticks = system_get_time();

                //以下代码注释了
                /*rx_buffer[len] = 0; // Null-terminate whatever we received and treat like a string...
                ESP_LOGI(TAG, "Received %d bytes from %s:", len, addr_str);
                ESP_LOGI(TAG, "%s", rx_buffer);

                int err = sendto(sock, rx_buffer, len, 0, (struct sockaddr *)&sourceAddr, sizeof(sourceAddr));
                if (err < 0) {
                    ESP_LOGE(TAG, "Error occured during sending: errno %d", errno);
                    break;
                }*/
            }

发现的现象就是打印的间隔时间有时候会超过150毫秒

TimXia, 2019-09-16:
请在 esp_wifi_init() 之后调用 esp_wifi_set_ps(WIFI_PS_NONE) 试试

Alvin1Zhang, 2020-01-15:
@hellosunvi Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

liuzfesp, 2020-03-25:
Hi @hellosunvi  

Could you describe "sometimes it has 150ms delay" in details? If it only have low probability, it's expected behavior.

To be detailed, the UDP receiving delay may cased by several possible reasons:

- The WiFi signal is weak, the router hardware requires about 150ms to send the UDP to ESP32
- The ESP32 fails to receives 1 UDP because of any reasons, then ESP32 goes to sleep because the  ESP32 will go to sleep if it hasn't receive any packets for 50ms. In other words, if ESP32 loses one UDP it will go to sleep. When ESP32 is in slept state, it can't receive any packets, the Router will NOT send any UDP to ESP32 and instead the Router buffers the UDP. In next DTIM interval, the ESP32 will wake up to receives the UDP again. The most common DTIM value is 1 or 3, and the beacon interval is 100ms. If the DTIM is 3, beacon interval is 100ms, then ESP32 has average 150ms delay to receive UDP if it goes to sleep.

liuzfesp, 2020-03-31:
Hi @hellosunvi, from our side, this issue look like expected behavior,  would you mind if I close it? Just free feel to reopen it if you have different opinions about it.

Alvin1Zhang, 2020-05-28:
@hellosunvi Thanks for reporting. Feel free to reopen if any opinions, thanks.

closed 2020-05-28 by Alvin1Zhang