espressif/esp-idf #3989: idf.py fails to compile because it cannot find and link common libraries!?

Micrified, 2019-08-29:
## Environment

- Development Kit: `ESP32-DevKitC`
- Kit version: `v4`
- Module or chip used:  `ESP32-WROOM-32`
- IDF version: `v4.0-dev-1443-g39f090a4f`
- Build System: `CMake`
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0`
- Operating System: `macOS 10.11.6`
- Power Supply: `USB`

## Problem Description

Invoking `idf.py build` on a custom project fails to compile, citing that the following libraries could not be found:

* stdc++
* libnewlib.a
* libhal.a
* lgcov
* lc
* lm
* lgcc

### Expected Behavior

I expect this project to compile successfully. **I am able to successfully build and flash this project on a machine running Ubuntu**. 

### Actual Behavior

I can successfully compile and flash the `hello-world` program from this machine. But I cannot compile this project or the example ones for BLE (e.g. BLE server example). The linker will consistently produce this output

### Steps to repropduce

1. Invoke `idf.py build`
2. Fails

### Code to reproduce this issue

The code is quite large spanning several files. The project is structured as follows:
```
- main
	| - include
	|	| - ble.h
	|	| - ble_profiles.h
	|	| - wifi.h
	| - src
	|	| - ble.c
	|	| - wifi.c
- CMakeLists.txt
- component.mk
- esp_32_wifi_main.c
```

Files `component.mk` is empty and unchanged. The file `CMakeLists.txt` contains:
```
idf_component_register(SRCS "esp32_wifi_main.c" "src/wifi.c" "src/ble.c"
                    INCLUDE_DIRS "include")
```

## Debug Logs

```
Checking Python dependencies...
Python requirements from /Users/Owatch/Documents/Somnox/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory /Users/Owatch/Documents/Somnox/esp32-wifi/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /Users/Owatch/Documents/Somnox/esp32-wifi"...
Warn about uninitialized values.
-- Found Git: /usr/local/bin/git (found version "2.19.0") 
-- Unexpected file in components directory: /Users/Owatch/Documents/Somnox/esp-idf/components/.DS_Store
[... 129 lines cut ...]
[61/62] Linking C executable bootloader.elf
[62/62] Generating binary image from built executable
esptool.py v2.7-dev
Generated /Users/Owatch/Documents/Somnox/esp32-wifi/build/bootloader/bootloader.bin
[1072/1073] Linking CXX executable esp32-wifi.elf
FAILED: esp32-wifi.elf 
: && /Users/Owatch/Documents/Somnox/esp-idf/tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/esp32-wifi.elf.dir/project_elf_src.c.obj  -o esp32-wifi.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/smartconfig_ack/libsmartconfig_ack.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/bt/libbt.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=/Users/Owatch/Documents/Somnox/esp32-wifi/build/esp32-wifi.map esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/smartconfig_ack/libsmartconfig_ack.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/bt/libbt.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp-tls/libesp-tls.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/protocomm/libprotocomm.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/nghttp/libnghttp.a esp-idf/bt/libbt.a -L/Users/Owatch/Documents/Somnox/esp-idf/components/bt/controller/lib -lbtdm_app esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjson.a esp-idf/cxx/libcxx.a esp-idf/newlib/libnewlib.a esp-idf/freertos/libfreertos.a esp-idf/heap/libheap.a esp-idf/log/liblog.a esp-idf/soc/libsoc.a esp-idf/esp_rom/libesp_rom.a esp-idf/esp_common/libesp_common.a esp-idf/xtensa/libxtensa.a esp-idf/esp32/libesp32.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/lwip/liblwip.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/spi_flash/libspi_flash.a esp-idf/efuse/libefuse.a esp-idf/app_update/libapp_update.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/smartconfig_ack/libsmartconfig_ack.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libcore.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libespnow.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libmesh.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libphy.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libpp.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/librtc.a /Users/Owatch/Documents/Somnox/esp-idf/components/esp_wifi/lib_esp32/libsmartconfig.a esp-idf/esp_event/libesp_event.a esp-idf/driver/libdriver.a es [... cut ...]
/Users/Owatch/Documents/Somnox/esp-idf/tools/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lgcc
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder): [Linked here](https://gist.github.com/Micrified/74b97855625e90e0475b65887fde705a)
- [ ] elf file in the ``build`` folder: **None**
- [ ] coredump: **None**

dobairoland, 2019-09-02:
Hi @Micrified. 

Please make sure that you get the same failure after you have removed the build directory (`rm -r build`).

Are you trying to create your own component or you just want an application which can be flashed? In the former case, I don't think that `esp_32_wifi_main.c` should be there (I'm assuming an `app_main` is inside). If the latter case, no `idf_component_register` is required.

Note that in the case of `idf_component_register` you probably want to specify its dependencies by the `REQUIRES` keyword (e.g. https://github.com/espressif/esp-idf/blob/master/components/app_update/CMakeLists.txt#L4).

Can you link please the exact path of the example which cannot be compiled at your end? That would be easier to use for tracking down your problem.

Micrified, 2019-09-02:
@dobairoland 

Hello. 

So I removed the build directory and tried again - same error. Since I'm trying to build an application that can be flashed, I want `esp_32_wifi_main.c` to be there. I build it by invoking: `idf.py build`, which is what produces these linker errors for me. 

What am I supposed to list in the requires section if these libraries don't exist? How come they exist if I build the exact same project on Linux? This is confusing to me a bit. The linker knows they should be there but it cannot find them for some reason. 

---

**Problem with compiling BLE example**

Alright I ran `tree` but only to three levels deep to show you the placement of the example with respect to `esp-idf`:

```
.
├── esp-idf
│   ├── CMakeLists.txt
│   ├── CONTRIBUTING.rst
│   ├── Kconfig
│   ├── LICENSE
│   ├── README.md
│   ├── add_path.sh
│   ├── components
│   │   ├── app_trace
│   │   ├── app_update
│   │   ├── asio
│   │   ├── bootloader
│   │   ├── bootloader_support
│   │   ├── bt
│   │   ├── coap
│   │   ├── console
│   │   ├── cxx
│   │   ├── driver
│   │   ├── efuse
│   │   ├── esp-tls
│   │   ├── esp32
│   │   ├── esp_adc_cal
│   │   ├── esp_common
│   │   ├── esp_eth
│   │   ├── esp_event
│   │   ├── esp_gdbstub
│   │   ├── esp_http_client
│   │   ├── esp_http_server
│   │   ├── esp_https_ota
│   │   ├── esp_https_server
│   │   ├── esp_local_ctrl
│   │   ├── esp_ringbuf
│   │   ├── esp_rom
│   │   ├── esp_websocket_client
│   │   ├── esp_wifi
│   │   ├── espcoredump
│   │   ├── esptool_py
│   │   ├── expat
│   │   ├── fatfs
│   │   ├── freemodbus
│   │   ├── freertos
│   │   ├── heap
│   │   ├── idf_test
│   │   ├── jsmn
│   │   ├── json
│   │   ├── libsodium
│   │   ├── log
│   │   ├── lwip
│   │   ├── mbedtls
│   │   ├── mdns
│   │   ├── mqtt
│   │   ├── newlib
│   │   ├── nghttp
│   │   ├── nvs_flash
│   │   ├── openssl
│   │   ├── partition_table
│   │   ├── protobuf-c
│   │   ├── protocomm
│   │   ├── pthread
│   │   ├── sdmmc
│   │   ├── smartconfig_ack
│   │   ├── soc
│   │   ├── spi_flash
│   │   ├── spiffs
│   │   ├── tcp_transport
│   │   ├── tcpip_adapter
│   │   ├── ulp
│   │   ├── unity
│   │   ├── vfs
│   │   ├── wear_levelling
│   │   ├── wifi_provisioning
│   │   ├── wpa_supplicant
│   │   └── xtensa
│   ├── docs
│   │   ├── Doxyfile
│   │   ├── README.md
│   │   ├── TEMPLATE_EXAMPLE_README.md
│   │   ├── _static
│   │   ├── check_doc_warnings.sh
│   │   ├── check_lang_folder_sync.sh
│   │   ├── conf_common.py
│   │   ├── docs_common.mk
│   │   ├── en
│   │   ├── gen-dxd.py
│   │   ├── gen-toolchain-links.py
│   │   ├── gen-version-specific-includes.py
│   │   ├── html_redirects.py
│   │   ├── issue_template.md
│   │   ├── link-roles.py
│   │   ├── local_util.py
│   │   ├── page_redirects.txt
│   │   ├── requirements.txt
│   │   ├── sphinx-known-warnings.txt
│   │   └── zh_CN
│   ├── examples
│   │   ├── README.md
│   │   ├── bluetooth
│   │   ├── build_system
│   │   ├── common_components
│   │   ├── ethernet
│   │   ├── get-started
│   │   ├── mesh
│   │   ├── peripherals
│   │   ├── protocols
│   │   ├── provisioning
│   │   ├── security
│   │   ├── storage
│   │   ├── system
│   │   └── wifi
│   ├── export.bat
│   ├── export.sh
│   ├── include_export.sh
│   ├── install.bat
│   ├── install.sh
│   ├── make
│   │   ├── common.mk
│   │   ├── component_common.mk
│   │   ├── component_wrapper.mk
│   │   ├── ldgen.mk
│   │   ├── project.mk
│   │   ├── project_config.mk
│   │   └── version.mk
│   ├── requirements.txt
│   ├── sdkconfig.rename
│   └── tools
│       ├── ble
│       ├── catch
│       ├── check_kconfigs.py
│       ├── check_python_dependencies.py
│       ├── ci
│       ├── cmake
│       ├── dist
│       ├── docker
│       ├── eclipse-code-style.xml
│       ├── elf_to_ld.sh
│       ├── esp_app_trace
│       ├── esp_prov
│       ├── format-minimal.sh
│       ├── format.sh
│       ├── gen_esp_err_to_name.py
│       ├── idf.py
│       ├── idf_monitor.py
│       ├── idf_size.py
│       ├── idf_tools.py
│       ├── kconfig
│       ├── kconfig_new
│       ├── ldgen
│       ├── mass_mfg
│       ├── python_env
│       ├── set-submodules-to-github.sh
│       ├── test_check_kconfigs.py
│       ├── test_idf_monitor
│       ├── test_idf_size
│       ├── test_idf_tools
│       ├── tiny-test-fw
│       ├── toolchain_versions.mk
│       ├── tools
│       ├── tools.json
│       ├── tools_schema.json
│       ├── unit-test-app
│       └── windows
└── gatt_server
    ├── CMakeLists.txt
    ├── Makefile
    ├── README.md
    ├── main
    │   ├── CMakeLists.txt
    │   ├── Kconfig
    │   ├── component.mk
    │   └── gatts_demo.c
    ├── sdkconfig
    ├── sdkconfig.defaults
    └── tutorial
        ├── Gatt_Server_Example_Walkthrough.md
        └── image
```

renzbagaporo, 2019-09-02:
Hi @Micrified,

Can you try increasing the open files limit? https://superuser.com/questions/827984/open-files-limit-does-not-work-as-before-in-osx-yosemite/1171026#1171026

dobairoland, 2019-09-02:
In that case, you don't need an `idf_component_register` statement. 
Your CMakeList.txt file should look as simple as https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server/CMakeLists.txt.

`esp-idf/examples/bluetooth/bluedroid/ble/gatt_server` compiles for me without any issues. You have the same error about missing standard libraries for that example as well?

Are you using the `install.sh` and `export.sh` scripts provided in the root ESP-IDF directory?

Micrified, 2019-09-03:
@dobairoland 

Hello again, 

Yeah, so my `CMakeList.txt` file basically is unchanged. The project simply spits out these errors when I try to compile. It compiles fine on my other machines. I've copied the example several times to compile it, and obtained the entire git repository again more than once. It always produces these compile errors. 

Yes I do use the `export.sh` scripts. I've also tried building using just `make`, and then `make flash monitor` but they basically fail for the same reasons. There must just be some misconfigured libraries somewhere but I'm not sure how to fix that because the build system shouldn't be looking only in my own system libraries to find these.

dobairoland, 2019-09-03:
I'm sorry but I have to ask again to confirm that you are talking about `examples/bluetooth/bluedroid/ble/gatt_server` and you didn't changed anything in it. Could you share the full build log for this example?

The given example can be compiled on Linux and MacOS as well and I cannot reproduce your issue.

Could you try a newer version of ESP-IDF (either the `master` branch or newer `release/v4.0`). There are a lot of related changes since the commit you are using. However, I can build even with the mentioned commit without an issue.

Another thing you could try is to remove the `~/.espressif/tools/xtensa-esp32-elf/` directory and reinstall it by `install.sh`.

Micrified, 2019-09-04:
Hey @dobairoland 

Yeah, it's that example precisely. In fact, I made a [screencast](https://youtu.be/9Xn5ztFG_ck) showing me successfully compiling `hello_world`, but **not** `gatt_server`. There are some time-stamps in the description you can jump to because my laptop took some time to compile them (all under 5m in total though)

I've yet to try increasing the open-file limit suggested by @renzbagaporo. I'll do that now.

dobairoland, 2019-09-09:
Hello @Micrified. Thank you checking all the ideas we gave you and making the screencast. I'm sorry we could not give you a direct and simple solution yet. Unfortunately, I still cannot reproduce the issue.  

We tried on macOS to build the same example using the same IDF version. 

Could you please give to me a verbose version of the build for the `gatt_server` example? I would like to examine it more closely. It will be long so it would be better to put into a file like this:
`rm -r build sdkconfig ; idf.py -v build > log_file 2>&1`

Thank you for your patience!

sieren, 2019-09-24:
Chiming in to say I'm having the same issue on macOS 10.14.6 (18G95) with exactly the same error messages.

sieren, 2019-09-24:
I solved it for myself by removing `~/.espressif` and reinstalling the toolchain.
But please, fix https://github.com/espressif/esp-idf/issues/4081 https://github.com/espressif/esp-idf/issues/4111 ... its kind of a joke that the toolchain server is unreachable for this long without making your install scripts ignore SSL certificates.

Micrified, 2019-09-24:
@dobairoland   Sorry I forgot about this thread when I took a break. Here's the [log file](https://gist.github.com/Micrified/4ab5db1f8379000970c4dbcf14c91334) if it's still relevant. I'll try what @sieren

dobairoland, 2019-09-25:
I'm happy for you @Micrified! So you did reinstall the toolchain and that helped? Or did you do anything else?

Micrified, 2019-09-25:
@dobairoland  I did what you asked I try, which is this: `rm -r build sdkconfig ; idf.py -v build > log_file 2>&1`

sieren, 2019-09-26:
I'm back at this error, turns out I was using the old toolchain when it resolved itself.
I tried to run the commands, but it seems like it breaks SPIFFS:

```
CMake Error at /Users/sierenmusic/Documents/Development/esp-idf/components/esptool_py/project_include.cmake:171 (file):
  file RELATIVE_PATH must be passed a full path to the file: FLASH_IN_PROJECT
Call Stack (most recent call first):
  /Users/sierenmusic/Documents/Development/esp-idf/components/spiffs/project_include.cmake:45 (esptool_py_flash_project_args)
  main/CMakeLists.txt:118 (spiffs_create_partition_image)
```

dobairoland, 2019-09-26:
Hi @sieren. Please create a new issue for the SPIFSS one because I believe these are unrelated. Thanks!

manuelbl, 2019-10-13:
After running into the same issue (*too many open files*), I've tried most of the options mentioned in this issue. Unfortunately, this issue does not go to the root of the cause. Instead, the conclusion is: *it resolved itself*.

I was able to fix it, and I can pinpoint a single measure: **increase the limit of open files and reboot**.

For increasing the limit, see https://wilsonmar.github.io/maximum-limits/ and follow the instructions for creating /Library/LaunchDaemons/limit.maxproc.plist. Then reboot. (I first tried with`sudo launchctl limit maxfiles 65536 200000` and it did not solve the issue.)

If this solution helps you as well, add a *thumbs up*.

julioppgca, 2019-10-14:
> After running into the same issue (_too many open files_), I've tried most of the options mentioned in this issue. Unfortunately, this issue does not go to the root of the cause. Instead, the [...]

Yeah! Thumbs up!
That's solved my problem!

Thank you.

lbrnmdev, 2019-10-16:
> Yeah! Thumbs up!
> That's solved my problem!
>
> Thank you.

I'm running into this issue as well (_too many open files_). In my case I'm trying to run the hello_world example from the esp-idf get-started guide.
I'm running macOS Mojave 10.14.6. My question is has the need to increase the open file limit always been the case or is this a recent issue? That is would using an older version of the esp-idf toolchain solve the issue?

manuelbl, 2019-10-16:
Yes, it's clearly related to ESP-IDF v4.x, either to changes in the build system or the newer toolchain (gcc etc.). Going back to ESP-IDF v3.3 and the older toolchain will solve it.

AniketRane, 2019-11-02:
> After running into the same issue (_too many open files_), I've tried most of the options mentioned in this issue. Unfortunately, this issue does not go to the root of the cause. Instead, the [...]

I am running macOS Mojave version 10.14.2 (18C54) and was having the same issue but using `sudo launchctl limit maxfiles 65536 200000` solved it.

**Remember**: The max files limit is reset at every reboot so make sure you add the command to your `~/.bash_profile `

Francois-Belanger, 2019-11-14:
Hi, having the same issue in macOS Catalina.

Using the sudo command works, but you need to close and re-open your terminal window to work. (or open a new terminal)

Hope this helps.

antoinevg, 2019-11-18:
Same issue in macOS Mojave 10.14.6

Neither the sudo command nor the launchctl solution worked.

For now I've dropped back to an earlier version (c19106876) that doesn't suffer from this problem.

Francois-Belanger, 2019-11-18:
Have you tried using a new console after using the sudo command ?

dobairoland, 2019-11-18:
Also there is a new version of the toolchain in the master branch (and will be available soon in v4.0 as well). Those who experience this problem could try to update to that and report back whether it solves the issue or not.

antoinevg, 2019-11-18:
Tried on new console, tried on reboot etc. etc.

Also, this was running against the latest commit to master branch: fb3d648

igrr, 2019-11-18:
Can you please try `ulimit -n 1024` (no sudo required, but has to be run in the same console where you are doing the build)

antoinevg, 2019-11-18:
Okay, that worked :-)

Some log files from building the `examples/get-started/hello_world` example against fb3d648.

Log for failing build: [fail.log](https://github.com/espressif/esp-idf/files/3860139/fail.log)

    rm -rf build sdkconfig && idf.py -v build > build.log 2>&1

Log for successful build: [success.log](https://github.com/espressif/esp-idf/files/3860149/success.log)

    ulimit -n 1024 && rm -rf build sdkconfig && idf.py -v build > build.log 2>&1

Thank you muchly for taking the time to look into this!

projectgus, 2019-12-08:
Reopening this as a fix has been merged on our internal branch, should push to GitHub soon.

Quentinb, 2019-12-15:
Running `sudo launchctl limit maxfiles 65536 200000` solved it for me on macOS Catalina 10.15.1

closed 2019-12-16 by espressif-bot

alexhenrie, 2021-06-17:
Is there any way to re-enable LTO when not building on Mac OS?