espressif/esp-idf #3990: VSPI as master not working!

Erichao990, 2019-08-30:
I use the ESP32-WROVER module VSPI as master,and try to send and receive data from a slave chip,but doesn't work.
I can't even get the CLK signal on oscilloscope. any one can help me figure out what problem it is ?
my init code is as below:

#define GPIO_VSPI_MOSI 23
#define GPIO_VSPI_MISO 19
#define GPIO_VSPI_SCLK 18
#define GPIO_VSPI_CS 5
void spi_master_init()
{
esp_err_t ret = ESP_OK;
spi_device_handle_t spi;
//Configuration for the SPI bus
spi_bus_config_t buscfg={
.mosi_io_num=GPIO_VSPI_MOSI,
.miso_io_num=GPIO_VSPI_MISO,
.sclk_io_num=GPIO_VSPI_SCLK,
.flags = SPICOMMON_BUSFLAG_NATIVE_PINS,
.quadwp_io_num=-1,
.quadhd_io_num=-1
};
//Configuration for the SPI device on the other side of the bus
spi_device_interface_config_t devcfg={
.command_bits=0,
.address_bits=0,
.dummy_bits=0,
.clock_speed_hz=510001000, //Clock out at 5 MHz,
.duty_cycle_pos=128, //50% duty cycle
.mode=0,
.spics_io_num=GPIO_VSPI_CS,
.cs_ena_posttrans=3, //Keep the CS low 3 cycles after transaction, to stop slave from missing the last bit when CS has less propagation delay than CLK
.queue_size=3
};
//Initialize the SPI bus and add the device we want to send stuff to.
ret=spi_bus_initialize(VSPI_HOST, &buscfg, 1);
assert(ret==ESP_OK);
ret=spi_bus_add_device(VSPI_HOST, &devcfg, &spi);
assert(ret==ESP_OK);
printf("[SPI] spi_master_init ok! SPI Type: %d ,mode: %d, speed: %d\r\n",VSPI_HOST,devcfg.mode,devcfg.clock_speed_hz);

uint8_t sendbuf[1] = {0x40};
uint8_t recvbuf[1] = {0};
spi_transaction_t t;
memset(&t, 0, sizeof(t));
t.length=8;
t.tx_buffer=sendbuf;
t.rx_buffer=recvbuf;
ret=spi_device_transmit(spi, &t);
printf("Received: %#x\n", recvbuf[0]);
}

Alvin1Zhang, 2019-08-30:
@Erichao990 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID and etc. would help us debug further. Thanks.

Erichao990, 2019-08-30:
My develop Environment is as below:
Module or chip used: [ESP32-WROVER|
IDF version : v4.0-dev-1287-gd7e659d
Build System: [Make|
Compiler version : // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: |Linux|
Power Supply: [USB|

Erichao990, 2019-08-30:
> @Erichao990 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? [...]

My develop Environment is as below:
Module or chip used: [ESP32-WROVER|
IDF version : v4.0-dev-1287-gd7e659d
Build System: [Make|
Compiler version : // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: |Linux|
Power Supply: [USB|

My chip type is :ESP32-DOWDQ6

edwin-oetelaar, 2019-08-30:
I can not be sure, but I do notice something here.

You do not initialize all fields of the spi_bus_config_t.
These do not default to 0x00 they are on the stack.
```

typedef struct {
    int mosi_io_num;                ///< GPIO pin for Master Out Slave In (=spi_d) signal, or -1 if not used.
    int miso_io_num;                ///< GPIO pin for Master In Slave Out (=spi_q) signal, or -1 if not used.
    int sclk_io_num;                ///< GPIO pin for Spi CLocK signal, or -1 if not used.
    int quadwp_io_num;              ///< GPIO pin for WP (Write Protect) signal which is used as D2 in 4-bit communication modes, or -1 if not used.
    int quadhd_io_num;              ///< GPIO pin for HD (HolD) signal which is used as D3 in 4-bit communication modes, or -1 if not used.
    int max_transfer_sz;            ///< Maximum transfer size, in bytes. Defaults to 4094 if 0.
    uint32_t flags;                 ///< Abilities of bus to be checked by the driver. Or-ed value of ``SPICOMMON_BUSFLAG_*`` flags.
    int intr_flags;    /**< Interrupt flag for the bus to set the priority, and IRAM attribute, see
                         *  ``esp_intr_alloc.h``. Note that the EDGE, INTRDISABLED attribute are ignored
                         *  by the driver. Note that if ESP_INTR_FLAG_IRAM is set, ALL the callbacks of
                         *  the driver, and their callee functions, should be put in the IRAM.
                         */
} spi_bus_config_t;
```

set some values for intr_flags and flags and for max_transfer_sz maybe that will help.

Erichao990, 2019-09-02:
> I can not be sure, but I do notice something here.
>
> You do not initialize all fields of the spi_bus_config_t.
> These do not default to 0x00 they are on the stack.
>
> ```
>
> typedef struct {
>     int [...]

Thanks for your answering,but I read all the examples of SPI init,no one use these two flags,I am not sure the correct value to config them,did you know that?

edwin-oetelaar, 2019-09-02:
I may have spoken too soon.
I have looked at C standards and it seems that doing something like this

`
typedef struct {
int a;
int b;
int c;
} bus_t;

bus_t bt = { .b=2 };
`

should initialise all members to 0x00 and bt.b to 2.
However, my recent experience with the ESP32 ESP-IDF 4.x (latest git) and gcc 8.2 tells me something else.
I will try to find an example later.
I will try to

ginkgm, 2019-09-03:
Hi @Erichao990 and all,
gd7e659d is not a valid IDF commit, did you modify it yourself?
About this code, I can see the clock output on my IDF commit b08dcefaf. however, `.clock_speed_hz=510001000, //Clock out at 5 MHz,` results in a output clock of 80MHz (510MHz rounded to 80MHz, the maximum supported by ESP32), maybe you should check whether this frequency is supported by your slave (usually ESP32 cannot read at such high speed, even when the slave is ideal, see https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#timing-considerations)

Also, the C standard guarantee that the following statement will clear all other members than member_a. It's also the suggested way to initialize the spi driver configuration structures.
```
type_t t = {
.member_a = X,
};
```

Erichao990, 2019-09-04:
> Hi @Erichao990 and all,
> gd7e659d is not a valid IDF commit, did you modify it yourself?
> About this code, I can see the clock output on my IDF commit b08dcefaf. however, `.clock_speed_hz=510001000, [...]

Thank you  so much for your detailed answer,I've got the problem and it's working now,hha.

negativekelvin, 2019-09-04:
@ginkgm  https://github.com/espressif/esp-idf/commit/d7e659d

I thought `.clock_speed_hz=510001000,` was just a problem with a markdown parser removing the asterisks from `.clock_speed_hz=5*1000*1000,` but maybe not

ginkgm, 2019-09-04:
@negativekelvin thanks for your clue~.

@Erichao990 Do you want to share the cause & can we close this thread?

closed 2019-09-05 by Erichao990